respeaker/seeed-voicecard #4: Voice response too fast when using google assistand SDK with Hotword

MyNameWasUnavailable, 2017-08-29:
Myself and quite a number of others are having issues with this device when using the google assistant SDK.

Audio tests seem to work correctly using the steps in "Configure and test the audio" from the official sdk, however when running the assistant the playback is sped up considerably, to the point where it is unintelligible.

This issue occurs with both the Supplied raspbian image and a manual installation.

xiongyihui, 2017-08-30:
Enabling pulseaudio will solve the issue.
It seems that google assistant's output is mono, but is played as stereo, so the speech voice is twice.

csprimary, 2017-09-03:
Enabling pulseaudio with the command

`pulseaudio &`

before running the Google Assistant demo results in the following error

`W: [alsa-sink-bcm2835-i2s-wm8960-hifi wm8960-hifi-0] alsa-sink.c: **Resume failed, couldn't restore original sample settings.**`

being scrolled infinitely down the terminal.

Disabling pulseaudio results in the original fast playback issue. Have still not been able to resolve this.

MyNameWasUnavailable, 2017-09-03:
I can confirm that my results are the same as the previous poster.

Can you please advise what the next steps would be?

xiongyihui, 2017-09-04:
My pulseaudio works. Can you post the log of `pulseaudio -vvvv`

xiongyihui, 2017-09-06:
try the 2 mic voice card with the latest raspbian image. The pulseaudio doesn't work by default. We must modify pulseaudio's configuration file `/etc/pulse/default.pa`. Find the two lines which contain `module-alsa` and change the lines as the following:
```
load-module module-alsa-sink device=hw:0,0
load-module module-alsa-source device=hw:0,0
```
and comment the following lines: 
```
### Automatically load driver modules depending on the hardware available
#.ifexists module-udev-detect.so
#load-module module-udev-detect
#.else
### Use the static hardware detection module (for systems that lack udev support)
#load-module module-detect
#.endif
```

If the on-board codec is enabled, the seeed voice card should be `hw:1,0`, replace `hw:0,0` to `hw:0,0`. Run `arecord -l` to check the voice card's hardware number.

Reboot to use the new configuration. Besides, pavucontrol and audacity can be used to debug audio R/W.

xiongyihui, 2017-09-06:
Another way to solve the issue is to use Ubuntu Mate for RPi 2/3 which is more better desktop OS.

xiongyihui, 2017-09-06:
@MyNameWasUnavailable @csprimary I got another solution. Run:

```
wget https://github.com/respeaker/seeed-voicecard/raw/master/asound.conf
sudo cp asound.conf /etc/asound.conf
```

The alsa configuration uses dmix plugin and dsnoop plugin. With the configuration, we can play and record with multiple applications simultaneously.

MyNameWasUnavailable, 2017-09-06:
I will try the above solutions, most likely on the weekend, and report back.

aceycv8, 2017-09-07:
i've just done a fresh install and i'm having issues getting this to work also..

if i apt get pulseaudio and set it up with your settings that you mention above then i run the SDK - when i say something it hears what i say and searches, but then the response freezes and nothing happens.

if i uninstall pulseaudio google speaks back very fast but atleast it's working - using 
wget https://github.com/respeaker/seeed-voicecard/raw/master/asound.conf
sudo cp asound.conf /etc/asound.conf

doesn't make any difference

xiongyihui, 2017-09-07:
@aceycv8 When google assistant freezes using pulseaudio, can you open pavucontrol, check the `Playback` tab and `Output Devices` tab and post a screenshot here?
To install pavucontrol, run: `sudo apt install pavucontrol`

My pavucontrol looks like:
[image]

aceycv8, 2017-09-07:
I'm using the light version so its all cli - is there anything I can do to check through cli?

xiongyihui, 2017-09-07:
You can use `pactl list sinks` to check output devices. and use `pactl list sink-inputs` to check applications  which are playing.

The log is like:
```
(env) pi@raspberrypi:~ $ pactl list sink-inputs 
Sink Input #19
	Driver: protocol-native.c
	Owner Module: 7
	Client: 32
	Sink: 0
	Sample Specification: s16le 2ch 48000Hz
	Channel Map: front-left,front-right
	Format: pcm, format.sample_format = "\"s16le\""  format.rate = "48000"  format.channels = "2"  format.channel_map = "\"front-left,front-right\""
	Corked: no
	Mute: no
	Volume: front-left: 63571 /  97% / -0.79 dB,   front-right: 63571 /  97% / -0.79 dB
	        balance 0.00
	Buffer Latency: 77605 usec
	Sink Latency: 19708 usec
	Resample method: speex-fixed-1
	Properties:
		media.name = "ALSA Playback"
		application.name = "ALSA plug-in [python3.5]"
		native-protocol.peer = "UNIX socket client"
		native-protocol.version = "32"
		application.process.id = "4889"
		application.process.user = "pi"
		application.process.host = "raspberrypi"
		application.process.binary = "python3.5"
		application.language = "C"
		window.x11.display = ":0.0"
		application.process.machine_id = "143b13e352bc41dc89c2ddbba64df665"
		application.process.session_id = "c1"
		application.icon_name = "/usr/share/pixmaps/python3.5.xpm"
		module-stream-restore.id = "sink-input-by-application-name:ALSA plug-in [python3.5]"
(env) pi@raspberrypi:~ $ pactl list sinks
Sink #0
	State: IDLE
	Name: alsa_output.hw_0_0
	Description: seeed-voicecard
	Driver: module-alsa-sink.c
	Sample Specification: s16le 2ch 44100Hz
	Channel Map: front-left,front-right
	Owner Module: 5
	Mute: no
	Volume: front-left: 63571 /  97% / -0.79 dB,   front-right: 63571 /  97% / -0.79 dB
	        balance 0.00
	Base Volume: 65536 / 100% / 0.00 dB
	Monitor Source: alsa_output.hw_0_0.monitor
	Latency: 28574 usec, configured 40000 usec
	Flags: HARDWARE DECIBEL_VOLUME LATENCY 
	Properties:
		alsa.resolution_bits = "16"
		device.api = "alsa"
		device.class = "sound"
		alsa.class = "generic"
		alsa.subclass = "generic-mix"
		alsa.name = ""
		alsa.id = "bcm2835-i2s-wm8960-hifi wm8960-hifi-0"
		alsa.subdevice = "0"
		alsa.subdevice_name = "subdevice #0"
		alsa.device = "0"
		alsa.card = "0"
		alsa.card_name = "seeed-voicecard"
		alsa.long_card_name = "seeed-voicecard"
		alsa.driver_name = "snd_soc_simple_card"
		device.bus_path = "platform-soc:sound"
		sysfs.path = "/devices/platform/soc/soc:sound/sound/card0"
		device.string = "hw:0,0"
		device.buffering.buffer_size = "352800"
		device.buffering.fragment_size = "176400"
		device.access_mode = "mmap+timer"
		device.description = "seeed-voicecard"
		device.icon_name = "audio-card"
	Formats:
		pcm

```

aceycv8, 2017-09-07:
thank you xiongyihui - here is my output where it still freezes.

pi@VoiceAI:~ $ pactl list sinks
Sink #0
        State: SUSPENDED
        Name: alsa_output.hw_0_0
        Description: seeed-voicecard
        Driver: module-alsa-sink.c
        Sample Specification: s16le 2ch 48000Hz
        Channel Map: front-left,front-right
        Owner Module: 5
        Mute: no
        Volume: front-left: 65536 / 100% / 0.00 dB,   front-right: 65536 / 100% / 0.00 dB
                balance 0.00
        Base Volume: 65536 / 100% / 0.00 dB
        Monitor Source: alsa_output.hw_0_0.monitor
        Latency: 0 usec, configured 0 usec
        Flags: HARDWARE DECIBEL_VOLUME LATENCY
        Properties:
                alsa.resolution_bits = "16"
                device.api = "alsa"
                device.class = "sound"
                alsa.class = "generic"
                alsa.subclass = "generic-mix"
                alsa.name = ""
                alsa.id = "bcm2835-i2s-wm8960-hifi wm8960-hifi-0"
                alsa.subdevice = "0"
                alsa.subdevice_name = "subdevice #0"
                alsa.device = "0"
                alsa.card = "0"
                alsa.card_name = "seeed-voicecard"
                alsa.long_card_name = "seeed-voicecard"
                alsa.driver_name = "snd_soc_simple_card"
                device.bus_path = "platform-soc:sound"
                sysfs.path = "/devices/platform/soc/soc:sound/sound/card0"
                device.string = "hw:0,0"
                device.buffering.buffer_size = "352800"
                device.buffering.fragment_size = "176400"
                device.access_mode = "mmap+timer"
                device.description = "seeed-voicecard"
                device.icon_name = "audio-card"
        Formats:
                pcm

pi@VoiceAI:~ $ pactl list sink-inputs
Sink Input #1
        Driver: protocol-native.c
        Owner Module: 7
        Client: 6
        Sink: 0
        Sample Specification: s16le 2ch 48000Hz
        Channel Map: front-left,front-right
        Format: pcm, format.sample_format = "\"s16le\""  format.rate = "48000"  format.channels = "2"  format.channel_map = "\"front-left,front-right\""
        Corked: no
        Mute: no
        Volume: front-left: 65536 / 100% / 0.00 dB,   front-right: 65536 / 100% / 0.00 dB
                balance 0.00
        Buffer Latency: 90000 usec
        Sink Latency: 0 usec
        Resample method: n/a
        Properties:
                media.name = "ALSA Playback"
                application.name = "ALSA plug-in [python2.7]"
                native-protocol.peer = "UNIX socket client"
                native-protocol.version = "29"
                application.process.id = "894"
                application.process.user = "pi"
                application.process.host = "VoiceAI"
                application.process.binary = "python2.7"
                application.language = "C"
                application.process.machine_id = "811d63baac94455983b6a5a0609efcba"
                application.process.session_id = "c1"
                application.icon_name = "/usr/share/pixmaps/python2.7.xpm"
                module-stream-restore.id = "sink-input-by-application-name:ALSA plug-in [python2.7]"

xiongyihui, 2017-09-08:
The only difference I found is:
```
pi@VoiceAI:~ $ pactl list sinks
Sink #0
State: SUSPENDED
Name: alsa_output.hw_0_0
Description: seeed-voicecard
Driver: module-alsa-sink.c
Sample Specification: s16le 2ch 48000Hz
```
vs
```
(env) pi@raspberrypi:~ $ pactl list sinks
Sink #0
	State: IDLE
	Name: alsa_output.hw_0_0
	Description: seeed-voicecard
	Driver: module-alsa-sink.c
	Sample Specification: s16le 2ch 44100Hz
```

@aceycv8 When the assistant is freezing, are you able to play sound using other applications such as aplay, audacity?

aceycv8, 2017-09-08:
Yep i'm able to do a speaker test fine once assistant freezes - 

i run  `speaker-test -c 2 -t wav -D plughw:0,0`

```
speaker-test 1.0.28

Playback device is plughw:0,0
Stream parameters are 48000Hz, S16_LE, 2 channels
WAV file(s)
Rate set to 48000Hz (requested 48000Hz)
Buffer size range from 139 to 142663
Period size range from 69 to 71332
Using max buffer size 142660
Periods = 4
was set period_size = 2691
was set buffer_size = 142660
 0 - Front Left
 1 - Front Right
```

and audio comes through while assistant is frozen.

here are the pactl commands while assistant is frozen and the speaker test is running.

```
pi@VoiceAI:~ $ pactl list sinks
Sink #0
        State: SUSPENDED
        Name: alsa_output.hw_0_0
        Description: seeed-voicecard
        Driver: module-alsa-sink.c
        Sample Specification: s16le 2ch 48000Hz
        Channel Map: front-left,front-right
        Owner Module: 5
        Mute: no
        Volume: front-left: 65536 / 100% / 0.00 dB,   front-right: 65536 / 100% / 0.00 dB
                balance 0.00
        Base Volume: 65536 / 100% / 0.00 dB
        Monitor Source: alsa_output.hw_0_0.monitor
        Latency: 0 usec, configured 0 usec
        Flags: HARDWARE DECIBEL_VOLUME LATENCY
        Properties:
                alsa.resolution_bits = "16"
                device.api = "alsa"
                device.class = "sound"
                alsa.class = "generic"
                alsa.subclass = "generic-mix"
                alsa.name = ""
                alsa.id = "bcm2835-i2s-wm8960-hifi wm8960-hifi-0"
                alsa.subdevice = "0"
                alsa.subdevice_name = "subdevice #0"
                alsa.device = "0"
                alsa.card = "0"
                alsa.card_name = "seeed-voicecard"
                alsa.long_card_name = "seeed-voicecard"
                alsa.driver_name = "snd_soc_simple_card"
                device.bus_path = "platform-soc:sound"
                sysfs.path = "/devices/platform/soc/soc:sound/sound/card0"
                device.string = "hw:0,0"
                device.buffering.buffer_size = "352800"
                device.buffering.fragment_size = "176400"
                device.access_mode = "mmap+timer"
                device.description = "seeed-voicecard"
                device.icon_name = "audio-card"
        Formats:
                pcm
pi@VoiceAI:~ $

pi@VoiceAI:~ $ pactl list sink-inputs
Sink Input #1
        Driver: protocol-native.c
        Owner Module: 7
        Client: 6
        Sink: 0
        Sample Specification: s16le 2ch 48000Hz
        Channel Map: front-left,front-right
        Format: pcm, format.sample_format = "\"s16le\""  format.rate = "48000"  format.channels = "2"  format.channel_map = "\"front-left,front-right\""
        Corked: no
        Mute: no
        Volume: front-left: 65536 / 100% / 0.00 dB,   front-right: 65536 / 100% / 0.00 dB
                balance 0.00
        Buffer Latency: 90000 usec
        Sink Latency: 0 usec
        Resample method: n/a
        Properties:
                media.name = "ALSA Playback"
                application.name = "ALSA plug-in [python2.7]"
                native-protocol.peer = "UNIX socket client"
                native-protocol.version = "29"
                application.process.id = "944"
                application.process.user = "pi"
                application.process.host = "VoiceAI"
                application.process.binary = "python2.7"
                application.language = "C"
                application.process.machine_id = "8196dca91c114c3ca89d5fccff66bf84"
                application.process.session_id = "c1"
                application.icon_name = "/usr/share/pixmaps/python2.7.xpm"
                module-stream-restore.id = "sink-input-by-application-name:ALSA plug-in [python2.7]"

```

xiongyihui, 2017-09-08:
`speaker-test` with `-D plughw:0,0` will bypass pulseaudio. How about just run `speaker-test`?

If the speaker-test is using pulseaudio, it will show in the output of `pactl list sink-inputs`
```
pi@raspberrypi:~ $ pactl list sink-inputs
Sink Input #7
	Driver: protocol-native.c
	Owner Module: 7
	Client: 10
	Sink: 0
	Sample Specification: s16le 1ch 48000Hz
	Channel Map: mono
	Format: pcm, format.sample_format = "\"s16le\""  format.rate = "48000"  format.channels = "1"  format.channel_map = "\"mono\""
	Corked: no
	Mute: no
	Volume: mono: 17364 /  26% / -34.61 dB
	        balance 0.00
	Buffer Latency: 4787625 usec
	Sink Latency: 39143 usec
	Resample method: speex-fixed-1
	Properties:
		media.name = "ALSA Playback"
		application.name = "ALSA plug-in [speaker-test]"
		native-protocol.peer = "UNIX socket client"
		native-protocol.version = "32"
		application.process.id = "1065"
		application.process.user = "pi"
		application.process.host = "raspberrypi"
		application.process.binary = "speaker-test"
		application.language = "en_US.UTF-8"
		window.x11.display = ":0.0"
		application.process.machine_id = "143b13e352bc41dc89c2ddbba64df665"
		application.process.session_id = "c1"
		module-stream-restore.id = "sink-input-by-application-name:ALSA plug-in [speaker-test]"
```

xiongyihui, 2017-09-08:
I just uploaded my pulseaudio configuration file `default.pa`. You can verify it.

```
wget https://github.com/respeaker/seeed-voicecard/raw/master/default.pa
diff default.pa /etc/pulse/default.pa
sudo cp default.pa /etc/pulse
```

aceycv8, 2017-09-08:
ok - 

i've copied your default.pa down.

running `speaker-test` by it's self the first time didn't work and i had to ctrl-c to close it - then the second time it worked with a hissing noise. 

So i ran assistant and exactly the same thing just freezes.

i've left assistant open and now re run `speaker-test` multiple times and i get no audio at all.

here is the `pactl list sinks` log while assistant is running and `speaker-test` is running ( but no audio at all )
```
pi@VoiceAI:~ $ pactl list sinks
Sink #0
        State: SUSPENDED
        Name: alsa_output.hw_0_0
        Description: seeed-voicecard
        Driver: module-alsa-sink.c
        Sample Specification: s16le 2ch 48000Hz
        Channel Map: front-left,front-right
        Owner Module: 5
        Mute: no
        Volume: front-left: 65536 / 100% / 0.00 dB,   front-right: 65536 / 100% / 0.00 dB
                balance 0.00
        Base Volume: 65536 / 100% / 0.00 dB
        Monitor Source: alsa_output.hw_0_0.monitor
        Latency: 0 usec, configured 0 usec
        Flags: HARDWARE DECIBEL_VOLUME LATENCY
        Properties:
                alsa.resolution_bits = "16"
                device.api = "alsa"
                device.class = "sound"
                alsa.class = "generic"
                alsa.subclass = "generic-mix"
                alsa.name = ""
                alsa.id = "bcm2835-i2s-wm8960-hifi wm8960-hifi-0"
                alsa.subdevice = "0"
                alsa.subdevice_name = "subdevice #0"
                alsa.device = "0"
                alsa.card = "0"
                alsa.card_name = "seeed-voicecard"
                alsa.long_card_name = "seeed-voicecard"
                alsa.driver_name = "snd_soc_simple_card"
                device.bus_path = "platform-soc:sound"
                sysfs.path = "/devices/platform/soc/soc:sound/sound/card0"
                device.string = "hw:0,0"
                device.buffering.buffer_size = "352800"
                device.buffering.fragment_size = "176400"
                device.access_mode = "mmap+timer"
                device.description = "seeed-voicecard"
                device.icon_name = "audio-card"
        Formats:
                pcm
```

and here is the `pactl list sink-inputs` at the same time.

```
pi@VoiceAI:~ $ pactl list sink-inputs
Sink Input #9
        Driver: protocol-native.c
        Owner Module: 7
        Client: 17
        Sink: 0
        Sample Specification: s16le 2ch 48000Hz
        Channel Map: front-left,front-right
        Format: pcm, format.sample_format = "\"s16le\""  format.rate = "48000"  format.channels = "2"  format.channel_map = "\"front-left,front-right\""
        Corked: no
        Mute: no
        Volume: front-left: 65536 / 100% / 0.00 dB,   front-right: 65536 / 100% / 0.00 dB
                balance 0.00
        Buffer Latency: 90000 usec
        Sink Latency: 0 usec
        Resample method: n/a
        Properties:
                media.name = "ALSA Playback"
                application.name = "ALSA plug-in [python2.7]"
                native-protocol.peer = "UNIX socket client"
                native-protocol.version = "29"
                application.process.id = "983"
                application.process.user = "pi"
                application.process.host = "VoiceAI"
                application.process.binary = "python2.7"
                application.language = "C"
                application.process.machine_id = "18fbcba7d1054997aab1e4a9d67e20e0"
                application.process.session_id = "c2"
                application.icon_name = "/usr/share/pixmaps/python2.7.xpm"
                module-stream-restore.id = "sink-input-by-application-name:ALSA plug-in [python2.7]"

Sink Input #26
        Driver: protocol-native.c
        Owner Module: 7
        Client: 29
        Sink: 0
        Sample Specification: s16le 1ch 48000Hz
        Channel Map: mono
        Format: pcm, format.sample_format = "\"s16le\""  format.rate = "48000"  format.channels = "1"  format.channel_map = "\"mono\""
        Corked: no
        Mute: no
        Volume: mono: 65536 / 100% / 0.00 dB
                balance 0.00
        Buffer Latency: 10922666 usec
        Sink Latency: 0 usec
        Resample method: copy
        Properties:
                media.name = "ALSA Playback"
                application.name = "ALSA plug-in [speaker-test]"
                native-protocol.peer = "UNIX socket client"
                native-protocol.version = "29"
                application.process.id = "1060"
                application.process.user = "pi"
                application.process.host = "VoiceAI"
                application.process.binary = "speaker-test"
                application.language = "C"
                application.process.machine_id = "18fbcba7d1054997aab1e4a9d67e20e0"
                application.process.session_id = "c3"
                module-stream-restore.id = "sink-input-by-application-name:ALSA plug-in [speaker-test]"

```

I then close assistant down and also close the speaker-test that was running and re run `speaker-test` multiple times with no sound at all

i then run `speaker-test -c 2 -t wav -D plughw:0,0` and i get sound.

i run `speaker-test` again and nothing.

however if i run `sudo speaker-test` i get a hissing sound like the very first time.

xiongyihui, 2017-09-08:
@aceycv8 would you mind to join the gitter seeed-voicecard chat room? It should be easier to discuss the issue in the chat room.

https://gitter.im/seeed-voicecard/Lobby?utm_source=share-link&utm_medium=link&utm_campaign=share-link

aceycv8, 2017-09-08:
just joined in

aceycv8, 2017-09-09:
@xiongyihui  - i downloaded your Pi Image and set everything back up.

i tried the following and rebooted then ran assistant with no joy - fast playback

`wget https://github.com/respeaker/seeed-voicecard/raw/master/default.pa
diff default.pa /etc/pulse/default.pa
sudo cp default.pa /etc/pulse`

I then did

`wget https://github.com/respeaker/seeed-voicecard/raw/master/asound.conf
sudo cp asound.conf /etc/asound.conf` 

and ran assistant - this fixed my problem right away - i'm using assistant with the LED script of 
`python google-assistant.py`

and so far so good, 

Thank you for your help 

now just to figure out how to get a beep to happen when you say "ok google"

thank you

MyNameWasUnavailable, 2017-09-12:
I have been as time allowed today trying the suggestions in the gitter chat room, using the seed supplied image and my outcome is as follows:

*just using the asounf.conf results in chipmunk speed speech.
* using the default.pa you uploaded, and using pulseaudio I get the following output:

ON_RESPONDING_STARTED:
  {'is_error_response': False}
W: [alsa-sink-bcm2835-i2s-wm8960-hifi wm8960-hifi-0] alsa-sink.c: Resume failed, couldn't restore original sample settings.
W: [alsa-sink-bcm2835-i2s-wm8960-hifi wm8960-hifi-0] alsa-sink.c: Resume failed, couldn't restore original sample settings.

Output from pactl list-sink-inputs:

pi@raspberrypi:~/env/bin $ pactl list sink-inputs
Sink Input #1
        Driver: protocol-native.c
        Owner Module: 11
        Client: 6
        Sink: 0
        Sample Specification: s16le 2ch 48000Hz
        Channel Map: front-left,front-right
        Format: pcm, format.sample_format = "\"s16le\""  format.rate = "48000"  format.channels = "2"  format.channel_map = "\"front-left,front-right\""
        Corked: no
        Mute: no
        Volume: front-left: 65536 / 100% / 0.00 dB,   front-right: 65536 / 100% / 0.00 dB
                balance 0.00
        Buffer Latency: 90000 usec
        Sink Latency: 0 usec
        Resample method: n/a
        Properties:
                media.name = "ALSA Playback"
                application.name = "ALSA plug-in [python3.4]"
                native-protocol.peer = "UNIX socket client"
                native-protocol.version = "29"
                application.process.id = "1251"
                application.process.user = "pi"
                application.process.host = "raspberrypi"
                application.process.binary = "python3.4"
                application.language = "C"
                application.process.machine_id = "aa0a4892531c42499224a4d1b2606e1c"
                application.process.session_id = "c3"
                application.icon_name = "/usr/share/pixmaps/python3.4.xpm"
                module-stream-restore.id = "sink-input-by-application-name:ALSA plug-in [python3.4]"
pi@raspberrypi:~/env/bin $

If I try and play some audio from another session, nothing plays, and I get another recurrance of the following message in the first session:
W: [alsa-sink-bcm2835-i2s-wm8960-hifi wm8960-hifi-0] alsa-sink.c: Resume failed, couldn't restore original sample settings.

output from pactl list sink-inputs

pi@raspberrypi:~/env/bin $ pactl list sink-inputs
Sink Input #1
        Driver: protocol-native.c
        Owner Module: 11
        Client: 6
        Sink: 0
        Sample Specification: s16le 2ch 48000Hz
        Channel Map: front-left,front-right
        Format: pcm, format.sample_format = "\"s16le\""  format.rate = "48000"  format.channels = "2"  format.channel_map = "\"front-left,front-right\""
        Corked: no
        Mute: no
        Volume: front-left: 65536 / 100% / 0.00 dB,   front-right: 65536 / 100% / 0.00 dB
                balance 0.00
        Buffer Latency: 90000 usec
        Sink Latency: 0 usec
        Resample method: n/a
        Properties:
                media.name = "ALSA Playback"
                application.name = "ALSA plug-in [python3.4]"
                native-protocol.peer = "UNIX socket client"
                native-protocol.version = "29"
                application.process.id = "1251"
                application.process.user = "pi"
                application.process.host = "raspberrypi"
                application.process.binary = "python3.4"
                application.language = "C"
                application.process.machine_id = "aa0a4892531c42499224a4d1b2606e1c"
                application.process.session_id = "c3"
                application.icon_name = "/usr/share/pixmaps/python3.4.xpm"
                module-stream-restore.id = "sink-input-by-application-name:ALSA plug-in [python3.4]"

Sink Input #4
        Driver: protocol-native.c
        Owner Module: 11
        Client: 21
        Sink: 0
        Sample Specification: s16le 2ch 44100Hz
        Channel Map: front-left,front-right
        Format: pcm, format.sample_format = "\"s16le\""  format.rate = "44100"  format.channels = "2"  format.channel_map = "\"front-left,front-right\""
        Corked: no
        Mute: no
        Volume: front-left: 65536 / 100% / 0.00 dB,   front-right: 65536 / 100% / 0.00 dB
                balance 0.00
        Buffer Latency: 500000 usec
        Sink Latency: 0 usec
        Resample method: speex-fixed-1
        Properties:
                media.name = "ALSA Playback"
                application.name = "ALSA plug-in [aplay]"
                native-protocol.peer = "UNIX socket client"
                native-protocol.version = "29"
                application.process.id = "1322"
                application.process.user = "pi"
                application.process.host = "raspberrypi"
                application.process.binary = "aplay"
                application.language = "en_GB.UTF-8"
                application.process.machine_id = "aa0a4892531c42499224a4d1b2606e1c"
                application.process.session_id = "c5"
                module-stream-restore.id = "sink-input-by-application-name:ALSA plug-in [aplay]"
pi@raspberrypi:~/env/bin $

I'm assuming I must be doing something wrong, however I cant for the life of me figure out what it is!

MyNameWasUnavailable, 2017-09-12:
ok, so I just rebooted the pi, and tried again without pulseaudio, and it worked!

I did receive the following error:

ALSA lib pcm.c:7843:(snd_pcm_recover) overrun occurred

However it did work correctly.

Looks like I am on the right track

MyNameWasUnavailable, 2017-09-13:
Just a final follow up comment before I close this issue.

I was able to successfully install and configure this hat using Stretch, the files / instructions from the aiyprojects-raspbian project, and the drivers and asound.conf from this git.

It worked more or less first go after fiddling for a while with the sound levels etc.  I am confident that the issue now is not with the hardware or drivers as such, simply a configuration issue on our parts, which it would appear in the official project, they have worked around.

I was also able to configure the button as a shutdown button, which also works.

ONly thing left is to integrate the LED's, which is the next thing I will be looking at.

closed 2017-09-13 by MyNameWasUnavailable

gangaselva, 2017-10-18:
I have followed the Google Assistant example. Exevrything works, but  Google Assistant voice is very quiet through the headphone jack and I can barely hear it (headphone is at full on ALSA mixer).  I have tried changing different mixer settings, but can't fix it. anyone know how to fix this? Seeed team please give me resolution to fix this issue. Thanks in advance

xiongyihui, 2017-10-18:
You can try to save https://github.com/respeaker/seeed-voicecard/blob/master/asound_2mic.conf as `/etc/asound.conf`

gangaselva, 2017-10-18:
I tried, but still same. any idea?
ThanksGanga

xiongyihui, 2017-10-18:
Is the sound OK when playing a wav or mp3 file?

gangaselva, 2017-10-18:
Sound was really good, when I play wav or mp3. I did, speaker-test -t wav test. It worked without any issues.

xiongyihui, 2017-10-19:
It seems a problem between alsa and google assistant.
Maybe have a try to use pulseaudio which has a util named pavucontrol to check the output sound level and amplify the level more than 100%. It will looks like:

[image]

gangaselva, 2017-10-19:
Thanks for your support. 
I am running headless mode. How do I increase volume to 100%.

xiongyihui, 2017-10-20:
You can try https://github.com/GeorgeFilipkin/pulsemixer

gangaselva, 2017-10-20:
I will try and let you know, if this resolve my issue. Did you guys figure it out actual root cause? It is so painful to try all sorts of options.

gangaselva, 2017-10-21:
sorry, still same. 
please give me clear direction to resolve the issue.

xiongyihui, 2017-10-21:
The google assistant demo (hotword version) has issues with ALSA on Raspberry Pi. The gRPC version works well.
Using PulseAudio will solve the issue. You should try the latest raspbian with pixel desktop and setup PulseAudio, and then use pavucontrol or audacity to make sure PulseAudio is working.
If PulseAudio works, `pulse` will be the default ALSA device. You can run `arecord -v -f cd s.wav` to check the default ALSA device. Then try the hotword version google assistant.

gangaselva, 2017-10-25:
Hi,
I resolved the issue. Its work very well. 
Thanks for your support.