respeaker/seeed-voicecard #9: Can this driver work on Ubuntu mate 16.04 ?

zhtsky, 2017-09-23:
I have installed this driver on raspberry (Ubuntu mate 16.04LTS), but I can't use it.

Pillar1989, 2017-09-23:
Can you show me your /boot/config.txt

zhtsky, 2017-09-24:
@Pillar1989 
[config.txt](https://github.com/respeaker/seeed-voicecard/files/1326973/config.txt)

Pillar1989, 2017-09-24:
It seems everything is fine!

zhtsky, 2017-09-24:
@Pillar1989  
I think so, too！ But  when I use cmd "aplay -l", I can't see seeed-2mic-voicecard.

Pillar1989, 2017-09-24:
do you insert the hat to your pi ? then reboot the board

zhtsky, 2017-09-24:
@Pillar1989 
Yes, I have installed this driver on raspbian-jessie in the same way, it works fine.
But on the ubuntu mate, it doesn't work.

Pillar1989, 2017-09-24:
#

Pillar1989, 2017-09-24:
I'll test it on mate tomorrow

Pillar1989, 2017-09-25:
I'm so sorry , Ubuntu  mate  used the old rasperry pi linux kernel(4.4) . We only support linux 4.9 kernel.

zhtsky, 2017-09-25:
@Pillar1989 
Ok, I know, thank you very much!

closed 2017-09-25 by zhtsky