respeaker/seeed-voicecard #14: Wake word not working

sumczak, 2017-11-04:
Hi, I have been trying to accomplish Alexa project for about a week now, trying it on Raspberry Pi Zero W, and Raspberry Pi 3, without a real success. I followed your manual, disabled onboard sound card, and everything work except wake word. I can start service by clicking button on java app. Sound otherwise works correctly. I have uploaded my alsa setting on: http://www.alsa-project.org/db/?f=7fd636fe60521a6709d83356a5c03fa401a7ebcb ,and this is terminal output when I try to use alexa avs wakeword terminal shows:

INFO:main: Starting Wake Word Agent
INFO:WakeWordAgent: State set to IDLE(2)
INFO:Initializing Sensory library | library name: TrulyHandsfree | library version: 5.0.0-beta.10.2 | model file: ../ext/resources/spot-alexa-rpi.snsr
WARNING:Library expires on: License expires on 28 Feb 2018 00:00:00 GMT
ALSA lib pcm_hw.c:1713:(_snd_pcm_hw_open) Invalid value for card
INFO:SensoryWakeWordEngine: mainLoop thread started
ERROR:An error happened in the mainLoop of SensoryWakeWord snsrRun(): task: Input channel audio-pcm: ALSA error: No such file or directory
INFO:SensoryWakeWordEngine: mainLoop thread ended
INFO:WakeWordIPCSocket::mainLoop thread started
INFO:WakeWordIPCSocket: init socket on port:5123
INFO:===> Connected to AVS client <===
INFO:WakeWordAgent: thread started

, and there is no response when I say "Alaxa". After pressing "java button" :

INFO:main: Starting Wake Word Agent
INFO:WakeWordAgent: State set to IDLE(2)
INFO:Initializing Sensory library | library name: TrulyHandsfree | library version: 5.0.0-beta.10.2 | model file: ../ext/resources/spot-alexa-rpi.snsr
WARNING:Library expires on: License expires on 28 Feb 2018 00:00:00 GMT
ALSA lib pcm_hw.c:1713:(_snd_pcm_hw_open) Invalid value for card
INFO:SensoryWakeWordEngine: mainLoop thread started
ERROR:An error happened in the mainLoop of SensoryWakeWord snsrRun(): task: Input channel audio-pcm: ALSA error: No such file or directory
INFO:SensoryWakeWordEngine: mainLoop thread ended
INFO:WakeWordIPCSocket::mainLoop thread started
INFO:WakeWordIPCSocket: init socket on port:5123
INFO:===> Connected to AVS client <===
INFO:WakeWordAgent: thread started
INFO:WakeWordAgent: IPC Command received:3
INFO:WakeWordAgent: State set to WAKE_WORD_PAUSE_REQUESTED(5)
INFO:SensoryWakeWordEngine: handling pause
INFO: *** THREAD JOINING: Sensory ***
INFO:WakeWordAgent: State set to WAKE_WORD_PAUSED(6)
INFO:WakeWordAgent: IPC Command received:4
INFO:WakeWordAgent: State set to WAKE_WORD_RESUME_REQUESTED(7)
INFO:SensoryWakeWordEngine: handling resume
ALSA lib pcm_hw.c:1713:(_snd_pcm_hw_open) Invalid value for card
INFO:WakeWordAgent: State set to IDLE(2)
INFO:SensoryWakeWordEngine: mainLoop thread started
ERROR:An error happened in the mainLoop of SensoryWakeWord snsrRun(): task: Input channel audio-pcm: ALSA error: No such file or directory
INFO:SensoryWakeWordEngine: mainLoop thread ended

Other problem, when I playing any radio stream or news, I can`t stop it only pause it.
I can see problem with ALSA, but I don`t really know how to solve it
Please can you help me with this.

Pillar1989, 2017-11-05:
which card? 2mic or 4mic?

xiongyihui, 2017-11-06:
You can try the python version alexa first. See https://github.com/respeaker/avs

aidex-hackability, 2017-11-10:
I had some issue with the sensory engine also, but the KITT_AI one works fine on my RPi3 + PiHAT 2 mic

sumczak, 2017-11-12:
It's a 2mic hat

sumczak, 2017-11-12:
I have manage to make it work with KITT_AI by changing asoundrc settings to default settings from: https://www.alsa-project.org/main/index.php/Asoundrc   but it fails when I'm asking Alexa to stop/cancel stream like news or radio.

sumczak, 2017-11-12:
Will try python version

KillingJacky, 2017-11-13:
`but it fails when I'm asking Alexa to stop/cancel stream like news or radio`
sounds like the echo cancellation technology should be involved. Can you wake Alexa when she's not streaming anything?

ppirrip, 2017-11-13:
I had similar issue with @KillingJacky, but with the Java client. Once I start play a  radio station, the client stop receiving voice commands. Everything works as expected when is not streaming audio. The python version I can't even get it steam radio - but everything else works

Pillar1989, 2017-11-14:
@ppirrip Could you share your code ? We work fine with python code.

ppirrip, 2017-11-14:
@Pillar1989 I am using the code from here:
https://github.com/respeaker/mic_hat.git

The only change I have to make is in the mic_hat/alexa.py line 22, which I have to change it to:
```
kws = KWS(model='alexa') # remove sensitivity=0.8
```
Python complained about an extra parameter. 
My problem is this setup won't play steam audio/radio, but the Java sample code does on the same HW. 

Should I run this python code in Python 2.7 or Python 3.5/3.6?

Pillar1989, 2017-11-15:
Yes, you can run this code in python 2.7 and Python 3.5/3.6. 
About  sensitivity , Please refer to 
https://github.com/voice-engine/voice-engine/blob/master/voice_engine/kws_snowboy.py#L20

ppirrip, 2017-11-15:
Seems like our voice-engine is at different revision.

Here is what I have from pip install voice-engine: (~/.local/lib/python2.7/site-packages/voice_engine)

class KWS(Element):
    def __init__(self, model='snowboy'):
        super(KWS, self).__init__()
[...]
Full file attached. Do you recommend me to use the github revision or the one install from pip? 

[kws_snowboy.py.zip](https://github.com/respeaker/seeed-voicecard/files/1475108/kws_snowboy.py.zip)

Pillar1989, 2017-11-23:
Yes,
Did you solve this issue?

ppirrip, 2017-11-23:
Hi, thanks for the follow up. I updated the voice engine and I can see the added sensitivity field. However, I still having problem playing a tunein radio from the Pi -- see brief log below. This features works on the Java sample app, as well as on the tunein radio's web browser. Is there any other way I can test the AudioPlayer directive from the voice engine?

thanks again for the follow up.

DEBUG:avs.alexa:{u'directive': {u'header': {u'namespace': u'TemplateRuntime', u'name': u'RenderPlayerInfo', u'messageId': u'43f56cd1-7ab3-4573-8339-e5d729dda292'}, u'payload': {u'content': {u'art': {u'sources': [{u'url': u'https://cdn-radiotime-logos.tunein.com/s54346t.png', u'size': u'x-small'}, {u'url': u'https://cdn-radiotime-logos.tunein.com/s54346t.png', u'size': u'small'}, {u'url': u'https://cdn-radiotime-logos.tunein.com/s54346q.png', u'size': u'medium'}, {u'url': u'https://cdn-radiotime-logos.tunein.com/s54346d.png', u'size': u'large'}, {u'url': u'https://cdn-radiotime-logos.tunein.com/s54346g.png', u'size': u'x-large'}]}, u'title': u'900', u'titleSubtext2': u'St. Michael, Barbados', u'titleSubtext1': u'CBC Radio', u'provider': {u'logo': {u'sources': [{u'url': u'https://s3.amazonaws.com/music-provider-logos/TuneIn.svg'}]}, u'name': u'TuneIn Live Radio'}, u'mediaLengthInMilliseconds': 0}, u'controls': [{u'type': u'BUTTON', u'selected': False, u'enabled': True, u'name': u'PLAY_PAUSE'}], u'audioItemId': u'amzn1.as-ct.v1.Dee-Domain-Music#ACRI#url#ACRI#bdcf253a-3062-46e4-ad33-30505c4226f9:1'}}}
DEBUG:avs.alexa:{
    "header": {
        "namespace": "TemplateRuntime", 
        "name": "RenderPlayerInfo", 
        "messageId": "43f56cd1-7ab3-4573-8339-e5d729dda292"
    }, 
    "payload": {
        "content": {
            "art": {
                "sources": [
                    {
                        "url": "https://cdn-radiotime-logos.tunein.com/s54346t.png", 
                        "size": "x-small"
                    }, 
                    {
                        "url": "https://cdn-radiotime-logos.tunein.com/s54346t.png", 
                        "size": "small"
                    }, 
                    {
                        "url": "https://cdn-radiotime-logos.tunein.com/s54346q.png", 
                        "size": "medium"
                    }, 
                    {
                        "url": "https://cdn-radiotime-logos.tunein.com/s54346d.png", 
                        "size": "large"
                    }, 
                    {
                        "url": "https://cdn-radiotime-logos.tunein.com/s54346g.png", 
                        "size": "x-large"
                    }
                ]
            }, 
            "title": "900", 
            "titleSubtext2": "St. Michael, Barbados", 
            "titleSubtext1": "CBC Radio", 
            "provider": {
                "logo": {
                    "sources": [
                        {
                            "url": "https://s3.amazonaws.com/music-provider-logos/TuneIn.svg"
                        }
                    ]
                }, 
                "name": "TuneIn Live Radio"
            }, 
            "mediaLengthInMilliseconds": 0
        }, 
        "controls": [
            {
                "type": "BUTTON", 
                "selected": false, 
                "enabled": true, 
                "name": "PLAY_PAUSE"
            }
        ], 
        "audioItemId": "amzn1.as-ct.v1.Dee-Domain-Music#ACRI#url#ACRI#bdcf253a-3062-46e4-ad33-30505c4226f9:1"
    }
}
INFO:avs.alexa:TemplateRuntime is not implemented yet

xiongyihui, 2017-11-24:
The AudioPlayer interface uses gstreamer to play audio from web. Maybe gstreamer doesn't support  tunein radio. You can try:
```
gst-launch-1.0 playbin uri={tunein radio audio url}
```
It tests if the audio can be played by gstreamer.  If gst-launch-1.0 is not found. run `sudo apt install gstreamer1.0-tools`

ppirrip, 2017-11-24:
I have a feeling that gst doesn't support tunein radio, as I play the uri in VLC is works. I have updated and installed all the gst plugins (the good, the bad and the ugly) and still no luck.

Based on the log for the directive AudioPlayer.Play, the uri from tunein is: 
https://opml.radiotime.com/Tune.ashx?serial=AHYZVBNCNMNXCMCIBJQ4ROIEAUUQ&formats=aac,mp3&id=s89259&partnerId=4JqugguZ

And the message I have from running gst on the console is:

pi@raspberrypi:~/Workspace/mic_hat $ gst-launch-1.0 playbin uri="https://opml.radiotime.com/Tune.ashx?serial=AHYZVBNCNMNXCMCIBJQ4ROIEAUUQ&formats=aac,mp3&id=s89259&partnerId=4JqugguZ"
Setting pipeline to PAUSED ...
Pipeline is PREROLLING ...
Missing element: text/uri-list decoder
WARNING: from element /GstPlayBin:playbin0/GstURIDecodeBin:uridecodebin0: No decoder available for type 'text/uri-list'.
Additional debug info:
gsturidecodebin.c(938): unknown_type_cb (): /GstPlayBin:playbin0/GstURIDecodeBin:uridecodebin0
ERROR: from element /GstPlayBin:playbin0/GstURIDecodeBin:uridecodebin0: Your GStreamer installation is missing a plug-in.
Additional debug info:
gsturidecodebin.c(1005): no_more_pads_full (): /GstPlayBin:playbin0/GstURIDecodeBin:uridecodebin0:
no suitable plugins found:
gstdecodebin2.c(4592): gst_decode_bin_expose (): /GstPlayBin:playbin0/GstURIDecodeBin:uridecodebin0/GstDecodeBin:decodebin0:
no suitable plugins found:
Missing decoder: text/uri-list (text/uri-list)

ERROR: pipeline doesn't want to preroll.
Setting pipeline to NULL ...
Freeing pipeline ...

-----------------
DEBUG:avs.alexa:{u'directive': {u'header': {u'dialogRequestId': u'c3da727ae38f4b3ab3abbc56b385bf67', u'namespace': u'AudioPlayer', u'name': u'Play', u'messageId': u'3d599876-5d3b-4009-b118-dd6b96be1de1'}, u'payload': {u'playBehavior': u'REPLACE_ALL', u'audioItem': {u'stream': {u'url': u'https://opml.radiotime.com/Tune.ashx?serial=AHYZVBNCNMNXCMCIBJQ4ROIEAUUQ&formats=aac,mp3&id=s89259&partnerId=4JqugguZ', u'progressReport': {u'progressReportIntervalInMilliseconds': 900000}, u'token': u'XXX', u'expiryTime': u'2017-11-24T14:38:59+0000', u'offsetInMilliseconds': 0}, u'audioItemId': u'amzn1.as-ct.v1.Dee-Domain-Music#ACRI#url#ACRI#9268482b-2fa9-4956-9f17-3c4f4ab9619e:1'}}}}

closed 2018-01-03 by Pillar1989

xiongyihui, 2018-01-04:
@ppirrip The avs library got an update which should support TuneIn stream. You can test it now:

```
pip install -U avs
```