respeaker/seeed-voicecard #16: Audio capture during audio playback

aidex-hackability, 2017-11-10:
I am running an AVS Java sample app with RPi3 + PiHAT 2mic on Raspian. I have the java client playing a radio, then I try to use the "listen" button to stop the radio, which works fine on a real Amazon Echo. At first I got a Java Exception "can't obtain source audio line error". After that I disabled the mp3 playback function in recordingStarted() in App.js, now the java client app just ignores my voice comment. 

My question is: Is the PiHAT 2 mic (HW or SW) capable of recording while playing back an audio?

Pillar1989, 2017-11-14:
@aidex-hackability Yes, Please test 
```sh
arecord -f cd -Dhw:1 | aplay -Dhw:
```

ppirrip, 2017-11-14:
Thanks @Pillar1989 . That test always worked (my -Dhw: is 0). I have this issue on both the Java sample client and the python avs, but the actual Amazon Echo doesn't seem to have this issue. Anyway, thanks. I will look into the Java/Python sample code more then.

closed 2017-11-23 by Pillar1989