respeaker/seeed-voicecard #17: Unable to get Status Code 204

tarunl, 2017-12-07:
I followed the sets of commands given in following article
http://wiki.seeed.cc/ReSpeaker_4_Mic_Array_for_Raspberry_Pi/

But after being done with the commands. No Status code 204 is received after running
"python ns_kws_doa_alexa_with_light.py"

Please help me to find the issue.

Following is the log file.........

ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.front
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround40
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround41
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround50
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround51
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround71
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4996:(snd_config_expand) Args evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4996:(snd_config_expand) Args evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa
ALSA lib pcm_dmix.c:990:(snd_pcm_dmix_open) The dmix plugin supports only playback stream
ALSA lib pcm_dmix.c:990:(snd_pcm_dmix_open) The dmix plugin supports only playback stream
ALSA lib pcm_ac108.c:469:(_snd_pcm_ac108_open) a108 is only for capture
connect(2) call to /tmp/jack-1000/default/jack_0 failed (err=No such file or directory)
attempt to connect to server failed
INFO:/home/pi/env/local/lib/python2.7/site-packages/voice_engine-0.0.3-py2.7.egg/voice_engine/source.pyc:0:bcm2835 ALSA: - (hw:0,0) with 0 input channels
INFO:/home/pi/env/local/lib/python2.7/site-packages/voice_engine-0.0.3-py2.7.egg/voice_engine/source.pyc:1:bcm2835 ALSA: IEC958/HDMI (hw:0,1) with 0 input channels
INFO:/home/pi/env/local/lib/python2.7/site-packages/voice_engine-0.0.3-py2.7.egg/voice_engine/source.pyc:2:seeed-4mic-voicecard: - (hw:1,0) with 2 input channels
INFO:/home/pi/env/local/lib/python2.7/site-packages/voice_engine-0.0.3-py2.7.egg/voice_engine/source.pyc:3:sysdefault with 0 input channels
INFO:/home/pi/env/local/lib/python2.7/site-packages/voice_engine-0.0.3-py2.7.egg/voice_engine/source.pyc:4:playback with 0 input channels
INFO:/home/pi/env/local/lib/python2.7/site-packages/voice_engine-0.0.3-py2.7.egg/voice_engine/source.pyc:5:dmixed with 0 input channels
INFO:/home/pi/env/local/lib/python2.7/site-packages/voice_engine-0.0.3-py2.7.egg/voice_engine/source.pyc:6:ac108 with 4 input channels
INFO:/home/pi/env/local/lib/python2.7/site-packages/voice_engine-0.0.3-py2.7.egg/voice_engine/source.pyc:Use ac108
INFO:avs.alexa:Refreshing access_token
DEBUG:urllib3.connectionpool:Starting new HTTPS connection (1): api.amazon.com
DEBUG:urllib3.connectionpool:https://api.amazon.com:443 "POST /auth/o2/token HTTP/1.1" 200 977
DEBUG:avs.alexa:{
"product_id": "ReSpeaker",
"access_token": "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx",
"refresh_url": "https://api.amazon.com/auth/o2/token",
"expiry": "Thu Dec 07 12:49:45 2017",
"api": "v20160207",
"client_id": "amzn1.application-oa2-client.xxxxxxxxxxxxxxxxxxxxxxxxxx",
"host_url": "avs-alexa-na.amazon.com",
"client_secret": "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx",
"refresh_token": "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx"
[... 89 lines cut ...]
DEBUG:hpack.hpack:Decoded 63, consumed 2 bytes
DEBUG:hpack.hpack:Decoded 44, consumed 1 bytes
DEBUG:hpack.hpack:Decoded ('x-amzn-requestid', '06c728fffe3dcd94-00004c6d-0000c350-3d16aba23a82e67b-6d39ad0b-3'), total consumed 47 bytes, indexed True
DEBUG:hpack.hpack:Decoded 31, consumed 1 bytes
DEBUG:hpack.hpack:Decoded 68, consumed 1 bytes
DEBUG:hpack.hpack:Decoded ('content-type', 'multipart/related;boundary=------abcde123;start=metadata.1512627619030;type="application/json"'), total consumed 70 bytes, indexed True
INFO:hyper.http20.connection:Received unhandled event <PriorityUpdated stream_id:3, weight:16, depends_on:0, exclusive:False>
INFO:avs.alexa:status code: 200
INFO:avs.alexa:Found 176 bytes of None application/json, first_payload_block=True
INFO:avs.alexa:Finished downloading JSON
DEBUG:avs.alexa:{u'directive': {u'header': {u'namespace': u'System', u'name': u'SetEndpoint', u'messageId': u'cc59261c-59ef-4629-8650-2816e28a37f7'}, u'payload': {u'endpoint': u'https://avs-alexa-eu.amazon.com'}}}
DEBUG:avs.alexa:{
"header": {
"namespace": "System",
"name": "SetEndpoint",
"messageId": "cc59261c-59ef-4629-8650-2816e28a37f7"
},
"payload": {
"endpoint": "https://avs-alexa-eu.amazon.com"
}
}
DEBUG:avs.alexa:ping at Thu Dec 07 11:53:48 2017
INFO:hyper.http20.connection:Received unhandled event 
DEBUG:avs.alexa:ping at Thu Dec 07 11:57:48 2017
DEBUG:avs.alexa:ping at Thu Dec 07 12:01:49 2017

Pillar1989, 2017-12-18:
Hi, Did  you test sound card ? you need confirm the sound card works fine first.

tarunl, 2017-12-18:
Can u suggest me a quick way to do the same ?
Though I was able to record and listen through audacity.

Pillar1989, 2017-12-18:
audacity  is fine.

closed 2018-01-03 by Pillar1989