respeaker/seeed-voicecard #19: 2mic not listed in aplay

noemu, 2017-12-12:
After driver installation and reboot on a vanilla Raspbian Stretch lite, i can't find the soundcard.

The RGB LED's are working over the SPI. However the aplay list only shows the bcm2835 default device.

Device: Raspberry Pi Zero W
Soundcard: ReSpeaker 2-mics Pi HAT v1.0
Linux raspberrypi 4.9.59+ #1047 Sun Oct 29 11:47:10 GMT 2017 armv6l

Install log:
[https://pastebin.com/3DLwkaU7](https://pastebin.com/3DLwkaU7)

Pillar1989, 2017-12-13:
Can you reboot the board ? Then send  `dmesg` log to me .

noemu, 2017-12-13:
Here the dmesg log
https://pastebin.com/VWueWjb0

noemu, 2017-12-13:
fixed it:
`if [ ! -f "/boot/overlays/seeed-4mic-voicecard.dtbo" ] && [ ! -f "/lib/modules/${uname_r}/kernel/sound/soc/codecs/snd-soc-ac108.ko" ] ; then`
failed somehow, even after unistall.

closed 2017-12-13 by noemu

beosro, 2018-01-13:
Well, I have the same problem ...
Any solution?
I don't understand noemu last comment.
Thanks in advance.

calinichta, 2018-01-19:
Hi beosro,

noemu took a look into the /seed-voicecard/install.sh file and fixed the problem by running the content of the if clause (start each line with sudo ):
  cp seeed-2mic-voicecard.dtbo /boot/overlays
  cp seeed-4mic-voicecard.dtbo /boot/overlays
  install -D ac108_plugin/libasound_module_pcm_ac108.so   /usr/lib/arm-linux-gn$

Unfortunately I'm not able to get this sound card work under Raspbian Stretch, yet - but I try hard!

If I find a solution, I'l post it hear.

giano, 2018-01-20:
Same here. Downgraded to Jessie 4.9.35+ kernel too, plain install on RPI zero w. Install goes well but device is not listed in aplay. I can see the boot modules in place.

giano, 2018-01-20:
BTW. Switched to Dev Branch but no joy there too.

noemu, 2018-01-20:
try to run the uninstall with sudo and for each card (2mic and 4mic) and after a reboot run the original install.sh once with sudo and post the error.

giano, 2018-01-20:
No 'blocking error' at all, it seems. After some yada about satisfied
dependencies here is the output:

Error! There are no instances of module: seeed-voicecard
0.2 located in the DKMS tree.

Creating symlink /var/lib/dkms/seeed-voicecard/0.2/source ->
/usr/src/seeed-voicecard-0.2

DKMS: add completed.

Kernel preparation unnecessary for this kernel. Skipping...

Building module:
cleaning build area......
make KERNELRELEASE=4.9.35+ -C /lib/modules/4.9.35+/build
M=/var/lib/dkms/seeed-voicecard/0.2/build.....................
cleaning build area.....

DKMS: build completed.

Kernel preparation unnecessary for this kernel. Skipping...

Building module:
cleaning build area.....
make KERNELRELEASE=4.9.35-v7+ -C /lib/modules/4.9.35-v7+/build
M=/var/lib/dkms/seeed-voicecard/0.2/build.....................
cleaning build area.....

DKMS: build completed.

snd-soc-wm8960:
Running module version sanity check.
- Original module
- No original module exists within this kernel
- Installation
- Installing to /lib/modules/4.9.35+/kernel/sound/soc/codecs/

snd-soc-ac108.ko:
Running module version sanity check.
- Original module
- No original module exists within this kernel
- Installation
- Installing to /lib/modules/4.9.35+/kernel/sound/soc/codecs/

snd-soc-simple-card.ko:
Running module version sanity check.

Good news! Module version for snd-soc-simple-card.ko
exactly matches what is already found in kernel 4.9.35+.
DKMS will not replace this module.
You may override by specifying --force.

depmod.....

DKMS: install completed.

snd-soc-wm8960:
Running module version sanity check.

Good news! Module version for snd-soc-wm8960.ko
exactly matches what is already found in kernel 4.9.35-v7+.
DKMS will not replace this module.
You may override by specifying --force.

snd-soc-ac108.ko:
Running module version sanity check.

Good news! Module version for snd-soc-ac108.ko
exactly matches what is already found in kernel 4.9.35-v7+.
DKMS will not replace this module.
You may override by specifying --force.

snd-soc-simple-card.ko:
Running module version sanity check.

Good news! Module version for snd-soc-simple-card.ko
exactly matches what is already found in kernel 4.9.35-v7+.
DKMS will not replace this module.
You may override by specifying --force.

depmod.....

DKMS: install completed.
cp wm8960_asound.state /var/lib/alsa/asound.state
------------------------------------------------------
Please reboot your raspberry pi to apply all settings
Enjoy!
------------------------------------------------------

But doing aplay -l no seeed sound card is listed

giano, 2018-01-20:
aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
Subdevices: 8/8
Subdevice #0: subdevice #0
Subdevice #1: subdevice #1
Subdevice #2: subdevice #2
Subdevice #3: subdevice #3
Subdevice #4: subdevice #4
Subdevice #5: subdevice #5
Subdevice #6: subdevice #6
Subdevice #7: subdevice #7
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
Subdevices: 1/1
Subdevice #0: subdevice #0

giano, 2018-01-22:
I can see in kern.log those lines:

Jan 20 19:13:15 jinni kernel: [    5.373739] wm8960 1-001a: Failed to issue reset
Jan 20 19:13:15 jinni kernel: [    5.373910] wm8960: probe of 1-001a failed with error -121

and 

Jan 20 19:36:33 jinni kernel: [    3.575627] snd_soc_ac108: loading out-of-tree module taints kernel.

I don't care about tainted kernel, but the first 2 lines seems an issue. So the module is loaded. Is the hardware malfunctioning?

SimonGurney, 2018-02-05:
Pulled my hair out A LOT then this fixed it:
sudo ./uninstall 2mic
reboot
sudo ./uninstall 4mic
reboot
sudo ./install 2mic
reboot

jabulonboaz, 2018-02-07:
@SimonGurney how did you do it please? 
I get "command not found"
pi@raspberrypi:~ $ sudo ./uninstall 2mic
sudo: ./uninstall: command not found
pi@raspberrypi:~ $ cd seeed-voicecard
pi@raspberrypi:~/seeed-voicecard $ sudo ./uninstall 2mic
sudo: ./uninstall: command not found

SimonGurney, 2018-02-07:
Sorry, try ./uninstall.sh

jabulonboaz, 2018-02-07:
Its ok Simon i figured it out but no luck yet. See my other post.