respeaker/seeed-voicecard #20: nothing recorded but errors in dmesg

hexxter, 2017-12-12:
[    3.179662]  i2c_id number :0
[    3.179674]  ac108  codec_index :0
[    3.179679]  ac108  I2S data protocol type :1
[    3.189993] ac108_write error->[REG-0x00,val-0x12]
[    3.196053] ac108_set_sysclk  :24000000
[    3.228844] asoc-simple-card soc:sound: ac108-codec0 <-> 3f203000.i2s mapping ok
...
[   28.566916] ac108_write error->[REG-0x33,val-0x0f]
[   28.568356] ac108_read error1
[   28.569789] ac108_read error1
[   28.571239] ac108_write error->[REG-0x35,val-0x33]
[   28.572681] ac108_read error1
[   28.574113] ac108_write error->[REG-0x60,val-0x04]
[   28.575545] ac108_write error->[REG-0x66,val-0x0f]
[   28.575551] AC108 PLL freq_in match:24000000, freq_out:22579200

[   28.576985] ac108_read error1
[   28.578418] ac108_write error->[REG-0x14,val-0xa9]
[   28.579849] ac108_read error1
[   28.581291] ac108_write error->[REG-0x13,val-0x40]
[   28.582729] ac108_read error1
[   28.584162] ac108_write error->[REG-0x12,val-0xa5]
[   28.585593] ac108_read error1
[   28.587024] ac108_write error->[REG-0x11,val-0x90]
[   28.588454] ac108_read error1
[   28.589884] ac108_write error->[REG-0x18,val-0xa7]
[   28.591331] ac108_read error1
[   28.592776] ac108_write error->[REG-0x10,val-0xaf]
[   28.594211] ac108_read error1
[   28.595642] ac108_write error->[REG-0x20,val-0x8f]
[   28.597073] ac108_write error->[REG-0x21,val-0x93]
....

if i record on the ac108 device (arecord/audacity) more from this errors will appened.

Pillar1989, 2017-12-13:
Hi , Do you plug-in the 4mic hat ?  Can you show me with `i2cdetect -y -r 1` log

hexxter, 2017-12-13:
0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- UU -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --

hexxter, 2017-12-13:
hardware is RPi3 Kernel 4.9.35-v7+ (raspbian)

Pillar1989, 2017-12-14:
It's seems your 4mic hat was broken.

Pillar1989, 2017-12-14:
Do you have other one ? Maybe you can try it again

hexxter, 2017-12-14:
yes i have a secound one but the same result.

hexxter, 2017-12-14:
i have resolder the irq pin and now it works more, but all channels if i record (audacity) is silents. I found this in the dmesg:

[   30.341142] AC108 PLL freq_in match:24000000, freq_out:22579200

[   30.423975] AC108 PLL freq_in match:24000000, freq_out:22579200

[   75.586491] AC108 PLL freq_in match:24000000, freq_out:24576000

[   75.596615] AC108 PLL freq_in match:24000000, freq_out:24576000

[   75.604638] AC108 PLL freq_in match:24000000, freq_out:24576000

[   75.611454] AC108 PLL freq_in match:24000000, freq_out:22579200

[   75.619383] AC108 PLL freq_in match:24000000, freq_out:24576000

[   75.627772] AC108 PLL freq_in match:24000000, freq_out:24576000

[   75.850985] AC108 PLL freq_in match:24000000, freq_out:24576000

[   75.859733] AC108 PLL freq_in match:24000000, freq_out:24576000

[   75.867803] AC108 PLL freq_in match:24000000, freq_out:24576000

[   75.874872] AC108 PLL freq_in match:24000000, freq_out:22579200

[   75.882711] AC108 PLL freq_in match:24000000, freq_out:24576000

[   75.891086] AC108 PLL freq_in match:24000000, freq_out:24576000

[   75.903019] AC108 PLL freq_in match:24000000, freq_out:24576000

[   76.665900] AC108 PLL freq_in match:24000000, freq_out:22579200

[   76.674597] AC108 PLL freq_in match:24000000, freq_out:22579200

[   76.681749] AC108 PLL freq_in match:24000000, freq_out:22579200

[   98.812820] AC108 PLL freq_in match:24000000, freq_out:24576000

[   98.829972] AC108 PLL freq_in match:24000000, freq_out:24576000

[   98.842422] AC108 PLL freq_in match:24000000, freq_out:24576000

[   98.849272] AC108 PLL freq_in match:24000000, freq_out:22579200

[   98.857313] AC108 PLL freq_in match:24000000, freq_out:24576000

[   98.865652] AC108 PLL freq_in match:24000000, freq_out:24576000

[   99.815352] AC108 PLL freq_in match:24000000, freq_out:24576000

[   99.824354] AC108 PLL freq_in match:24000000, freq_out:24576000

[   99.832041] AC108 PLL freq_in match:24000000, freq_out:24576000

[   99.839050] AC108 PLL freq_in match:24000000, freq_out:22579200

[   99.847025] AC108 PLL freq_in match:24000000, freq_out:24576000

[   99.855363] AC108 PLL freq_in match:24000000, freq_out:24576000

[   99.874664] AC108 PLL freq_in match:24000000, freq_out:24576000

[   99.883552] AC108 PLL freq_in match:24000000, freq_out:24576000

[   99.891302] AC108 PLL freq_in match:24000000, freq_out:24576000

[   99.898314] AC108 PLL freq_in match:24000000, freq_out:22579200

[   99.906298] AC108 PLL freq_in match:24000000, freq_out:24576000

[   99.914681] AC108 PLL freq_in match:24000000, freq_out:24576000

[   99.924175] AC108 PLL freq_in match:24000000, freq_out:24576000

[  104.032880] AC108 PLL freq_in match:24000000, freq_out:22579200

[  156.334478] AC108 PLL freq_in match:24000000, freq_out:22579200

[  207.385731] AC108 PLL freq_in match:24000000, freq_out:24576000

[  211.865453] AC108 PLL freq_in match:24000000, freq_out:22579200

hexxter, 2017-12-15:
more information:

[    3.093611]  i2c_id number :0
[    3.093623]  ac108  codec_index :0
[    3.093628]  ac108  I2S data protocol type :1
[    3.095494] ac108_write error->[REG-0x00,val-0x12]
[    3.148356] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)
[    3.191610] ac108_set_sysclk  :24000000
[    3.250199] asoc-simple-card soc:sound: ac108-codec0 <-> 3f203000.i2s mapping ok
[    3.500931] brcmfmac: F1 signature read @0x18000000=0x1541a9a6
[    3.507674] usbcore: registered new interface driver brcmfmac
[    3.729987] brcmfmac: Firmware version = wl0: Aug  7 2017 00:46:29 version 7.45.41.46 (r666254 CY) FWID 01-f8a78378
[    3.730897] brcmfmac: brcmf_c_preinit_dcmds: CLM version = API: 12.2 Data: 7.11.15 Compiler: 1.24.2 ClmImport: 1.24.1 Creation: 2014-05-26 10:53:55 Inc Data: 9.10.41 Inc Compiler: 1.29.4 Inc ClmImport: 1.36.3 Creation: 2017-08-07 00:37:47 
[    4.035530] EXT4-fs (mmcblk0p3): recovery complete
[    4.035554] EXT4-fs (mmcblk0p3): mounted filesystem with ordered data mode. Opts: (null)
[    4.216238] systemd-journald[129]: Received request to flush runtime journal from PID 1
[    6.315611] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[    6.315637] brcmfmac: power management disabled
[    6.518111] random: crng init done
[    6.531754] uart-pl011 3f201000.serial: no DMA platform data
[    7.211376] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[    8.727095] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0xCDE1
[   11.098206] Bluetooth: Core ver 2.22
[   11.098292] NET: Registered protocol family 31
[   11.098297] Bluetooth: HCI device and connection manager initialized
[   11.098318] Bluetooth: HCI socket layer initialized
[   11.098330] Bluetooth: L2CAP socket layer initialized
[   11.098368] Bluetooth: SCO socket layer initialized
[   11.108455] Bluetooth: HCI UART driver ver 2.3
[   11.108470] Bluetooth: HCI UART protocol H4 registered
[   11.108475] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   11.108647] Bluetooth: HCI UART protocol Broadcom registered
[   11.315158] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   11.315168] Bluetooth: BNEP filters: protocol multicast
[   11.315182] Bluetooth: BNEP socket layer initialized
[   19.429908] AC108 PLL freq_in match:24000000, freq_out:22579200

[   19.504118] AC108 PLL freq_in match:24000000, freq_out:22579200
.....

Pillar1989, 2017-12-18:
Hi, I think your 4mic hat was not installed correctly. Please do
```
sudo ./uninstall.sh 4mic
sudo ./uninstall.sh 2mic
```
then reboot 
```
sudo ./install.sh 4mic
```
reboot again.

If you still have the same issue. Please show me `alsamixer -c 0`

closed 2018-01-03 by Pillar1989