respeaker/seeed-voicecard #21: Recording not working

bshogeman, 2017-12-24:
arecord -Dac108 -f S32_LE -r 16000 -c 4 a.wav -d 10
arecord: main:788: audio open error: File or folder doesn't exist

The pixel ring is working with the example files. Only can't seem to get the mic's working on the ReSpeaker 4mic model.

What can I do to fix this?

Pillar1989, 2017-12-26:
Hi , Where do you want to store the a.wav ?  maybe you can try 
`arecord -Dac108 -f S32_LE -r 16000 -c 4  /tmp/a.wav -d 10`

bshogeman, 2017-12-26:
In the home folder. The writing of the file isn't the problem. If I skip the -Dac108 and -f S32_LE parameters the command (not specify the soundcard or just the command for the 2mic version) is working and writes a file. Only the file doesn't create anything but a long silence.

Pillar1989, 2018-01-03:
which pi image do you use ?

bshogeman, 2018-01-03:
Raspberry Pi 3 B

Pillar1989, 2018-01-15:
#25

closed 2018-01-15 by Pillar1989