respeaker/seeed-voicecard #23: 安装驱动后，树莓派没有了声音

1158114251, 2018-01-03:
我安装驱动后，树莓派没有了声音，即使我在面板下修改成3.5还是不行 raspi-config，只要我卸载了就没问题了，后来我定位是 /etc/asound 删除就可以了，我看安装脚本 是asound_4mic.conf 拷贝进去的，问下如何解决？

Pillar1989, 2018-01-03:
你用的是树莓派型号？镜像版本？

1158114251, 2018-01-04:
树莓派3b 镜像是2017年10月的，我看到安装脚本里面有升级内核，安装过程一直联网，内核也升级了。

xiongyihui, 2018-01-04:
可以尝试把文件里面的`555555`改成`777777`或其它值

```
# The IPC key of dmix or dsnoop plugin must be unique
# If 555555 or 666666 is used by other processes, use another one

pcm.!default {
    type asym
    playback.pcm "playback"
    capture.pcm "ac108"
}

pcm.playback {
    type plug
    slave.pcm "dmixed"
}

pcm.dmixed {
    type dmix
    slave.pcm "hw:0,0"
    ipc_key 777777
}

pcm.ac108 {
        type ac108
        slavepcm "hw:1,0"
        channels 4
}
```

xiongyihui, 2018-01-04:
或者不用`dmix`插件，把default设备中的`playback.pcm "playback"`改成`playback.pcm "hw:0,0"`

但没有dmix插件，不能同时播放多个音频，运行[avs](https://github.com/respeaker/avs)会有问题

1158114251, 2018-01-04:
好的，晚上我回去试试，感谢回复，还想问下，在用户层用alsa基于c语言编程录音，但是一直录不到音，用arecord可以，下面是我的录音程序，录音保存的pcm文件，我用Audacity打开，完全没有波形，也就没有声音
#include <alsa/asoundlib.h>  
int main() {  
long loops;  
int rc,i = 0;  
int size;  
FILE *fp ;
snd_pcm_t *handle;  
snd_pcm_hw_params_t *params;  
unsigned int val,val2;  
int dir;  
snd_pcm_uframes_t frames;  
char *buffer;  
if(  (fp =fopen("sound.wav","w")) < 0)
printf("open sound.wav fial\n");
/* Open PCM device for recording (capture). */  
rc = snd_pcm_open(&handle, "default",  
SND_PCM_STREAM_CAPTURE, 0);  
if (rc < 0) {  
fprintf(stderr,  "unable to open pcm device: %s/n",  snd_strerror(rc));  
exit(1);  
}  
/* Allocate a hardware parameters object. */  
snd_pcm_hw_params_alloca(&params);  
/* Fill it in with default values. */  
snd_pcm_hw_params_any(handle, params);  
/* Set the desired hardware parameters. */  
/* Interleaved mode */  
snd_pcm_hw_params_set_access(handle, params,  
SND_PCM_ACCESS_RW_INTERLEAVED);  
/* Signed 16-bit little-endian format */  
snd_pcm_hw_params_set_format(handle, params,  
SND_PCM_FORMAT_S16_LE);  
/* Two channels (stereo) */  
snd_pcm_hw_params_set_channels(handle, params, 2);  
/* 44100 bits/second sampling rate (CD quality) */  
val = 44100;  
snd_pcm_hw_params_set_rate_near(handle, params,  &val, &dir);  
/* Set period size to 32 frames. */  
frames = 32;  
snd_pcm_hw_params_set_period_size_near(handle,  params, &frames, &dir);  
/* Write the parameters to the driver */  
rc = snd_pcm_hw_params(handle, params);  
if (rc < 0) {  
fprintf(stderr,  "unable to set hw parameters: %s/n",  
snd_strerror(rc));  
exit(1);  
}  
/* Use a buffer large enough to hold one period */  
snd_pcm_hw_params_get_period_size(params,  &frames, &dir);  
size = frames * 4; /* 2 bytes/sample, 2 channels */  
printf("size = %d\n",size);
buffer = (char *) malloc(size);  
/* We want to loop for 5 seconds */  
snd_pcm_hw_params_get_period_time(params,  &val, &dir);  
loops = 10000000 / val;  
while (loops > 0) {  
loops--;  
rc = snd_pcm_readi(handle, buffer, frames); 
printf("%d\n",i++); 
if (rc == -EPIPE) {  
/* EPIPE means overrun */  
fprintf(stderr, "overrun occurred/n");  
snd_pcm_prepare(handle);  
} else if (rc < 0) {  
fprintf(stderr,  
"error from read: %s/n",  
snd_strerror(rc));  
} else if (rc != (int)frames) {  
fprintf(stderr, "short read, read %d frames/n", rc);  
}  
//rc = fwrite( buffer,1, size,fp);  
rc = write(1,buffer,size);
if (rc != size)  
fprintf(stderr,  "short write: wrote %d bytes/n", rc);  
else printf("fwrite buffer success\n");
}

1158114251, 2018-01-04:
采样率我已经改成16000，通道改为4通道 ，snd_pcm_open(&handle, "hw:1,0")，size = frames * 8; / 2 bytes/sample, 4channels */ 依旧录不到音，也不报错。

1158114251, 2018-01-05:
用audacity可以录到音，那个声音播放还是不行，安装驱动后，树莓派就没声音了，我今天下载了你们的镜像回去试试，录音我写的程序应该也没问题呀，就是录不到，用arecord和audacity都可以录到

1158114251, 2018-01-05:
好的，那我自己看看解决，打扰您了，十分感谢您的回复，上面那段代码在Ubuntu，x86平台是可以运行的，可能程序健壮性不行，所以树莓派没有成功

closed 2018-01-15 by Pillar1989