respeaker/seeed-voicecard #24: Respeaker 2-mic channel 1 has failed

dm1010111, 2018-01-12:
Following is the command execution and error:

$>google-assistant-demo --device_model_id respeaker
device_model_id: respeaker
device_id: FB29E975434B822C7AD45E3AADB1AFFF

ON_MUTED_CHANGED:
  {'is_muted': False}
ON_START_FINISHED
[1528:1558:ERROR:mic_failure_detector.cc(140)] [assistant] Channel 1 has failed. Powers -inf and -inf

Also test "arecord -f cd -Dhw:1 | aplay -Dhw:1"  is failed.  The mics could not pick up the voice.

dm1010111, 2018-01-12:
Please help to rootcause or resolve this issue. Thanks.

dm1010111, 2018-01-12:
More information:
OS is Linux raspberrypi 4.9.59-v7+ #1047 SMP Sun Oct 29 12:19:23 GMT 2017 armv7l GNU/Linux
dmesg: bcm2835-i2s 3f203000.i2s: ASoC: can't set 3f203000.i2s hw params: -16
Hope this can help.  By the way.  This system is working for 4-mic "Respeaker 4-Mic Array" board..

Pillar1989, 2018-01-15:
@dm1010111 Maybe you hardware device choice is wrong. if you show me  info about `aplay -l ` and `arecord -l`, I can help you

dm1010111, 2018-01-15:
Hi Pillar1989, Thnak you for your response. Originally I used Respeaket 4-mic with Raspberry Pi 3 and it works fine.  And then I replaced Respeaket 4mic with a Respeaker-2 mic.  The issue happened after I rebuild the driver.  I wnet back Repekaer 4-mic after I found the issue.  But this time, Respeaker 4mic dose not work even I rebuild the driver for Respeaker 4-mic.  The same problem as Respeaker 2-mic. Following is the aplay -l output and .asoundrc :     
pi@raspberrypi:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 7/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
---------------------asoundrc----------------------------------------------------------
pi@raspberrypi:~ $ cat .asoundrc 
pcm.!default {
  type asym
  capture.pcm "mic"
  playback.pcm "speaker"
}

pcm.mic {
  type plug
  slave {
    pcm "hw:0,0"
  }
}

pcm.speaker {
  type plug
  slave {
   pcm "hw:1,0"
  }
}

dm1010111, 2018-01-15:
I did "sudo ./uninstall.sh 2mic, sudo ./uninstall 4mic; reboot; sudo ./install 4mic; reboot".  But the issue is still there.   

dmesg:
[    4.008843]  i2c_id number :0
[    4.008857]  ac108  codec_index :0
[    4.008862]  ac108  I2S data protocol type :1
[    4.019158] ac108_write error->[REG-0x00,val-0x12]
[    4.020552] ac108_set_sysclk  :24000000
brcmfmac: Firmware version = wl0: Aug  7 2017 00:46:29 version 7.45.41.46 (r666254 CY) FWID 01-f8a78378
[    4.523656] brcmfmac: brcmf_c_preinit_dcmds: CLM version = API: 12.2 Data: 7.11.15 Compiler: 1.24.2 ClmImport: 1.24.1 Creation: 2014-05-26 10:53:55 Inc Data: 9.10.41 Inc Compiler: 1.29.4 Inc ClmImport: 1.36.3 Creation: 2017-08-07 00:37:4

...........................................................................................................
[   13.719301] AC108 PLL freq_in match:24000000, freq_out:22579200

[   13.814846] AC108 PLL freq_in match:24000000, freq_out:22579200

dm1010111, 2018-01-15:
pi@raspberrypi:~ $ arecord -l
**** List of CAPTURE Hardware Devices ****
card 1: seeed4micvoicec [seeed-4mic-voicecard], device 0: bcm2835-i2s-ac108-codec0 ac108-codec0-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0

Pillar1989, 2018-01-16:
Note: https://github.com/respeaker/seeed-voicecard/blob/master/asound_4mic.conf#L25 must be set right device

dm1010111, 2018-01-17:
Hi, I tried this method but it does not work.  I start from scratch to install the driver and google package and 4-mic is working.  I will close this comment and open a new Respeaker 2mic comment.

closed 2018-01-17 by dm1010111

dm1010111, 2018-01-17:
Start from scratch and re-install OS, driver and google assistant SDK.  Now the 4mic respeaker works again.