respeaker/seeed-voicecard #25: Cannot set pcm format

GadgetmanStewart, 2018-01-13:
I am trying on Pi raspbian "stretch" (all up to date) to record with the 4mic respeaker card and following this guide:-
http://wiki.seeed.cc/ReSpeaker_4_Mic_Array_for_Raspberry_Pi/

Here is the error:-
pi@pigoogle:~ $ arecord -D ac108 -f S32_LE -r 16000 -c 4 hello.wav
Recording WAVE 'hello.wav' : Signed 32 bit Little Endian, Rate 16000 Hz, Channels 4
ALSA lib pcm_ac108.c:48:(ac108_slave_hw_params_half) **Cannot set pcm format**
ALSA lib pcm_ac108.c:323:(ac108_hw_params) ac108_slave_hw_params_half error

arecord: set_params:1363: Unable to install hw params:
ACCESS:  RW_INTERLEAVED
FORMAT:  S32_LE
SUBFORMAT:  STD
SAMPLE_BITS: 32
FRAME_BITS: 128
CHANNELS: 4
RATE: 16000
PERIOD_TIME: 125000
PERIOD_SIZE: 2000
PERIOD_BYTES: 32000
PERIODS: 4
BUFFER_TIME: 500000
BUFFER_SIZE: 8000
BUFFER_BYTES: 128000
TICK_TIME: [0 0]
pi@pigoogle:~ $

Further the cards is installed:-
pi@pigoogle:~ $ arecord -L
null
    Discard all samples (playback) or generate zero samples (capture)
default
playback
dmixed
ac108
sysdefault:CARD=Audio
    USB Audio, USB Audio
    Default Audio Device
front:CARD=Audio,DEV=0
    USB Audio, USB Audio
    Front speakers
surround21:CARD=Audio,DEV=0
    USB Audio, USB Audio
    2.1 Surround output to Front and Subwoofer speakers
surround40:CARD=Audio,DEV=0
    USB Audio, USB Audio
    4.0 Surround output to Front and Rear speakers
surround41:CARD=Audio,DEV=0
    USB Audio, USB Audio
    4.1 Surround output to Front, Rear and Subwoofer speakers
surround50:CARD=Audio,DEV=0
    USB Audio, USB Audio
    5.0 Surround output to Front, Center and Rear speakers
surround51:CARD=Audio,DEV=0
    USB Audio, USB Audio
    5.1 Surround output to Front, Center, Rear and Subwoofer speakers
surround71:CARD=Audio,DEV=0
    USB Audio, USB Audio
    7.1 Surround output to Front, Center, Side, Rear and Woofer speakers
iec958:CARD=Audio,DEV=0
    USB Audio, USB Audio
    IEC958 (S/PDIF) Digital Audio Output
dmix:CARD=Audio,DEV=0
    USB Audio, USB Audio
    Direct sample mixing device
dsnoop:CARD=Audio,DEV=0
    USB Audio, USB Audio
    Direct sample snooping device
hw:CARD=Audio,DEV=0
    USB Audio, USB Audio
    Direct hardware device without any conversions
plughw:CARD=Audio,DEV=0
    USB Audio, USB Audio
    Hardware device with all software conversions
sysdefault:CARD=seeed4micvoicec
    seeed-4mic-voicecard,
    Default Audio Device
dmix:CARD=seeed4micvoicec,DEV=0
    seeed-4mic-voicecard,
    Direct sample mixing device
dsnoop:CARD=seeed4micvoicec,DEV=0
    seeed-4mic-voicecard,
    Direct sample snooping device
hw:CARD=seeed4micvoicec,DEV=0
    seeed-4mic-voicecard,
    Direct hardware device without any conversions
plughw:CARD=seeed4micvoicec,DEV=0
    seeed-4mic-voicecard,
    Hardware device with all software conversions
pi@pigoogle:~ $

Pillar1989, 2018-01-15:
Hi
I have tested in latest image. everything works fine.
[image]
```
pi@raspberrypi:~ $ arecord -L
null
    Discard all samples (playback) or generate zero samples (capture)
default
playback
dmixed
ac108
sysdefault:CARD=seeed4micvoicec
    seeed-4mic-voicecard, 
    Default Audio Device
dmix:CARD=seeed4micvoicec,DEV=0
    seeed-4mic-voicecard, 
    Direct sample mixing device
dsnoop:CARD=seeed4micvoicec,DEV=0
    seeed-4mic-voicecard, 
    Direct sample snooping device
hw:CARD=seeed4micvoicec,DEV=0
    seeed-4mic-voicecard, 
    Direct hardware device without any conversions
plughw:CARD=seeed4micvoicec,DEV=0
    seeed-4mic-voicecard, 
    Hardware device with all software conversions
pi@raspberrypi:~ $ lsb
lsblk        lsb_release  
pi@raspberrypi:~ $ lsb_release -a
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 9.1 (stretch)
Release:	9.1
Codename:	stretch
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.9.59-v7+ #1047 SMP Sun Oct 29 12:19:23 GMT 2017 armv7l GNU/Linux
pi@raspberrypi:~ $ arecord -D ac108 -f S32_LE -r 16000 -c 4 -v hello.wav
Recording WAVE 'hello.wav' : Signed 32 bit Little Endian, Rate 16000 Hz, Channels 4
AC108 decode Plugin
Its setup is:
  stream       : CAPTURE
  access       : RW_INTERLEAVED
  format       : S32_LE
  subformat    : STD
  channels     : 4
  rate         : 16000
  exact rate   : 16000 (16000/1)
  msbits       : 32
  buffer_size  : 8000
  period_size  : 2000
  period_time  : 125000
  tstamp_mode  : NONE
  tstamp_type  : GETTIMEOFDAY
  period_step  : 1
  avail_min    : 2000
  period_event : 0
  start_threshold  : 1
  stop_threshold   : 8000
  silence_threshold: 0
  silence_size : 0
  boundary     : 2097152000
^CAborted by signal Interrupt...

```

GadgetmanStewart, 2018-01-15:
You are running the full UI version of Raspbian (I see from the use of audacity).
I have built a new SD card with full UI, and it works.
Thanks

closed 2018-01-15 by GadgetmanStewart