respeaker/seeed-voicecard #26: Respeaker 2mic driver does not work on PI 3 B

dm1010111, 2018-01-17:
I have Raspberry Pi 3 B and hooked with respeaker 2-Mics Pi HAT.  I use Linux Rasberrypi OS 4.9.59-v7+ built on 10/29/2017. When the system up, "speaker-test -t wav" is OK.  I did following to install the 2-mics driver:
    $git clone https://github.com/respeaker/seeed-voicecard.git
    $cd seeed-voicecard
    $sudo ./install.sh 2mic
    $reboot
after system reboot, the "speaker-test -t wav" is broken.  The test 
     $ arecord -f cd -Dhw:1 | aplay -Dhw:1
Is broken. No record no playback.
Please help rootcause the issue.  It looks like "sudo ./install 2mic" 2mic driver has some issue.

dm1010111, 2018-01-17:
Add some observations:
If I use "sudo ./uninstall 2mic", then the speaker-test will work again. fro some reason, after installing 2mic driver, the on Pi 3 board speaker is disabled.

Pillar1989, 2018-01-17:
```
pi@raspberrypi:~/seeed-voicecard $ arecord -l
**** List of CAPTURE Hardware Devices ****
card 1: seeed2micvoicec [seeed-2mic-voicecard], device 0: bcm2835-i2s-wm8960-hifi wm8960-hifi-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
pi@raspberrypi:~/seeed-voicecard $ 
```
 seeed2micvoicec is   card 1 ?

dm1010111, 2018-01-17:
Hi Baozhu, on my Pi 3 and respeaker 2mic system, I can pass pass " arecord -f cd -Dhw:1 | aplay -Dhw:1"echo test.  

I got google-assistant-demo stuck.  Following is the message:
$google-assistant-demo --device_model_id 2micberry
device_model_id: 2micberry
device_id: ACDCBE4E571C34433D72A4DB1AFC502C

ON_MUTED_CHANGED:
  {'is_muted': False}
ON_START_FINISHED
[2360:2386:ERROR:mic_failure_detector.cc(140)] [assistant] Channel 1 has failed. Powers -inf and -inf

$aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: seeed2micvoicec [seeed-2mic-voicecard], device 0: bcm2835-i2s-wm8960-hifi wm8960-hifi-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0

$arecord -l
**** List of CAPTURE Hardware Devices ****
card 1: seeed2micvoicec [seeed-2mic-voicecard], device 0: bcm2835-i2s-wm8960-hifi wm8960-hifi-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0

I just no clue how to debug this issue.  Please let me know what is the solution.

Pillar1989, 2018-01-18:
https://github.com/respeaker/seeed-voicecard/blob/master/asound_2mic.conf#L22
https://github.com/respeaker/seeed-voicecard/blob/master/asound_2mic.conf#L29
should be "hw:1,0"

dm1010111, 2018-01-18:
Finally I resolve the issue with following .asoundrc file:
# The IPC key of dmix or dsnoop plugin must be unique
# If 555555 or 666666 is used by other processes, use another one

pcm.!default {
    type asym
    playback.pcm "playback"
    capture.pcm "capture"
}

pcm.playback {
    type plug
    slave.pcm "dmixed"
}

pcm.capture {
    type plug
    slave.pcm "array"
}

pcm.dmixed {
    type dmix
    slave.pcm "hw:1,0"
    ipc_key 555555
}

pcm.array {
    type dsnoop
    slave {
        pcm "hw:1,0"
        channels 2
    }
    ipc_key 666666
}
Thank you for all your helps, Baozhu.

calinichta, 2018-01-20:
I just updated and upgraded my Raspbian Linux Kernel to the mentioned kernel (Linux Rasberrypi OS 4.9.59-v7+) but I'm still not able to install the drivers.

"journalctl -xe" lists the following:

-- Unit systemd-modules-load.service has begun starting up.
Jan 20 00:49:30 raspberrypi systemd-modules-load[1036]: Failed to find module 'snd-soc-ac108'
Jan 20 00:49:30 raspberrypi systemd[1]: systemd-modules-load.service: Main process exited, code=exited, status=1/FAILURE
Jan 20 00:49:30 raspberrypi systemd[1]: Failed to start Load Kernel Modules.
-- Subject: Unit systemd-modules-load.service has failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- Unit systemd-modules-load.service has failed.
-- 
-- The result is failed.
Jan 20 00:49:30 raspberrypi systemd[1]: systemd-modules-load.service: Unit entered failed state.
Jan 20 00:49:30 raspberrypi systemd[1]: systemd-modules-load.service: Failed with result 'exit-code'.
Jan 20 00:49:30 raspberrypi sudo[1030]: pam_unix(sudo:session): session closed for user root

Seems that snd-soc-ac108 is not generated, so I'm not able to see the seed2mic-device in alsamixer or somewhere else. I'm stuck since 4 hours.

Does anyone now what's wrong?

dkmerona, 2018-01-21:
have a similar problem; did sudo ./install.sh 2mic; reboot and aplay -l dont list the card

Pillar1989, 2018-01-22:
@calinichta  I think you need reflash the sd card. then reinstall the seeed-voicecard.

dkmerona, 2018-01-22:
i just boot on sd card, all my system file is on an usb without problem till now;

calinichta, 2018-01-22:
I reinstalled the image about 4 times - but on berryboot with an external
USB drive.

Setting up the latest raspbian with kernel 4.9.57 on the SD card only woks
fine.

Taking a berryboot image with kernel 4.9.57 also works fine, but not on my
time intensive and carefully installed image based on kernel 4.9.26v7-aufs.

There, the systemd complains about not finding the right drivers
snd-soc-ac108.ko and snd-soc-wm9860.ko

When I start "sudo ./install.sh" it tells me that there is no build folder
in the given kernel what is right indeed.

Does anyone got successfully build drivers for kernel 4.9.26v7-aufs in
folder "/lib/modules/4.9.26v7-aufs/kernel/sound/soc/codecs"?

Thank you very much in advance.

Pillar1989, 2018-01-22:
4.9.26v7-aufs？ we only tested seeed-voicecard on raspbian.

calinichta, 2018-01-22:
This is the raspbian image 2017-09 for berryboot with kernel 4.9.26v7-aufs -
might this cause the problem?
As I already mentioned, running an image with kernel 4.9.57 is fine.

dkmerona, 2018-01-23:
for the comment on linux file, for sure its linux file because my linux system is running well for a lot of applications. typing yname -a give 4.9.59-v7

Pillar1989, 2018-01-26:
@calinichta Have you solved your issue ?

calinichta, 2018-01-26:
@Baozhu Zuo
Unfortunately I didn't have time to fix it, maybe next week...but I try hard

dkmerona, 2018-01-26:
i have kernel 4.9.59-v7 and it's not working. Any help?

calinichta, 2018-01-26:
I had updated to that kernel from 4.9.27 and it also didn't work, because
typing "uname -r" it still says "kernel 4.9.27...after setting up a
complete new image based on kernel 4.9.59-v7 it worked, because" uname -r"
says 4.9.59-v7.

I know that this is no help for you, but maybe you experience the same.

darles, 2018-01-26:
Same here - not working with 4.9.26v7-aufs

Pillar1989, 2018-01-29:
@darles  @dkmerona Where can I download berryboot image ?

calinichta, 2018-01-29:
You'll find berryboot images on https://berryboot.alexgoldcheidt.com/images/

dkmerona, 2018-01-29:
anyway can we hope have a patch of driver if needed  for 4.9.59-v7 or anybody from seeed confirming that actual drivers are ok for this kernel?

calinichta, 2018-01-29:
As I already wrote, drivers work fine on an image with kernel
4.9.57-v7...but not if it was updated from 4.9.26v7-aufs.

dkmerona, 2018-01-29:
arg^^; so what to do when you are using a 4.9.57-v7; what are the steps to follow because i updated the kernel and after that i download the driver and tried to install these drivers. May be i uninstall drivers and try to install drivers once more time?

calinichta, 2018-01-29:
That didn't work for me. I had to get an image that comes with kernel
4.9.57-v7

dkmerona, 2018-01-29:
well i cannot install a new image as i have other application running ; i need a runnable driver

Pillar1989, 2018-01-30:
@dkmerona If you have raspberrypi-kernel-headers for 4.9.57-v7 version, you can ignore the new image.

ukayzm, 2018-01-30:
I'm having the same issue with the initial issue of dm1010111, but 4 mic in my case. I have Raspberry Pi 3 B, respeaker 4-Mics Pi HAT, using Raspbian stretch lite, 4.9.59-v7+ built on 10/29/2017.

Before installing 4mic driver, speaker-test -t wav works fine. After installing 4mic driver by sudo ./install.sh 4mic and reboot, speaker-test -t wav stops forever with this output on console:

```
$ speaker-test -t wav
speaker-test 1.1.3

Playback device is default
Stream parameters are 48000Hz, S16_LE, 1 channels
WAV file(s)
Rate set to 48000Hz (requested 48000Hz)
Buffer size range from 12000 to 18000
Period size range from 6000 to 6000
Using max buffer size 18000
Periods = 4
was set period_size = 6000
was set buffer_size = 18000
 0 - Front Left
```

After the above output, it does not go any more, but stops forever. I have to press ^C to get out of the stopped speaker-test. It gives this output like this when pressing ^C.

```
^CWrite error: -4,Interrupted system call
xrun_recovery failed: -4,Interrupted system call
Transfer failed: Interrupted system call
```

When I uninstall the driver by sudo ./uninstall.sh 4mic; sudo ./uninstall.sh 2mic, and reboot, speaker-test -t wav works fine. 
 I have tried this from re-installing OS several times, but the result is always same. 

The strange thing is that the recording works well, even when the playback fails. So I don't think that my 4 mic hat is physically damaged. 

Another strange thing is this log on booting up:

```
[    3.460557]  i2c_id number :0
[    3.460567]  ac108  codec_index :0
[    3.460571]  ac108  I2S data protocol type :1
[    3.461115] ac108_write error->[REG-0x00,val-0x12]
```

ukayzm, 2018-01-30:
FYI, This is the output of arecord and aplay, which is different from the one in http://wiki.seeed.cc/ReSpeaker_4_Mic_Array_for_Raspberry_Pi/ 

```
$ arecord -L
null
    Discard all samples (playback) or generate zero samples (capture)
default
playback
dmixed
ac108
sysdefault:CARD=seeed4micvoicec
    seeed-4mic-voicecard,
    Default Audio Device
dmix:CARD=seeed4micvoicec,DEV=0
    seeed-4mic-voicecard,
    Direct sample mixing device
dsnoop:CARD=seeed4micvoicec,DEV=0
    seeed-4mic-voicecard,
    Direct sample snooping device
hw:CARD=seeed4micvoicec,DEV=0
    seeed-4mic-voicecard,
    Direct hardware device without any conversions
plughw:CARD=seeed4micvoicec,DEV=0
    seeed-4mic-voicecard,
    Hardware device with all software conversions
$ aplay -L
null
    Discard all samples (playback) or generate zero samples (capture)
default
playback
dmixed
ac108
sysdefault:CARD=ALSA
    bcm2835 ALSA, bcm2835 ALSA
    Default Audio Device
dmix:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Direct sample mixing device
dmix:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Direct sample mixing device
dsnoop:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Direct sample snooping device
dsnoop:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Direct sample snooping device
hw:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Direct hardware device without any conversions
hw:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Direct hardware device without any conversions
plughw:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Hardware device with all software conversions
plughw:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Hardware device with all software conversions
$ 
```
Could you please review my issue? Thank you in advance.

Pillar1989, 2018-02-01:
@ukayzm we are so sorry for your encounter. We have been optimized seeed-voicecard installation scheme.  Please use the new image to install it.
`git clone https://github.com/respeaker/seeed-voicecard -b install `

ukayzm, 2018-02-01:
@Pillar1989, I re-installed driver and now it works fine. Thank you very much.

jabulonboaz, 2018-02-07:
@Pilar1989 and all you have a new image for 2mic too? Or a way to fix this issue please?
I am on a Pi3 and running AlexaPi but i can't get the the record to work. 
Even if i replace the input_device with the error log suggestions bellow then i get different errors: Error log follows:

  pi@raspberrypi:~ $ sudo systemctl status AlexaPi.service
? AlexaPi.service - Alexa client for all your devices
   Loaded: loaded (/usr/lib/systemd/system/AlexaPi.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Wed 2018-02-07 01:35:46 UTC; 3min 56s ago
     Docs: https://github.com/alexa-pi/AlexaPi/wiki
  Process: 590 ExecStart=/usr/bin/python /opt/AlexaPi/src/main.py --daemon (code=exited, status=1/FAILURE)
 Main PID: 590 (code=exited, status=1/FAILURE)

Feb 07 01:35:45 raspberrypi python[590]: ALSA lib pcm_dsnoop.c:556:(snd_pcm_dsnoop_open) The dsnoop plugin supports only capture stream
Feb 07 01:35:46 raspberrypi python[590]: CRITICAL: Your input_device 'seeed-2mic-voicecard' is invalid. Use one of the following:
Feb 07 01:35:46 raspberrypi python[590]: seeed-2mic-voicecard: - (plughw:0,0)
Feb 07 01:35:46 raspberrypi python[590]: sysdefault
Feb 07 01:35:46 raspberrypi python[590]: capture
Feb 07 01:35:46 raspberrypi python[590]: array
Feb 07 01:35:46 raspberrypi python[590]: default
Feb 07 01:35:46 raspberrypi systemd[1]: AlexaPi.service: Main process exited, code=exited, status=1/FAILURE
Feb 07 01:35:46 raspberrypi systemd[1]: AlexaPi.service: Unit entered failed state.
Feb 07 01:35:46 raspberrypi systemd[1]: AlexaPi.service: Failed with result 'exit-code'.

Pillar1989, 2018-02-07:
@jabulonboaz Please use a new raspbian image install `git clone https://github.com/respeaker/seeed-voicecard -b install` . Then setup AlexaPi to your system. install will be merge to master branch quickly.

jabulonboaz, 2018-02-07:
@Pillar1989 is this link o for the 2mic drivers yes?

SimonGurney, 2018-02-07:
I had this on a Pi Zero.

Did the following

sudo ./uninstall.sh 2mic
reboot
sudo ./uninstall.sh 4mic
reboot
sudo ./install.sh 2mic

jabulonboaz, 2018-02-07:
I tried that @SimonGurney i saw it on your other post then figured out the ".sh" missing bit from your other post.
No luck the above git link does not install it but the old .git url cloned does but i was advised to clone with this new  -b install way.
Do i have to uninstall with the hat off?
I saw u just wrote that somewhere. 
Btw my final goal is what u have to move the build on a Pi0w from the Pi3.
Any pointers are welcomed here as i been "fighting" with this thing for ages.

SimonGurney, 2018-02-07:
I did the uninstalls and installs with the hat off as it would kill my pi when I connected it.

Not sure what the new method is but I cloned with the link off the wiki which pulled the git repo.

Do you get anything?  I almost gave up hope but when I checked the button on it was working and I could see it’s state changing in python.

jabulonboaz, 2018-02-07:
I am using AlexaPi and going headless no button Simon.
Are you building Alexa with AlexaPi or AVS? 
I will try your method with the normal git.

SimonGurney, 2018-02-07:
There is a button on the 2mic hat that you can read the state from via the gpio.

I’m actually just using it for audio out in a different project, also headless.  Might try the alexapi project if it gets a bit better.  Would really want drop in support before I went with it.

jabulonboaz, 2018-02-07:
@Pillar1989 your kind support please how do i install with just this command ?
git clone https://github.com/respeaker/seeed-voicecard -b install

jabulonboaz, 2018-02-07:
@SimonGurney so what did you use for your build?

SimonGurney, 2018-02-07:
This guide http://wiki.seeed.cc/Respeaker_2_Mics_Pi_HAT/

jabulonboaz, 2018-02-07:
@Pillar1989 

● AlexaPi.service - Alexa client for all your devices
   Loaded: loaded (/usr/lib/systemd/system/AlexaPi.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Wed 2018-02-07 22:05:22 UTC; 1min 30s ago
     Docs: https://github.com/alexa-pi/AlexaPi/wiki
  Process: 562 ExecStart=/usr/bin/python /opt/AlexaPi/src/main.py --daemon (code=exited, status=1/FAILURE)
 Main PID: 562 (code=exited, status=1/FAILURE)

Feb 07 22:05:22 raspberrypi python[562]:     self.fetch_more_tokens()
Feb 07 22:05:22 raspberrypi python[562]:   File "/usr/local/lib/python2.7/dist-packages/yaml/scanner.py", line 220, in fetch_more_tokens
Feb 07 22:05:22 raspberrypi python[562]:     return self.fetch_value()
Feb 07 22:05:22 raspberrypi python[562]:   File "/usr/local/lib/python2.7/dist-packages/yaml/scanner.py", line 576, in fetch_value
Feb 07 22:05:22 raspberrypi python[562]:     self.get_mark())
Feb 07 22:05:22 raspberrypi python[562]: yaml.scanner.ScannerError: mapping values are not allowed here
Feb 07 22:05:22 raspberrypi python[562]:   in "/etc/opt/AlexaPi/config.yaml", line 12, column 37
Feb 07 22:05:22 raspberrypi systemd[1]: AlexaPi.service: Main process exited, code=exited, status=1/FAILURE
Feb 07 22:05:22 raspberrypi systemd[1]: AlexaPi.service: Unit entered failed state.
Feb 07 22:05:22 raspberrypi systemd[1]: AlexaPi.service: Failed with result 'exit-code'.

jabulonboaz, 2018-02-07:
i can hear the test sound and i can verify that recording works but alexa does not star up due to audio errors. this happens only with the Respeaker, when i do the same build with a USB mic and the audio jack of the RPi3 Alexa Pi works prefect;y. @yexiaobo-seeedstudio @Pillar1989 we need some support with details please.

Pillar1989, 2018-02-08:
@jabulonboaz  We didnt test with Alexa PI. Do you read this document https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging ? 
Alexa PI have answered the issue.
```
Feb 07 22:05:22 raspberrypi python[562]: in "/etc/opt/AlexaPi/config.yaml", line 12, column 37
```

jabulonboaz, 2018-02-08:
@Pillar1989
Yes i read it over and over and yes i have already explained that i tried all the combinations and replaced line 12 that is referring to he input device with all the combinations of names for Respeaker 2mic. What is your suggestion please ? And u still did not answer my original question on how to install your you drivers as the git URL u provided does not do anything. Your support would be highly appreciated.

SimonGurney, 2018-02-08:
Might not work but I reference the speaker card as “hw:1,0”

jabulonboaz, 2018-02-08:
Thank you @SimonGurney i will try that format. I have even tried  plughw:0,0 but i only got as far as being able to play back and record a test wav. I know respeaker works cuz i used it with another git than AlexaPi which  was AssistantsPi but there is some issue here with the naming of the Respeaker. 
Will give your suggestion a go and report back mate cheers.

jabulonboaz, 2018-02-08:
"hw:0,0" did not work.

@Pillar1989 @yexiaobo-seeedstudio  @anyone that can help please.
This in the URL link bellow please find the detailed debug information, i cant get this to work :(

Your ALSA information is located at http://www.alsa-project.org/db/?f=37022a4253e49bbbc09120eb9411cbe71b51c9f8

pi@raspberrypi:~ $ cd ~ && wget https://raw.githubusercontent.com/alexa-pi/AlexaPi-misc/master/alsa-info.sh && bash alsa-info.sh
--2018-02-08 20:07:28--  https://raw.githubusercontent.com/alexa-pi/AlexaPi-misc/master/alsa-info.sh
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 151.101.60.133
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|151.101.60.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 27872 (27K) [text/plain]
Saving to: ‘alsa-info.sh.1’

alsa-info.sh.1      100%[===================>]  27.22K  --.-KB/s    in 0.03s

2018-02-08 20:07:29 (865 KB/s) - ‘alsa-info.sh.1’ saved [27872/27872]

ALSA Information Script v 0.4.64
--------------------------------

This script visits the following commands/files to collect diagnostic
information about your ALSA installation and sound related hardware.

  dmesg
  lspci
  lsmod
  aplay
  amixer
  alsactl
  /proc/asound/
  /sys/class/sound/
  ~/.asoundrc (etc.)

See 'alsa-info.sh --help' for command line options.

cat: '/sys/module/snd_soc_simple_card/parameters/*': No such file or directory
Automatically upload ALSA information to www.alsa-project.org? [y/N] : y
Uploading information to www.alsa-project.org ...  Done!

Your ALSA information is located at http://www.alsa-project.org/db/?f=37022a4253e49bbbc09120eb9411cbe71b51c9f8

SimonGurney, 2018-02-08:
How about “hw:1,0”

jabulonboaz, 2018-02-08:
@SimonGurney show me your aplay -l please.  your cardmust be =1 that is why hw:1,0 works for ypu but my card is at 0 and my device is at 0 as you can see from my debug file.

Why should i try hw:1,0 ?

SimonGurney, 2018-02-08:
Yes you’re right, hdmi is my 0 card

SimonGurney, 2018-02-08:
> I have given up on Alexa pi and i have successfully used AssistantsPi https://github.com/shivasiddharth/Assistants-Pi<https://eur01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fshivasiddharth%2FAssistants-Pi&data=02%7C01%7C%7C0cbf870b59a94544fdc808d56f4e25b4%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C636537303779886293&sdata=XCI4RhjaxjYLBAR7YeQp%2FrGmN4OBC5Br%2F61tksAsFfg%3D&reserved=0>
> I have not only completed the install on a Pi3 but all so on a PiOw and it works like a charm. I have only installed Alexa from the options of assistant pi as with Google assistant and Alexa running simultaneously it won't work.
> The way i done it was to install the AssitantsPi by choosing option 2 Alexa only and choose 1 for AIY hat as soundcard option. That leaves the asound.conf empty. I have an AIY hat so i could test the sound but u don't need to have one i believe you can just go ahead and finish the install any way without confirming the sound check. Then when i finished installing on the pi3 (btw i skipped the very last command of the instructions on the Assistant Pi git that activated Google assistant) i removed the AIY hat and placed the Respeaker 2 mic on the Pi3. I Then downloaded the Rspeaker 2mic drivers from their git https://github.com/respeaker/seeed-voicecard<https://eur01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Frespeaker%2Fseeed-voicecard&data=02%7C01%7C%7C0cbf870b59a94544fdc808d56f4e25b4%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C636537303779886293&sdata=AH9leQ%2FFfPVovh5HYLBEKIdR884Jsg7q10Gbj6G5kuI%3D&reserved=0> and the rebooted.
> [... cut ...]
> Any way very please Alexa from AssistantsPi works headless and with the hotword activation -no button needed, and the packages is so small it can fit into anything. My next step it to use Snowboy for custom wake words and activate some of the GPiO pins to be able to control a relay from the PiO. Thank you

Nice, do you know if it supports drop in calls?

jabulonboaz, 2018-02-09:
I will test it and report back Simon. For sure it supports music streaming from tunein and controls my smarthome devices.

jabulonboaz, 2018-02-09:
I still like to know how to do it in assistant pi too though. The respeaker works no drought the issue is how to make it work with AlexaPi.

Pillar1989, 2018-02-09:
@jabulonboaz 
The squeaking wheel gets the oil. 
You win. 
That's page for you. https://github.com/respeaker/seeed-voicecard/wiki/AlexaPi

jabulonboaz, 2018-02-09:
@Pillar1989 WOW! you fixed it  i will try it immediately!!  
One question it will work on the latest Raspbian jassie lite OS - (no desktop GUI) right?  
Thank you!

closed 2018-02-09 by Pillar1989

jabulonboaz, 2018-02-09:
"tutorial for nincompoop" that a bit rood mate... 
Again @Pillar1989 like i mentioned to you in different streams thank you for trying but the new build command does not build anything.

pi@raspberrypi:~ $ arecord -f cd | aplay
arecord: main:788: audio open error: No such file or directory
aplay: playback:2787: read error
pi@raspberrypi:~ $

I followed your new guide to the letter:

"git clone https://github.com/respeaker/seeed-voicecard -b install
cd seeed-voicecard/
pi@raspberrypi:~/seeed-voicecard $ sudo ./install.sh "

pi@raspberrypi:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
Subdevices: 8/8
Subdevice #0: subdevice #0
Subdevice #1: subdevice #1
Subdevice #2: subdevice #2
Subdevice #3: subdevice #3
Subdevice #4: subdevice #4
Subdevice #5: subdevice #5
Subdevice #6: subdevice #6
Subdevice #7: subdevice #7
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
Subdevices: 1/1
Subdevice #0: subdevice #0

is this suppose to run on Raspbian stretch with desktop only? I m on Raspbian Jessie lite. DO i need to install a specific version of python maybe?

Pillar1989, 2018-02-09:
Please login gitter at Beijing time 10 in the morning. I will teach you how to do this.

dkmerona, 2018-02-25:
Well. Coming back hoping thinks Evolved but still install not OK.
Linux raspberrypi 4.9.59-v7+ #1047 SMP Sun Oct 29 12:19:23 GMT 2017 armv7l GNU/Linux

Whe following the guide to the letter (^^) i get the following error message 👍 
[image]
[image]

i've tried the --force command but same result and aplay -l give 👍 
aplay -l
**** Liste des Périphériques Matériels PLAYBACK ****
carte 0: ALSA [bcm2835 ALSA], périphérique 0: bcm2835 ALSA [bcm2835 ALSA]
  Sous-périphériques: 8/8
  Sous-périphérique #0: subdevice #0
  Sous-périphérique #1: subdevice #1
  Sous-périphérique #2: subdevice #2
  Sous-périphérique #3: subdevice #3
  Sous-périphérique #4: subdevice #4
  Sous-périphérique #5: subdevice #5
  Sous-périphérique #6: subdevice #6
  Sous-périphérique #7: subdevice #7
carte 0: ALSA [bcm2835 ALSA], périphérique 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Sous-périphériques: 1/1
  Sous-périphérique #0: subdevice #0

Well what a weird product!!

jabulonboaz, 2018-02-26:
execute raspi-config [you need sudo for this]
-Go to Interfacing options Enable SPI
-Reboot

[4 more replies]