respeaker/seeed-voicecard #28: Raspberry Pi zero W incompatibility

giano, 2018-01-25:
Hi, I'm trying to use the 2mic version of Seeed HAT on a Raspberry Pi zero W. I managed to compile the modules on 4.9.59+ kernel (the 4.9.77+ does not have headers at the moment, so I freezed it at the stable branch). 

The HAT is not recognized in aplay/arecord and I have these errors:

  60.592823] gpiomem-bcm2835 20200000.gpiomem: Initialised: Registers at 0x20200000
[   60.947960] wm8960 1-001a: Failed to issue reset
[   60.948135] wm8960: probe of 1-001a failed with error -121

on dmesg

and 

 snd_soc_ac108: loading out-of-tree module taints kernel.

before. I expected the latter.

I can't use the HAT in any way. 

I tested the same exact configuration on a RPI3 (I swapped the microSDs and mounted the HAT on it) and it's fully working here, led and Google Assistant tested and functioning with audio in and out.

This is the HAT mounting (without the speaker at the moment, to simplify things)

[image]

I have no error on compilation and already tried the "./uninstall 4mic ./uninstall 2mic reboot ./install 2mic" procedure with no success. 

Thanks a lot, can't wait to use this product to its full extents.

Pillar1989, 2018-01-26:
Did you meet with "Hash mismatch" like #27 ?

giano, 2018-01-26:
Sorry but no. Must say that I already upgraded (and dist-upgraded) before
launching installer. Install logs are clean and modules are in place,
loaded at boot and instantiated, just can't signal to the hat.

Pillar1989, 2018-01-26:
@giano I have confirmed `sudo ./install 2mic` works on Raspberry Pi zero W with https://downloads.raspberrypi.org/raspbian_lite_latest 

- Check  the hardware connection, 2mic should be inserted in the board and then powered on and do play/recored test.
- Exclude other system setting. You can flash the new sd card with the lastest image , the install 2mic hat software

giano, 2018-01-26:
Ok, I'll redo the process with the attached image. Thank you. Do you suggest apt-get upgrade before?

Pillar1989, 2018-01-26:
No, I didnt do apt-get upgrade.

giano, 2018-01-26:
ok, thank you. I'll follow you along on it. Will post on this later or close the issue if all ok :)

giano, 2018-01-26:
Same issue. I'm attaching the entire log. As you can see no errors occurs during installation.

The hat seems correctly attached to me. The pins seems stable and I'm using an original 2x20 GPIO Header. How can I check this? Do I have to "overpower" the hat using the usb input on RPI zero?

Thank you.

[rpi.log](https://github.com/respeaker/seeed-voicecard/files/1667354/rpi.log)

Pillar1989, 2018-01-29:
```
[    8.753167] wm8960 1-001a: Failed to issue reset
[    8.753337] wm8960: probe of 1-001a failed with error -121
```
It's means the 2mic-hat connection is not correctly.

giano, 2018-01-29:
Thank you. So the issue is "hardware related". I'll test it on another RPI Zero with pre-soldered headers. As the Hat is working on the RPI3 I think the issue is on the Zero itself or the way it's connected. Closing this issue, as not software related. Thank you for your help anyway.

closed 2018-01-29 by giano