respeaker/seeed-voicecard #30: This is not working - No Respeaker card gets installed

jabulonboaz, 2018-02-09:
Again @Pillar1989 like i mentioned to you in different streams thank you for trying but the new build command does not build anything. 

I followed your new guide to the letter:

"git clone https://github.com/respeaker/seeed-voicecard -b install
cd seeed-voicecard/
pi@raspberrypi:~/seeed-voicecard $ sudo ./install.sh "

pi@raspberrypi:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

Pillar1989, 2018-02-10:
@jabulonboaz Please login gitter at Beijing time zoon 10 am. I will teach you how to do this. setup by setup.

Pillar1989, 2018-02-10:
https://gitter.im/seeed-voicecard/Lobby?utm_source=badge&utm_medium=badge&utm_campaign=pr-badge&utm_content=badge

Pillar1989, 2018-02-10:
@jabulonboaz Do you  online ? Please  response me.

jabulonboaz, 2018-02-10:
Big thanks to @Pillar1989 for finding in the issue and fixing it for me. Apparently the new drivers would not install due to an i2c configuration or enable issue that neded to be performed in raspi-config. Standing by for @Pillar1989 to post here the options that he changed for others that might run into the same issue to beniefit from too.

cogmeta, 2018-02-10:
Can someone post the fix here please?

cogmeta, 2018-02-10:
I enabled i2c using raspi-config but it still does not work. Can @Pillar1989 please put the details of the fix here. Thanks.

jabulonboaz, 2018-02-10:
@cogmeta i am waiting for @Pillar1989 to to clarify it as when he did it it was via team viewer and he told me he will provide full instructions, He did it to fast so i could not follow everything but its like a few commands only, whilst you wait here is the relevant "history" of cmd form my pi that gives a clue as to what he did and by following the bellow i belive you will succeed:
(I have added some comments #IN CAPITAL LETTERS for you next to the commands for you @cogmeta, line 27 to 34 is when he took over and before that i was just following his lead - the commands are bellow use them without the numbers and the comments):
 
 8  git clone https://github.com/respeaker/seeed-voicecard -b install
    9  cd seeed-voicecardsudo ./install.sh
   10  pi@raspberrypi:~ $
   11  cd seeed-voicecard
   12  sudo ./install.sh
   13  sudo reboot
   14  aplay -l #AT THIS POINT THE DRIVER HAD INSTALLED BUT THE RE-SPEAKER CARD WAS NOT APPEARING - THIS WAS THE PROBLEM IN THE FIRST PLACE 
   15  jabulonboaz @jabulonboaz 09:32
   16  i2cdetect -y 1
   17  i2cdetect -y
   18  i2cdetect -y 1
   19  sudo apt install i2c-tools
   20  i2cdetect -y 1
   21  cat /boot/config.txt
   22  cat /boot/config.txt | grep i2c
   23  ls /dev/i2c* #THIS WAS NOT WORKING 
   24  ls /dev/i2c #THIS WAS NOT WORKING
   25  ls /dev/i2c* #THIS WAS NOT WORKING
   26  ls /dev/ # THIS COMMAND WORKED -PILLAR WAS IN THE TERMINAL NOW VIA TEAM VIEWER 
   27  sudo raspi-config #HERE I BELIEVE HE ENABLED I2C IN THE RPI MENU UNDER INTERFACES 
   28  ls /dev/i2c-1
   29  cat /boot/config.txt | grep i2c
   30  sudo i2cdetect -y 1
   31  sudo systemctl restart seeed-voicecard
   32  aplay -l #SUCCESS THE RE-SPEAKER CARD NOW APPEARS !!!
   33  arecord -l #SUCCESS THE RE-SPEAKER CARD NOW APPEARS !!!
   34  arecord -f cd | aplay  #THIS COMAND IS TESING THE MIC AND SPEAKER DO NOT WORRY ABUT THE SOUND AFTER YOU HEAR IT HIT CONTROL+C

@cogmeta bear in mind that we did this with anew clean install of a Raspbian Jessie lite and instaling the respeaker 2mic drivers was the first step before i preceded with the installation of AlexaPi . Now before i started installing AlexaPi i blacklisted the RPi internal sound card just in case like this: 
#Create this file like this:
cd /etc/modprobe.d
sudo nano alsa-blacklist.conf
#PUT THIS LINE IN THE FILE 
blacklist snd_bcm2835
sudo reboot 

then when you come back to the terminals you do aplay -l and you are only left with the respeaker 2mic voice card. When i completed the AlexaPi installation its was working straight away i never need to do anything in the audio settings of the Alexa file configuration file.  I did made one change though and that was to activate GPIO 17 on the AlexaPi configuratio file for the repseaker button to work and be able to activate Alexa from there too as well as my voice - this you can find in the final part of the instructions give in the respeaker git by @Pillar1989 
https://github.com/respeaker/seeed-voicecard/wiki/AlexaPi

good luck @cogmeta

jabulonboaz, 2018-02-10:
@Pillar1989 one thing i cannot figure out is when i adjust the volume in "alsamixer" and then i store it using "sudo alsactl store" the seeings reset after reboot and Alexa is too loud at maximum audio level - how can i save the alsamixer settings for ever?
Also 2 more questions are:
1. How can i make the Respeaker LEDs trigger when Alexa listens and speaks ? - i tried changing the GPIO settings in the AlexaPi configuration file with GPIO 10 and 11 but nothing
2: Ho can i map 2 GPIO pins to be used with a button each reference to Ground so that when i press one button the volume of the "alsamixer" "Playback" to increase and when i press the other button for the volume to decrease ? - basically i  like to be able to control the volume with buttons

cogmeta, 2018-02-11:
Thank you so much @jabulonboaz. I followed all the steps but no luck so far. Will keep digging further.

cogmeta, 2018-02-11:
It is little frustrating and annoying that seed studio is makes us wait (possibly hundreds of us) for what might be a small fix. This is not a professional behaviour. I can expect and live with it if it  were a open source contribution  but not from a company who got paid for the product they have sold and expected to provide a speedy fix when things are not working.

jabulonboaz, 2018-02-11:
@cogmeta try also to do this:

"Enable/Disable SPI interfaces and automatic loading of the SPI kernel module, needed for products such as PiFace."

Enable the "automatic loading of SPI kernel" i have no idea what it is but just emailed Pillar1989 asking if he can come back to us and he provided this enigmatic short answer from the Raspberry pi foundation web site:
 https://www.raspberrypi.org/documentation/configuration/raspi-config.md

cogmeta, 2018-02-11:
@jabulonboaz You are awesome. That fixed it. You have to enable enable SPI and I2C intefacing options using raspi-config. Thank you so much.

For others facing similar problem. Try this! This might resolve the issue for you as well.

1. execute raspi-config [you need sudo for this]
2.  Go to 5. Interfacing options
3. Enable SPI
4. Enable I2C (not sure whether this is required)
5. reboot

then reinstall seeed-voicecard and reboot.

jabulonboaz, 2018-02-11:
It was @Pillar1989 that provided it so thanks to him. I am happy yours is working too now and we have a way to reproduce it @cogmeta.
Btw if you know how to make the LEDs trigger please email @cogmeta at jabulonboaz@hotmail.com i like the Respeaker LEDs to trigger when alexapi listens and responds i found some basic info on Seeed's wiki but cant go further to integrate it with alexapi

cogmeta, 2018-02-11:
@jabulonboaz  Surely, we will certainly help you with that.  We have 10 more such  devices to install and check and then I will get back to you with solution for the LED. We owe you one :) Thanks a lot

jabulonboaz, 2018-02-11:
Do not enable SPI @congmeta i just did to try play with the LEDs but it rigs AlexaPi and will just say yes but no answer

cogmeta, 2018-02-11:
Folks, in addition you

jabulonboaz, 2018-02-11:
Help anyone please ? Even if i restore my build from a clone i build of my SD card when it was working i have the same behaviour. Alexa says hello at boot and responds with hello when u talk to her but does not respond further with an answer. Something to do with these SPI settings?

jabulonboaz, 2018-02-11:
AlexaPi stoped working after i enabled SPI then i disable it again but nothing.
Alexa says hello at boot and recognises the hot word when i say alexa she comes back with hello but never responds further to any questions.

Please help.

Sent from Blue<http://www.bluemail.me/r?b=11998>

cogmeta, 2018-02-11:
@jabulonboaz we are actually building our own assistant like Alexa so we are not really testing Alexa or google home. But, yes I was able to install the seeed voice card successfully and both mic and audio jack are working fine.

cogmeta, 2018-02-11:
I think for your case problem is more likely with alexa itself, most likely network issue. SPI settings should not affect it. Can you do both aplay and arecord tests to check if the card is installed properly.

jabulonboaz, 2018-02-11:
Happy to test it too when u guys are ready

jabulonboaz, 2018-02-11:
pi@raspberrypi:~ $ sudo arecord -f S16_LE -r 16000 -D your_input_device test.wav
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM your_input_device
arecord: main:788: audio open error: No such file or directory
pi@raspberrypi:~ $ cd ~ && wget https://raw.githubusercontent.com/alexa-pi/AlexaPi-misc/master/alsa-info                 .sh && bash alsa-info.sh
--2018-02-11 17:01:06--  https://raw.githubusercontent.com/alexa-pi/AlexaPi-misc/master/alsa-info.sh
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 151.101.60.133
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|151.101.60.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 27872 (27K) [text/plain]
Saving to: ‘alsa-info.sh’

alsa-info.sh              100%[=====================================>]  27.22K  --.-KB/s    in 0.02s

2018-02-11 17:01:07 (1.26 MB/s) - ‘alsa-info.sh’ saved [27872/27872]

ALSA Information Script v 0.4.64
--------------------------------

This script visits the following commands/files to collect diagnostic
information about your ALSA installation and sound related hardware.

  dmesg
  lspci
  lsmod
  aplay
  amixer
  alsactl
  /proc/asound/
  /sys/class/sound/
  ~/.asoundrc (etc.)

See 'alsa-info.sh --help' for command line options.

cat: '/sys/module/snd_soc_simple_card/parameters/*': No such file or directory
Automatically upload ALSA information to www.alsa-project.org? [y/N] : y
Uploading information to www.alsa-project.org ...  Done!

Your ALSA information is located at http://www.alsa-project.org/db/?f=b74386dedb4be67b75c6bbdf4674c8939b                 4958d1

jabulonboaz, 2018-02-11:
http://www.alsa-project.org/db/?f=b74386dedb4be67b75c6bbdf4674c8939b 4958d4958d1

jabulonboaz, 2018-02-11:
http://www.alsa-project.org/db/?f=b74386dedb4be67b75c6bbdf4674c8939b4958d4958d1

jabulonboaz, 2018-02-11:
Feb 11 16:55:50 raspberrypi python[466]: INFO: AVS token: Requesting a new one
Feb 11 16:56:00 raspberrypi python[466]: INFO: AVS token: Obtained successfully
Feb 11 16:56:05 raspberrypi python[466]: [0263ca58] core input error: ES_OUT_SET_(GROUP_)PCR  is called too late (pts_delay increased to 300 ms)
Feb 11 16:56:05 raspberrypi python[466]: [0263ca58] core input error: ES_OUT_RESET_PCR called
Feb 11 16:56:13 raspberrypi python[466]: INFO: Triggered: pocketsphinx
Feb 11 16:56:17 raspberrypi python[466]: INFO: (process_response Error) Status Code: 403
Feb 11 16:56:34 raspberrypi python[466]: INFO: Triggered: pocketsphinx
Feb 11 16:56:38 raspberrypi python[466]: INFO: (process_response Error) Status Code: 403
Feb 11 17:08:02 raspberrypi python[466]: INFO: Triggered: pocketsphinx
Feb 11 17:08:06 raspberrypi python[466]: INFO: (process_response Error) Status Code: 403

jabulonboaz, 2018-02-11:
these are my logs help anyone pls?

jabulonboaz, 2018-02-11:
Feb 11 17:17:38 raspberrypi python[470]: ALSA lib pcm_dmix.c:990:(snd_pcm_dmix_open) The dmix plugin supports only playback stream
Feb 11 17:17:39 raspberrypi python[470]: ALSA lib pcm_dsnoop.c:556:(snd_pcm_dsnoop_open) The dsnoop plugin supports only capture stream
Feb 11 17:20:06 raspberrypi python[470]: INFO: Checking Internet Connection ...
Feb 11 17:20:07 raspberrypi python[470]: INFO: Connection OK
Feb 11 17:20:07 raspberrypi python[470]: INFO: AVS token: Requesting a new one
Feb 11 17:20:07 raspberrypi python[470]: INFO: AVS token: Obtained successfully
Feb 11 17:20:17 raspberrypi python[470]: INFO: Triggered: pocketsphinx
Feb 11 17:20:21 raspberrypi python[470]: INFO: (process_response Error) Status Code: 403
Feb 11 17:20:23 raspberrypi python[470]: INFO: Triggered: pocketsphinx
Feb 11 17:20:31 raspberrypi python[470]: INFO: (process_response Error) Status Code: 403

jabulonboaz, 2018-02-11:
I rebuild-ed everything from scratch over and over, the sound card installed and works fine, Alexa listens and response with yes but never continues to replay to the question i asked. 
She just says yes and then silence, here are my logs, any idea anyone please?

Feb 11 19:30:17 raspberrypi python[474]: INFO: Checking Internet Connection ...
Feb 11 19:30:18 raspberrypi python[474]: INFO: Connection OK
Feb 11 19:30:18 raspberrypi python[474]: INFO: AVS token: Requesting a new one
Feb 11 19:30:18 raspberrypi python[474]: INFO: AVS token: Obtained successfully
Feb 11 19:30:30 raspberrypi python[474]: INFO: Triggered: pocketsphinx
Feb 11 19:30:33 raspberrypi python[474]: INFO: (process_response Error) Status Code: 403
Feb 11 19:30:42 raspberrypi python[474]: INFO: Triggered: pocketsphinx
Feb 11 19:30:46 raspberrypi python[474]: INFO: (process_response Error) Status Code: 403
Feb 11 19:32:19 raspberrypi python[474]: INFO: Triggered: pocketsphinx
Feb 11 19:32:23 raspberrypi python[474]: INFO: (process_response Error) Status Code: 403

jabulonboaz, 2018-02-11:
I have also raised an issue at the AlexaPi git:
https://github.com/alexa-pi/AlexaPi/issues/376

jabulonboaz, 2018-02-11:
http://www.alsa-project.org/db/?f=1bfc46ac5afe839401bc592ecf43f70c3b25e93c

detailed audio debug logs here @Pillar1989

jabulonboaz, 2018-02-11:
pi@raspberrypi:~ $ sudo journalctl -u AlexaPi
-- Logs begin at Sun 2018-02-11 20:17:26 UTC, end at Sun 2018-02-11 20:21:50 UTC. --
Feb 11 20:17:47 raspberrypi systemd[1]: Started Alexa client for all your devices.
Feb 11 20:17:59 raspberrypi python[468]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.front
Feb 11 20:17:59 raspberrypi python[468]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
Feb 11 20:17:59 raspberrypi python[468]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
Feb 11 20:17:59 raspberrypi python[468]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
Feb 11 20:17:59 raspberrypi python[468]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
Feb 11 20:17:59 raspberrypi python[468]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround40
Feb 11 20:17:59 raspberrypi python[468]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround41
Feb 11 20:17:59 raspberrypi python[468]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround50
Feb 11 20:17:59 raspberrypi python[468]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround51
Feb 11 20:17:59 raspberrypi python[468]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround71
Feb 11 20:17:59 raspberrypi python[468]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
Feb 11 20:17:59 raspberrypi python[468]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
Feb 11 20:17:59 raspberrypi python[468]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
Feb 11 20:17:59 raspberrypi python[468]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
Feb 11 20:17:59 raspberrypi python[468]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
Feb 11 20:17:59 raspberrypi python[468]: ALSA lib pcm_dmix.c:990:(snd_pcm_dmix_open) The dmix plugin supports only playba
Feb 11 20:18:00 raspberrypi python[468]: ALSA lib pcm_dsnoop.c:556:(snd_pcm_dsnoop_open) The dsnoop plugin supports only
Feb 11 20:18:00 raspberrypi python[468]: ALSA lib pcm_dmix.c:990:(snd_pcm_dmix_open) The dmix plugin supports only playba
Feb 11 20:18:00 raspberrypi python[468]: ALSA lib pcm_dsnoop.c:556:(snd_pcm_dsnoop_open) The dsnoop plugin supports only
Feb 11 20:18:04 raspberrypi python[468]: INFO: Checking Internet Connection ...
Feb 11 20:18:15 raspberrypi python[468]: INFO: Connection OK
Feb 11 20:18:15 raspberrypi python[468]: INFO: AVS token: Requesting a new one
Feb 11 20:18:16 raspberrypi python[468]: INFO: AVS token: Obtained successfully
Feb 11 20:18:22 raspberrypi python[468]: INFO: Triggered: pocketsphinx
Feb 11 20:18:25 raspberrypi python[468]: INFO: (process_response Error) Status Code: 403
Feb 11 20:18:30 raspberrypi python[468]: INFO: Triggered: pocketsphinx
Feb 11 20:18:34 raspberrypi python[468]: INFO: (process_response Error) Status Code: 403
Feb 11 20:18:37 raspberrypi python[468]: INFO: Triggered: pocketsphinx
Feb 11 20:18:41 raspberrypi python[468]: INFO: (process_response Error) Status Code: 403
Feb 11 20:18:47 raspberrypi python[468]: INFO: Triggered: pocketsphinx
Feb 11 20:18:56 raspberrypi python[468]: INFO: (process_response Error) Status Code: 403
Feb 11 20:18:57 raspberrypi python[468]: INFO: Triggered: platform
Feb 11 20:19:01 raspberrypi python[468]: INFO: (process_response Error) Status Code: 403
Feb 11 20:19:04 raspberrypi python[468]: INFO: Triggered: platform
Feb 11 20:19:13 raspberrypi python[468]: INFO: (process_response Error) Status Code: 403

hexbioc, 2018-02-13:
I am using the currently latest version of Raspbian Lite (2017-11-29), and I came across the same problem after installing seeed-voicecard. I solved it by installing the i2c-tools package and restarting the voicecard service as follows:
``sudo apt-get install i2c-tools # Status of the voicecard service hinted at the use of i2cdetect, which it couldn't find
sudo systemctl restart seeed-voicecard``
Recording seems to be perfect now.

jabulonboaz, 2018-02-15:
@hexbioc thank you! I was missing that part with the i2c-tools

Pillar1989, 2018-02-16:
(https://github.com/respeaker/seeed-voicecard/commit/b462e8a724573bec45d387cfea06a9774751d251)  I'm sorry for this issue. I have updated install scripts.

closed 2018-02-16 by Pillar1989

trbedwards, 2019-05-24:
> @jabulonboaz You are awesome. That fixed it. You have to enable enable SPI and I2C in interfacing options using raspi-config. Thank you so much.
> 
> For others, who are facing similar problem, please try this.
> 
> 1. execute raspi-config [you need sudo for this]
> 2. Go to 5. Interfacing options
> 3. Enable SPI
> 4. Enable I2C (not sure whether this is required)
> 5. reboot
> 
> then install seeed-voicecard and reboot. Additionally you might have to do this
> 
> sudo systemctl restart seeed-voicecard

Thanks! This worked for me, but the key thing is that you have to reboot multiple times! Reboot after enabling SPI and I2C, uninstall seed-voicecard then reboot, install seeed-voicecard then reboot.