respeaker/seeed-voicecard #31: JST connector Speaker

Warbowhunter, 2018-02-10:
Before installing Pulseaudio I can speak to the google assistant and It recognizes my speech. Using "speaker-test -t wav" works and you can hear the test sound. But whenever I'm asking something to the Google Assistant I can't hear the response. I can hear the JST connected speaker making a little beep when giving output and the beep disappearing when the response finished. I've tried a few things in Alsamixer but nothing (I tried) improved this. 

So then I installed Pulseaudio and ran it in the background (as in the tutorial). The problem now here is I can select only one profile: Output or Input. I want to use both. The JST connected speaker and the microphone on the board. How can I do this?

Raspberry Pi 3 B with the latest Raspbian image. Using the ReSpeaker 2-Mics Pi HAT and I followed this tutorial: http://wiki.seeed.cc/Respeaker_2_Mics_Pi_HAT/

Any help is really appreciated!

Pillar1989, 2018-02-16:
Hi
I didn't test pulseaudio with respeaker 2mic. Why do you need pulseaudio?

Warbowhunter, 2018-02-16:
Thanks for the reply! In the tutorial on the seeed website there are, when installing google assistant, a few commands to install Pulseaudio. So I thought you should

Without it I can' hear anything. The sound appears to be really soft. But I can't find the right setting in Alsamixer to fix this. Or I'm doing something else wrong but I don't know what.

Pillar1989, 2018-02-16:
let me ask for the demo author. We are all in Chinese New Year now, I can't reply you very quickly, but I will definitely figure out this issue.

Bhavitrajput, 2018-02-23:
Have you tried rpi 3's 3.5 mm audio output?
I think there is some issue in hat's audio output.
I am using rpi's audio output without installing pulse audio and it's working fine.

Warbowhunter, 2018-02-23:
Got it working! Someone on the Google + community had the same problem and a solution. If you ask the google assistant to Max the volume it highers a (virtual?) Volume. Deleting the .asoundrc config file (and rebooting) fixed my issue!

closed 2018-03-08 by Pillar1989