respeaker/seeed-voicecard #32: How to control the LEDs of Respeaker2mic via AlexaPi ?

jabulonboaz, 2018-02-10:
How can I control the top 3 rob leds of respeaker 2mic on a RaspberryPi zero w with an AlexaPi build please?

Pillar1989, 2018-02-16:
For LEDs of Respeaker 2mic  you can refer to this https://github.com/respeaker/pixel_ring

jabulonboaz, 2018-02-16:
@Pillar1989 thank u. This refers to the array jat but i have the 2mic hat. Anything relevant to 2mic hat please?

Pillar1989, 2018-02-16:
2mic hat use 3 LEDs. Please change PIXELS_N to 3.
https://github.com/respeaker/pixel_ring/blob/master/pixel_ring/apa102_pixel_ring.py#L14

jabulonboaz, 2018-02-16:
Can you make instructions to implement the respeaker 2mic leds with google assistant and AlexaPi projects please?

Pillar1989, 2018-02-16:
Sorry, We are all in Chinese New Years. maybe you can ask for google assistant and AlexaPi projects projects developer to implement  LEDs.

xiongyihui, 2018-02-27:
@jabulonboaz We have demos to use Google Assistant and Alexa Voice Service with LED light. See https://github.com/respeaker/mic_hat

jabulonboaz, 2018-02-27:
Thank you i have implemented the LEDs with GA for now. I have read the git instructions in the past but they are not detailed enough. It would be good if you could update these with more detail.

closed 2018-03-13 by Pillar1989