respeaker/seeed-voicecard #33: Capturing 4mic with ffmpeg spikes cpu to 100%+ and crashes

rpaddock, 2018-02-22:
Using latest master (b462e8a) with ffmpeg 3.4.2. I can capture with arecord with no issues. 

**Versions:**
```
$ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```
```
$ ffmpeg -version
ffmpeg version 3.4.2 Copyright (c) 2000-2018 the FFmpeg developers
built with gcc 6.3.0 (Raspbian 6.3.0-18+rpi1) 20170516
configuration: --prefix=/usr --disable-static --enable-gpl --enable-libass --enable-libfdk-aac --enable-libfreetype --enable-libx264 --enable-omx --enable-omx-rpi --enable-nonfree --enable-shared --enable-version3
libavutil      55. 78.100 / 55. 78.100
libavcodec     57.107.100 / 57.107.100
libavformat    57. 83.100 / 57. 83.100
libavdevice    57. 10.100 / 57. 10.100
libavfilter     6.107.100 /  6.107.100
libswscale      4.  8.100 /  4.  8.100
libswresample   2.  9.100 /  2.  9.100
libpostproc    54.  7.100 / 54.  7.100
```
```
$ arecord --version
arecord: version 1.1.5 by Jaroslav Kysela <perex@perex.cz>
```

**ffmpeg Command:**
```
ffmpeg -f alsa -c:a pcm_s32le -ac 4 -ar 16000 -i ac108 -c:a copy test.wav
ffmpeg version 3.4.2 Copyright (c) 2000-2018 the FFmpeg developers
  built with gcc 6.3.0 (Raspbian 6.3.0-18+rpi1) 20170516
  configuration: --prefix=/usr --disable-static --enable-gpl --enable-libass --enable-libfdk-aac --enable-libfreetype --enable-libx264 --enable-omx --enable-omx-rpi --enable-nonfree --enable-shared --enable-version3
  libavutil      55. 78.100 / 55. 78.100
  libavcodec     57.107.100 / 57.107.100
  libavformat    57. 83.100 / 57. 83.100
  libavdevice    57. 10.100 / 57. 10.100
  libavfilter     6.107.100 /  6.107.100
  libswscale      4.  8.100 /  4.  8.100
  libswresample   2.  9.100 /  2.  9.100
  libpostproc    54.  7.100 / 54.  7.100
Guessed Channel Layout for Input Stream #0.0 : stereo
Input #0, alsa, from 'ac108':
  Duration: N/A, start: 1519289577.617550, bitrate: 1536 kb/s
    Stream #0:0: Audio: pcm_s16le, 48000 Hz, stereo, s16, 1536 kb/s
Output #0, wav, to 'test.wav':
  Metadata:
    ISFT            : Lavf57.83.100
    Stream #0:0: Audio: pcm_s16le ([1][0][0][0] / 0x0001), 48000 Hz, stereo, s16, 1536 kb/s
Stream mapping:
  Stream #0:0 -> #0:0 (copy)
Press [q] to stop, [?] for help
[alsa @ 0x61b850] ALSA buffer xrun.
```

**Kernel Log:**
```
Feb 22 01:23:42 raspberrypi kernel: [ 1744.752723] ac10x_read error1 -11 tried 0
Feb 22 01:23:42 raspberrypi kernel: [ 1744.752726] ac10x_read error1 -11 tried 0
Feb 22 01:23:42 raspberrypi kernel: [ 1744.752729] ac10x_read error1 -11 tried 0
Feb 22 01:23:46 raspberrypi kernel: [ 1749.428370] ac10x_read error1 -11 tried 0
Feb 22 01:23:46 raspberrypi kernel: [ 1749.428377] ac10x_read error1 -11 tried 0
Feb 22 01:23:46 raspberrypi kernel: [ 1749.428380] ac10x_read error1 -11 tried 0
Feb 22 01:23:46 raspberrypi kernel: [ 1749.428382] ac10x_read error1 -11 tried 0
Feb 22 01:23:52 raspberrypi kernel: [ 1755.120939] i2c-bcm2835 3f804000.i2c: i2c transfer timed out
Feb 22 01:23:52 raspberrypi kernel: [ 1755.120949] ac10x_read error1 -110 tried 0
Feb 22 01:23:53 raspberrypi kernel: [ 1756.160930] i2c-bcm2835 3f804000.i2c: i2c transfer timed out
Feb 22 01:23:53 raspberrypi kernel: [ 1756.160939] ac10x_read error1 -110 tried 0
Feb 22 01:23:54 raspberrypi kernel: [ 1757.200933] i2c-bcm2835 3f804000.i2c: i2c transfer timed out
Feb 22 01:23:54 raspberrypi kernel: [ 1757.200942] ac10x_read error1 -110 tried 0
Feb 22 01:23:55 raspberrypi kernel: [ 1758.240930] i2c-bcm2835 3f804000.i2c: i2c transfer timed out
Feb 22 01:23:55 raspberrypi kernel: [ 1758.240939] ac10x_read error1 -110 tried 0
Feb 22 01:23:55 raspberrypi kernel: [ 1758.241490] ac10x_read error1 -5 tried 0
Feb 22 01:23:55 raspberrypi kernel: [ 1758.241515] ac10x_read error1 -5 tried 0
Feb 22 01:24:01 raspberrypi kernel: [ 1764.000935] i2c-bcm2835 3f804000.i2c: i2c transfer timed out
Feb 22 01:24:01 raspberrypi kernel: [ 1764.000944] ac10x_read error1 -110 tried 0
Feb 22 01:24:02 raspberrypi kernel: [ 1765.040933] i2c-bcm2835 3f804000.i2c: i2c transfer timed out
Feb 22 01:24:02 raspberrypi kernel: [ 1765.040942] ac10x_read error1 -110 tried 0
Feb 22 01:24:03 raspberrypi kernel: [ 1766.080926] i2c-bcm2835 3f804000.i2c: i2c transfer timed out
Feb 22 01:24:03 raspberrypi kernel: [ 1766.080934] ac10x_read error1 -110 tried 0
Feb 22 01:24:04 raspberrypi kernel: [ 1767.120931] i2c-bcm2835 3f804000.i2c: i2c transfer timed out
Feb 22 01:24:04 raspberrypi kernel: [ 1767.120939] ac10x_read error1 -110 tried 0
Feb 22 01:24:04 raspberrypi kernel: [ 1767.121185] ac10x_read error1 -5 tried 0
Feb 22 01:24:04 raspberrypi kernel: [ 1767.121216] ac10x_read error1 -5 tried 0
```

**arecord Command**
```
arecord -Dac108 -f S32_LE -r 16000 -c 4 --verbose out.wav
Recording WAVE 'out.wav' : Signed 32 bit Little Endian, Rate 16000 Hz, Channels 4
Plug PCM: Hardware PCM card 1 'seeed-4mic-voicecard' device 0 subdevice 0
Its setup is:
  stream       : CAPTURE
  access       : RW_INTERLEAVED
  format       : S32_LE
  subformat    : STD
  channels     : 4
  rate         : 16000
  exact rate   : 16000 (16000/1)
  msbits       : 32
  buffer_size  : 8000
  period_size  : 2000
  period_time  : 125000
  tstamp_mode  : NONE
  tstamp_type  : MONOTONIC
  period_step  : 1
  avail_min    : 2000
  period_event : 0
  start_threshold  : 1
  stop_threshold   : 8000
  silence_threshold: 0
  silence_size : 0
  boundary     : 2097152000
  appl_ptr     : 0
  hw_ptr       : 0
```

turmary, 2018-02-25:
hi rpaddock,
I tried to fix the problem with version [d225c6](https://github.com/respeaker/seeed-voicecard/commit/d225c664d787faf3668061753fd86d88c901bfd2), please test with it, tks.

rpaddock, 2018-03-01:
Seems to be resolved. I will reopen if I see it again. Thanks @turmary.

closed 2018-03-01 by rpaddock