respeaker/seeed-voicecard #34: Driver for banana pi

Bhavitrajput, 2018-02-23:
Can i use this respeaker 2 mic hat in banana pi.
How can i use this driver in banana pi.

RPiGuyXY, 2023-03-09:
Hi there @HinTak , first of all, thank you f all your efforts to keep respeaker going. It's ideal hardware for so many projects and a shame to be left behind.

With RPi's now impossible to find, I find myself in dire need of respeaker support in Banana Pi versions of Raspbian, where the WM8960 (2 mic) voicecard driver has the same issues in compilation shown by newer RPi kernels. I have tried diving into the logs, tried your repo, and still don't have a working solution. Please can we make at least the 2 mic voice card install compatible with banana-pi's raspbian kernel? How can I best contact you to discuss and explain? It's of great importance to me, and will be to the entire community now that RPis are essentially off the market and we must turn to alternatives. I'd really appreciate your help.

HinTak, 2023-03-09:
Quick look at some of the things that banana pi ships, look like they use a rather old and custom kernel , although more recent variant (from  3rd party) exists. So to support banana pi, looks like a choice between "back port driver to older kernel" or "graft a newer kernel onto older banana pi distro".. The latter is harder but more future-proof. Either options are somewhat involved so we probably should continue in a private discussion on commercial/contract work to get this done, if you want it done.