respeaker/seeed-voicecard #35: Muffled audio via 3.5mm audio jack

nicklee1990, 2018-02-24:
I have installed the linux kernel drivers for my Respeaker 2 mic pHat on my Pi Zero W and can confirm the microphone works perfectly (recorded audio out to file and played it back via a bluetooth speaker). However, when playing audio with the headphones connected to the headphone jack, the audio is barely audible and sounds garbled/muffled.

Does this sound like a hardware issue that may require a replacement Resepeaker 2 mic board?

Command used to test `arecord -f cd -Dhw:1 | aplay -Dhw:1`

Pillar1989, 2018-02-25:
Maybe you can record a video for me, I can help you solve the issue.

nicklee1990, 2018-02-25:
Tried on my Pi3 and it worked fine so looks like it could be a hardware issue specific to my Pi Zero. I need to take a look at the GPIO pins as I think there is an issue there. Will close this issue as it is unrelated to the Respeaker phat

closed 2018-02-25 by nicklee1990