respeaker/seeed-voicecard #36: Your kernel headers for kernel 4.14.21+ cannot be found at /lib/modules/4.14.21+/build or /lib/modules/4.14.21+/source.

nicklee1990, 2018-02-26:
## Description
I had previously had this working on a Raspberry Pi 3 B however after recently updating via `rpi-update` the seeed cards no longer appeared. `aplay -l` only lists the default card, and `arecord -l` lists nothing.

**OS**
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"

**uname -a**
Linux raspberrypi3 4.14.21-v7+ #1095 SMP Fri Feb 23 18:21:29 GMT 2018 armv7l GNU/Linux

I have ran an uninstall and observed the following error messages:

```
rm: cannot remove '/lib/modules/4.14.21-v7+/kernel/sound/soc/codecs/snd-soc-wm8960.ko': No such file or directory
rm: cannot remove '/lib/modules/4.14.21-v7+/kernel/sound/soc/codecs/snd-soc-ac108.ko': No such file or directory
rm: cannot remove '/lib/modules/4.14.21-v7+/kernel/sound/soc/generic/snd-soc-simple-card.ko': No such file or directory
```

During install I see the following errors:
```
DKMS: build completed.
Error! echo
Your kernel headers for kernel 4.14.21+ cannot be found at
/lib/modules/4.14.21+/build or /lib/modules/4.14.21+/source.
Error! echo
Your kernel headers for kernel 4.14.21-v7+ cannot be found at
/lib/modules/4.14.21-v7+/build or /lib/modules/4.14.21-v7+/source.
```

After reading #3 I have ran `sudo apt-get install dkms raspberrypi-kernel-headers` which tells me that both those packages are already the latest version.

The `/lib/modules/4.14.21+` folder and `/lib/modules/4.14.21-v7` exist but there is no build or source directories in there

nicklee1990, 2018-02-26:
Will downgrading help this?

closed 2018-03-08 by Pillar1989

stoneWeb, 2018-03-08:
@nicklee1990 so how did you fixed it? I had the same error.
`
cannot be found at
/lib/modules/4.14.24+/build or /lib/modules/4.14.24-v7+/source
`

nicklee1990, 2018-03-08:
I uninstalled this package, downgraded to 4.9.xx, rebooted then reinstalled.

You need to use rpi-update with the relevant commit hash as described in the rpi-update docs https://github.com/Hexxeh/rpi-update/blob/master/README.md. You can get the commit hash from the rpi-firmware repo https://github.com/Hexxeh/rpi-firmware/commits/master

tfontoura, 2018-04-04:
I solved it installing sources utilizing rpi-source : https://github.com/notro/rpi-source/wiki.

ash-lat, 2018-04-16:
Hi 

Is there other workaround while keeping the latest Kernel 4.14.24.
As one of my application requires rpi-update and after downgrade it stops working

Thanks

Geronimo2015, 2018-05-22:
I also have this problem after using rpi-update.

After downgrading to sudo rpi-update 5c80565c5c0c7f820258c792a98b56f22db2dd03
(4.9.80) I now get:

```Your kernel headers for kernel 4.14.41+ cannot be found at
/lib/modules/4.14.41+/build or /lib/modules/4.14.41+/source.
Error! echo
Your kernel headers for kernel 4.14.41-v7+ cannot be found at
/lib/modules/4.14.41-v7+/build or /lib/modules/4.14.41-v7+/source.
Error! echo
Your kernel headers for kernel 4.9.80+ cannot be found at
/lib/modules/4.9.80+/build or /lib/modules/4.9.80+/source.
Error! echo
Your kernel headers for kernel 4.9.80-v7+ cannot be found at
/lib/modules/4.9.80-v7+/build or /lib/modules/4.9.80-v7+/source.
```
So my problem remained or even worsened.

Any ideas how to solve this?

Pillar1989, 2018-05-23:
What's news in rpi-update?  Why do people like to use these bad future code ?

Geronimo2015, 2018-05-23:
@Pillar1989 Its not that clear it installs a non-stable build.

Geronimo2015, 2018-05-23:
I rebuilt Linux myself and installed modules from there which fixed my problem.
[https://www.raspberrypi.org/documentation/linux/kernel/building.md](url)

Afterwards I reinstalled Seeed voicecard.

nicklee1990, 2018-05-24:
@Pillar1989 I agree with @Geronimo2015 - I don't think it's a case of people using "bad" future code. It's just a case of people upgrading, perhaps unknowingly, to what is bleeding edge firmware and expecting this package will be compatible. It could probably be resolved adding something to the README regarding compatibility?

Michael5490, 2019-04-28:
Hello I still havé this Problem and When i downgrade to 4.9 m’y image fails because I have a rpi 3b+ 
Have you Gottéron an idea how to get kernel since version 4.14
Ive found Kernel until 4.9

Thx!!

turmary, 2019-04-29:
Hi Michael5490,
Please try install again.
or
`apt-get -y install raspberrypi-kernel-headers raspberrypi-kernel `

Michael5490, 2019-05-05:
Hi Turmary

Thanks for your answer. I tried this already but it doesn't work.

I found a solution to install mkarcadejoystick

I've done this to copy old kernel to new: 
`sudo ln -s /lib/modules/4.9.35-v7+/build /lib/modules/4.14.29-v7+/build`

Then the installation seems ok but at the end I've this error
ERROR: could not insert 'mk_arcade_joystick_rpi': Exec format error

I'm getting an error when running:
sudo modprobe mk_arcade_joystick_rpi map=1,2
it says the same error: 

ERROR: could not insert 'mk_arcade_joystick_rpi': Exec format error

reukiodo, 2021-05-26:
I solved a similar issue where the `build` and `source` folders were missing in `/lib/modules/5.10.36-v7+` by creating the missing symlinks:
```
ln -s /usr/src/linux-headers-5.10.17-v7+ /lib/modules/5.10.36-v7+/build
ln -s /usr/src/linux-headers-5.10.17-v7+ /lib/modules/5.10.36-v7+/source
```