respeaker/seeed-voicecard #38: Can this driver work on other linux distributions like Ubuntu? What about some linux-based OSes like OpenWRT?

HSID, 2018-02-28:
I want to use other OSes rather than raspbian for some reasons. Will that work?

Pillar1989, 2018-02-28:
let's open a vote, If 50+ people need this feature, we'll develop it.

HSID, 2018-02-28:
So the current version will only work for Raspbian?

Pillar1989, 2018-02-28:
Yes

Pillar1989, 2018-03-26:
Support ArchlinuxARM

keelobight, 2018-04-03:
PLEASE let this work on ubuntu for pi.  Many of us hate using raspbian and paid for this card before realizing it's "raspian only"  come on guys, support your customers!!!!!

Pillar1989, 2018-04-03:
@keelobight which ubuntu distribution do you like?

keelobight, 2018-04-03:
Currently i use xubuntu and the most.  
I also have a libre computer ROC-RK3328-CC that runs ubuntu.  If you could make it work on that, i would buy another unit probably the more more expensive one .

keelobight, 2018-04-06:
Are you really going to make it ubuntu compatible or are you just joking?

Pillar1989, 2018-04-08:
@keelobight  I think current project support raspberry pi ubuntu system. Do you test it? what's the issue?

keelobight, 2018-04-08:
sudo ./install.sh 2mic[sudo] password for alexapi: Hit:1 http://ppa.launchpad.net/ubuntu-pi-flavour-makers/ppa/ubuntu xenial InReleaseHit:2 http://ports.ubuntu.com xenial InReleaseHit:3 http://ports.ubuntu.com xenial-updates InReleaseHit:4 http://ports.ubuntu.com xenial-security InReleaseHit:5 http://ports.ubuntu.com xenial-backports InReleaseReading package lists... Done                     Building dependency tree       Reading state information... Done35 packages can be upgraded. Run 'apt list --upgradable' to see them.Reading package lists... DoneBuilding dependency tree       Reading state information... Doneraspberrypi-kernel-headers is already the newest version (1.20161215-1~xenial1.0).raspberrypi-kernel is already the newest version (1.20161215-1~xenial1.0).0 upgraded, 0 newly installed, 0 to remove and 35 not upgraded.Reading package lists... DoneBuilding dependency tree       Reading state information... Donedkms is already the newest version (2.2.0.3-2ubuntu11.5).git is already the newest version (1:2.7.4-0ubuntu1.3).i2c-tools is already the newest version (3.1.2-3~xenial1.0).0 upgraded, 0 newly installed, 0 to remove and 35 not upgraded.
------------------------------Deleting module version: 0.3completely from the DKMS tree.------------------------------Done.
Creating symlink /var/lib/dkms/seeed-voicecard/0.3/source ->                 /usr/src/seeed-voicecard-0.3
DKMS: add completed.Error! Your kernel headers for kernel 4.14.32+ cannot be found.Please install the linux-headers-4.14.32+ package,or use the --kernelsourcedir option to tell DKMS where it's locatedError! Your kernel headers for kernel 4.14.32-v7+ cannot be found.Please install the linux-headers-4.14.32-v7+ package,or use the --kernelsourcedir option to tell DKMS where it's located
Kernel preparation unnecessary for this kernel.  Skipping...
Building module:cleaning build area....make KERNELRELEASE=4.4.38+ -C /lib/modules/4.4.38+/build M=/var/lib/dkms/seeed-voicecard/0.3/build.......(bad exit status: 2)ERROR (dkms apport): binary package for seeed-voicecard: 0.3 not foundError! Bad return status for module build on kernel: 4.4.38+ (armv7l)Consult /var/lib/dkms/seeed-voicecard/0.3/build/make.log for more information.
Kernel preparation unnecessary for this kernel.  Skipping...
Building module:cleaning build area....make KERNELRELEASE=4.4.38-v7+ -C /lib/modules/4.4.38-v7+/build M=/var/lib/dkms/seeed-voicecard/0.3/build.......(bad exit status: 2)ERROR (dkms apport): binary package for seeed-voicecard: 0.3 not foundError! Bad return status for module build on kernel: 4.4.38-v7+ (armv7l)Consult /var/lib/dkms/seeed-voicecard/0.3/build/make.log for more information.------------------------------------------------------Please reboot your raspberry pi to apply all settingsEnjoy!------------------------------------------------------
This is what I get not matter what distro I use.  Sorry I'm kind of a noob, but I've spent SO many hours trying to get this thing to work.  Can you please give me specific instructions on what to do?  I can get it to work on an old version of raspbian from 9-2017, so I know the hardware works, but I hate raspbian and only use other distros.  Please give me a little help here.  I've tried every ubuntu distro there is and even newer versions of raspian with no luck.  I'm starting to wish I hadn't spent money on this product that has such poor documentation.

Pillar1989, 2018-04-08:
```
DKMS: add completed.Error! Your kernel headers for kernel 4.14.32+ cannot be found.Please install the linux-headers-4.14.32+ package,or use the --kernelsourcedir option to tell DKMS where it's locatedError! Your kernel headers for kernel 4.14.32-v7+ cannot be found.Please install the linux-headers-4.14.32-v7+ package,or use the --kernelsourcedir option to tell DKMS where it's located
```

raspberrypi.org only provide  4.14.30-v7+  headers. seeed-voicecard total support raspberry pi 4.14+ linux kernel.  But raspberry pi ubuntu lack of software package.

keelobight, 2018-04-08:
Ok so i just need an older kernel?  No problem.  Which one works best?  Ive tried older ones and get the same result....

Pillar1989, 2018-04-08:
@keelobight  4.14.30-v7+ headers and 4.14.30-v7+  kernel image ？

Pillar1989, 2018-05-15:
Please reopen this issue if  it really generates demand.

closed 2018-05-15 by Pillar1989

bogeuh, 2018-12-25:
i'd love to be able to use this on ubuntu (lubuntu). had to install it for my ros robot on a raspberry

sofar i got the leds to work

bogeuh, 2018-12-25:
got it to work

whb883, 2019-01-18:
I also need to use it in XXX Ubuntu Mate 16.04，Still no support plan?