respeaker/seeed-voicecard #40: ReSpeaker 4mic : unable to record (arecord: main:722: audio open error: Invalid argument)

channelz, 2018-03-05:
Hello. Please help. Have installed/uninstalled many times. (all failed) Please advise....

1) Enabled SPI and i2c. 
2) sudo apt-get install i2c-tools
3) sudo ./install.sh 4mic
4) reboot
5) sudo systemctl restart seeed-voicecard

$arecord -Dac108 -f S32_LE -r 16000 -c 4 hello.wav
ALSA lib pcm_direct.c:999:(snd1_pcm_direct_initialize_slave) unable to install hw params
ALSA lib pcm_dsnoop.c:629:(snd_pcm_dsnoop_open) unable to initialize slave
arecord: main:722: audio open error: Invalid argument

$dmesg:
[   33.349595]  i2c_id number      : 0
[   33.349610]  ac10x data protocol: 0
[   33.370589]  ac10x codec count  : 1
[   33.371536] ac108_set_sysclk  :24000000
[   33.380092] asoc-simple-card soc:sound: ac10x-codec0 <-> 3f203000.i2s mapping ok
[   91.940526] bcm2835-i2s 3f203000.i2s: bcm2835_i2s_hw_params:bad format
[   91.940612] bcm2835-i2s 3f203000.i2s: ASoC: can't set 3f203000.i2s hw params: -22

$i2cdetect -y -r 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- UU -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

$ arecord -L
null
    Discard all samples (playback) or generate zero samples (capture)
pulse
    PulseAudio Sound Server
playback
dmixed
ac108
multiapps
default
sysdefault:CARD=seeed4micvoicec
    seeed-4mic-voicecard,
    Default Audio Device
dmix:CARD=seeed4micvoicec,DEV=0
    seeed-4mic-voicecard,
    Direct sample mixing device
dsnoop:CARD=seeed4micvoicec,DEV=0
    seeed-4mic-voicecard,
    Direct sample snooping device
hw:CARD=seeed4micvoicec,DEV=0
    seeed-4mic-voicecard,
    Direct hardware device without any conversions
plughw:CARD=seeed4micvoicec,DEV=0
    seeed-4mic-voicecard,
    Hardware device with all software conversions

channelz, 2018-03-05:
$ arecord -Dac108 -f S32_LE -r 16000 -c 4 hello.wav
ALSA lib pcm_direct.c:999:(snd1_pcm_direct_initialize_slave) unable to install hw params
ALSA lib pcm_dsnoop.c:629:(snd_pcm_dsnoop_open) unable to initialize slave
arecord: main:722: audio open error: Invalid argument

 $ arecord -l
**** List of CAPTURE Hardware Devices ****
card 1: seeed4micvoicec [seeed-4mic-voicecard], device 0: bcm2835-i2s-ac10x-codec0 ac10x-codec0-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0

aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

turmary, 2018-03-06:
Hi,
Have you used the old version driver which install a plugin /usr/lib/arm-linux-gnueabihf/alsa-lib/libasound_module_pcm_ac108.so ?
please try to delete it and try again.
sudo rm /usr/lib/arm-linux-gnueabihf/alsa-lib/libasound_module_pcm_ac108.so

channelz, 2018-03-06:
*file dies not exist

$ sudo ls /usr/lib/arm-linux-gnueabihf/alsa-lib/libasound_module_pcm_ac108.*
ls: cannot access /usr/lib/arm-linux-gnueabihf/alsa-lib/libasound_module_pcm_ac108.*: No such file or directory

$uname -a
Linux raspberrypi 4.9.35-v7+ #1014 SMP Fri Jun 30 14:47:43 BST 2017 armv7l GNU/Linux

$cat /etc/os-release

PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
NAME="Raspbian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

channelz, 2018-03-06:
I have just successfully installed this on a pi3 running stretch.  
But would like this to run on my pi2 b running jessie

channelz, 2018-03-06:
What should $HOME/.asoundrc look like in "Raspbian GNU/Linux 8 (jessie)"  in an operational Respeaker 4mic environment?

(I feel this is where the issue is)

turmary, 2018-03-07:
It's should not exist.

turmary, 2018-03-07:
Hi channelz,
I found the old kernel 4.9.35 don't support DSP_A mode which current version driver used.
So, I recommend you use the history [version](https://github.com/respeaker/seeed-voicecard/tree/b9d9cde471358855ff66b3d18eb3d0f2c8a73987) using I2S-mode.
Or else update your kernel.

channelz, 2018-03-08:
Thank you. I will try tomorrow. What are the drawbacks of using /seeed-voicecard/tree/b9d9cde471358855ff66b3d18eb3d0f2c8a73987 ?

stoneWeb, 2018-03-09:
@turmary I tried to install this [version](https://github.com/respeaker/seeed-voicecard/tree/b9d9cde471358855ff66b3d18eb3d0f2c8a73987).

my kernel is also 4.9.35.
In this case：
```
$ arecord -L

null
    Discard all samples (playback) or generate zero samples (capture)
pulse
    PulseAudio Sound Server
playback
dmixed
ac108
```

and I tried to restart:
```
$ sudo systemctl restart seeed-voicecard

Failed to restart seeed-voicecard.service: Unit seeed-voicecard.service failed to load: No such file or directory.
```

why??????????

turmary, 2018-03-09:
@stoneWeb 
Firstly Make sure the installing process you doing has no any errors, paste here if it has.
Secondly, check system modules (generated by command lsmod)  include the item snd_soc_ac108.

stoneWeb, 2018-03-10:
There was no error in the install process. and the lsmod command include snd_soc_ac108. so what's the problem?

stoneWeb, 2018-03-10:
this is install logs:
```
Creating symlink /var/lib/dkms/seeed-voicecard/0.2/source ->
                 /usr/src/seeed-voicecard-0.2

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area....
make KERNELRELEASE=4.9.35+ -C /lib/modules/4.9.35+/build M=/var/lib/dkms/seeed-voicecard/0.2/build......
cleaning build area....

DKMS: build completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area....
make KERNELRELEASE=4.9.35-v7+ -C /lib/modules/4.9.35-v7+/build M=/var/lib/dkms/seeed-voicecard/0.2/build......
cleaning build area....

DKMS: build completed.

snd-soc-wm8960:
Running module version sanity check.

Good news! Module version  for snd-soc-wm8960.ko
exactly matches what is already found in kernel 4.9.35+.
DKMS will not replace this module.
You may override by specifying --force.

snd-soc-ac108.ko:
Running module version sanity check.

Good news! Module version  for snd-soc-ac108.ko
exactly matches what is already found in kernel 4.9.35+.
DKMS will not replace this module.
You may override by specifying --force.

depmod.....

DKMS: install completed.

snd-soc-wm8960:
Running module version sanity check.

Good news! Module version  for snd-soc-wm8960.ko
exactly matches what is already found in kernel 4.9.35-v7+.
DKMS will not replace this module.
You may override by specifying --force.

snd-soc-ac108.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.9.35-v7+/kernel/sound/soc/codecs/

depmod.....

DKMS: install completed.
cp ac108_asound.state /var/lib/alsa/asound.state
------------------------------------------------------
Please reboot your raspberry pi to apply all settings
Enjoy!
```

turmary, 2018-03-11:
try to use command
i2cdetect -y -r 1
to see if the hardware connection is broken.
then paste result here.

stoneWeb, 2018-03-11:
@turmary 
```
$ sudo i2cdetect -y -r 1

     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- UU -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```

turmary, 2018-03-11:
and try to use dmesg, paste the result start from "i2c_id number", like this
[ 33.349595] i2c_id number : 0
[ 33.349610] ac10x data protocol: 0
[ 33.370589] ac10x codec count : 1
[ 33.371536] ac108_set_sysclk :24000000
[ 33.380092] asoc-simple-card socsound ac10x-codec0 <-> 3f203000.i2s mapping ok
[ 91.940526] bcm2835-i2s 3f203000.i2s: bcm2835_i2s_hw_params:bad format
[ 91.940612] bcm2835-i2s 3f203000.i2s: ASoC: can't set 3f203000.i2s hw params: -22

stoneWeb, 2018-03-12:
@turmary I've tried the new raspbian Image, and the drive is normal, but run the voice-engine example get the error:

```
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4996:(snd_config_expand) Args evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4996:(snd_config_expand) Args evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa
ALSA lib pcm_dmix.c:990:(snd_pcm_dmix_open) The dmix plugin supports only playback stream
ALSA lib pcm_dmix.c:990:(snd_pcm_dmix_open) The dmix plugin supports only playback stream
ALSA lib pcm_ac108.c:469:(_snd_pcm_ac108_open) a108 is only for capture
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
```

And put headphones on the raspberry, using this command, without any sound.
```
$ aplay hello.wav
Playing WAVE 'hello.wav' : Signed 32 bit Little Endian, Rate 16000 Hz, Channels 4
```

```
$ aplay -L
null
    Discard all samples (playback) or generate zero samples (capture)
default
playback
dmixed
ac108
sysdefault:CARD=ALSA
    bcm2835 ALSA, bcm2835 ALSA
    Default Audio Device
dmix:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Direct sample mixing device
dmix:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Direct sample mixing device
dsnoop:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Direct sample snooping device
dsnoop:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Direct sample snooping device
hw:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Direct hardware device without any conversions
hw:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Direct hardware device without any conversions
plughw:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Hardware device with all software conversions
plughw:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Hardware device with all software conversions

$ arecord -L
null
    Discard all samples (playback) or generate zero samples (capture)
default
playback
dmixed
ac108
sysdefault:CARD=seeed4micvoicec
    seeed-4mic-voicecard,
    Default Audio Device
dmix:CARD=seeed4micvoicec,DEV=0
    seeed-4mic-voicecard,
    Direct sample mixing device
dsnoop:CARD=seeed4micvoicec,DEV=0
    seeed-4mic-voicecard,
    Direct sample snooping device
hw:CARD=seeed4micvoicec,DEV=0
    seeed-4mic-voicecard,
    Direct hardware device without any conversions
plughw:CARD=seeed4micvoicec,DEV=0
    seeed-4mic-voicecard,
    Hardware device with all software conversions
```

turmary, 2018-03-13:
Hi,
about [headphone output](https://www.jianshu.com/p/674145fe98fa).

gemaizi, 2018-03-13:
@turmary when i install a old version as you mention, then no card device detect, if i dont want to install a new raspbian Image, since i have a bunch of installed software on my current image, do i have any othe way? my current  kernel is 4.9.35, thanks!

turmary, 2018-03-14:
Following the content of the old version install.sh, check configuration files /etc/modules, /boot/config.txt, /boot/overlays/seeed-4mic-voicecard.dtbo in order. 
then check system modules (generated by command lsmod) include the item snd_soc_ac108.
then result of aplay -L && arecord -L.

gemaizi, 2018-03-14:
@turmary the command cat /etc/modules get the reault :
i2c-dev
snd-soc-simple-card
snd-soc-wm8960
snd-soc-ac108
the command cat /boot/config.txt,  get the reault :
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details
# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1
# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1
# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16
# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720
# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1
# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1
# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2
# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4
# uncomment for composite PAL
#sdtv_mode=2
#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800
# Uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c_arm=on
#dtparam=i2s=on
dtparam=spi=on
# Uncomment this to enable the lirc-rpi module
#dtoverlay=lirc-rpi
# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on
dtoverlay=pi3-disable-bt
dtoverlay=i2s-mmap

There is exist a file /boot/overlays/seeed-4mic-voicecard.dtbo
The there is an error in dmesg comand:

[    3.723089]  i2c_id number :0
[    3.723105]  ac108  codec_index :0
[    3.723113]  ac108  I2S data protocol type :1
[    3.723678] ac108_write error->[REG-0x00,val-0x12]

Is the "ac108_write error->[REG-0x00,val-0x12]" that matters?
Thanks!

turmary, 2018-03-15:
OK，the lastest driver changed kernel module snd-soc-simple-card.ko.
So, please run command 
'sudo apt-get install --reinstall raspberrypi-kernel' to recover it.

gemaizi, 2018-03-15:
@turmary it works! thank you very much!

closed 2018-03-19 by Pillar1989

creekhead, 2018-11-05:
> OK，the lastest driver changed kernel module snd-soc-simple-card.ko.
> So, please run commands
> 
> ```
> sudo rm -rf `find /lib/modules/ -name "*simple-card.ko*"`
> sudo apt-get install --reinstall raspberrypi-kernel
> ```
> to recover it, then reboot.

Careful with this command, it made my version unbootable and i had to start from scratch