respeaker/seeed-voicecard #41: Installation works but recording does not as the configuration seems to be wrong

CoolM8, 2018-03-05:
Im on Raspbian Jessie Lite and for the installation of the ac108 driver to work i had to do the following steps:

1. sudo dpkg --configure -a
2. edit the following line in install.sh:
   _apt-get -y install raspberrypi-kernel-headers raspberrypi-kernel_ to _apt-get -y --force-yes install raspberrypi-kernel-headers raspberrypi-kernel_
3. run sudo ./install.sh

After rebooting and entering _arecord -L_ I only see PulseAudio. Then when i enter _sudo systemctl restart seeed-voicecard_ i get the voice card to be listed, but it looks like the following:
`null
    Discard all samples (playback) or generate zero samples (capture)
pulse
    PulseAudio Sound Server
default
    Playback/recording through the PulseAudio sound server
playback
dmixed
ac108
multiapps
sysdefault:CARD=seeed4micvoicec
    seeed-4mic-voicecard,
    Default Audio Device
dmix:CARD=seeed4micvoicec,DEV=0
    seeed-4mic-voicecard,
    Direct sample mixing device
dsnoop:CARD=seeed4micvoicec,DEV=0
    seeed-4mic-voicecard,
    Direct sample snooping device
hw:CARD=seeed4micvoicec,DEV=0
    seeed-4mic-voicecard,
    Direct hardware device without any conversions
plughw:CARD=seeed4micvoicec,DEV=0
    seeed-4mic-voicecard,
    Hardware device with all software conversions`

When trying to record anything I get the following message:
_ALSA lib pcm_direct.c:999:(snd1_pcm_direct_initialize_slave) unable to install hw params
ALSA lib pcm_dsnoop.c:629:(snd_pcm_dsnoop_open) unable to initialize slave
arecord: main:722: audio open error: Invalid argument_

How can I bring this to work?

turmary, 2018-03-06:
Hi,
Do you used the old version driver which install a plugin /usr/lib/arm-linux-gnueabihf/alsa-lib/libasound_module_pcm_ac108.so.
please try to delete it  and try again.
sudo rm /usr/lib/arm-linux-gnueabihf/alsa-lib/libasound_module_pcm_ac108.so

CoolM8, 2018-03-06:
No, i followed the wiki for installation.
sudo rm /usr/lib/arm-linux-gnueabihf/alsa-lib/libasound_module_pcm_ac108.so
tells
me that there is no such file.

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

turmary, 2018-03-06:
Hi CoolM8,
OK, I want to know your kernel & image version generated by command below,
uname -a
cat /etc/os-release
Tell me the Raspbian Image download URL is prefer, which made me have the identical environment with you. Thanks.

CoolM8, 2018-03-06:
Just missed, but it is exactly the same issue as [Issue #4](https://github.com/respeaker/seeed-voicecard/issues/40)

turmary, 2018-03-20:
Duplicate of #40