respeaker/seeed-voicecard #43: Can respeak 4-mic be used with hifiberry sound card for audio output

CoolM8, 2018-03-10:
My respeaker 4-mic array is working properly and i can output the recorded audio with aplay onto the headphone. But now i would like to use my hifiberry amp+ to output sound.
From the documentation for the amp+ i have to edit /boot/config.txt and do the following:
 - remove 'dtparam=audio=on'
- add 'dtoverlay=hifiberry-amp'

Now when i reboot with the changes above, then i cannot find my respeaker wiht arecord -L but see the hifiberry device listest with aplay -L. I then added 'dtoverlay=seeed-4mic-voicecard' to the config.txt with no luck. When i then add  'dtparam=audio=on' again, the respeaker mic is listed again but the hifiberry is not.

Can this somehow be resolved? I really want to output audio with a proper soundcard and not raspberry pi's integrated audio channel.

Pillar1989, 2018-03-12:
Hi, How do you connect 4-mic hat and hifiberry sound card together ?

CoolM8, 2018-03-12:
I stacked them together through the GPIO connector. I think both hats would support i2c and therefore allow daisy-chaining them together. Or can / must i connect the respeaker hat differently?

CoolM8, 2018-03-12:
Can you explain me the device tree overlay? Do I have to somehow merge the dts.files for both hats together? In the seeed-4mic-voicecard-overlay.dts i see the following:
`fragment@3 {
		target = <&sound>;

		sound_overlay: __overlay__ {
			compatible = "simple-audio-card";
			simple-audio-card,format = "dsp_a";
			simple-audio-card,name = "seeed-4mic-voicecard"; 
			status = "okay";
		
			simple-audio-card,bitclock-master = <&codec_dai>;
			simple-audio-card,frame-master = <&codec_dai>;
			simple-audio-card,channels-playback-override = <4>;
			simple-audio-card,channels-capture-override  = <4>;

			cpu_dai: simple-audio-card,cpu {
				sound-dai = <&i2s>;
				dai-tdm-slot-num     = <2>;
				dai-tdm-slot-width   = <32>;
				dai-tdm-slot-tx-mask = <1 1 0 0>;
				dai-tdm-slot-rx-mask = <1 1 0 0>;
			};
			codec_dai: simple-audio-card,codec {
				sound-dai = <&ac108_a>;
				clocks =  <&ac108_mclk>;
			};
		};
	};`
in the overlay file for the hifiberry amp (hifiberry-amp-overlay.dts) i see the following:
`fragment@0 {
		target = <&sound>;
		__overlay__ {
			compatible = "hifiberry,hifiberry-amp";
			i2s-controller = <&i2s>;
			status = "okay";
		};
	};

	fragment@1 {
		target = <&i2s>;
		__overlay__ {
			status = "okay";
		};
	};

	fragment@2 {
		target = <&i2c1>;
		__overlay__ {
			#address-cells = <1>;
			#size-cells = <0>;
			status = "okay";

			tas5713@1b {
				#sound-dai-cells = <0>;
				compatible = "ti,tas5713";
				reg = <0x1b>;
				status = "okay";
			};
		};
	};`

It seems to me that the sound overlay from the respeaker hat is overriding the part for the hifiberry amp.

Pillar1989, 2018-03-13:
@CoolM8 It's impossible to merge them together.
- Them use different transmit format, hifiberry use i2s format(https://github.com/raspberrypi/linux/blob/rpi-4.9.y/sound/soc/bcm/hifiberry_amp.c#L53 ) , 4mic hat use dsp_a format.
- codec working clk is hard to choose

CoolM8, 2018-03-13:
What else then can I do that both devices are loaded? Why is one device tree overlay removing the other one to be detected by aplay -L respectively arecord -L?

Pillar1989, 2018-03-15:
Sorry, linux alsa driver framework are not allowed you to do this.

CoolM8, 2018-03-16:
So there is no other chance at all to bring those two cards to work together?

Pillar1989, 2018-03-16:
At least I can't do it.

closed 2018-03-21 by Pillar1989