respeaker/seeed-voicecard #44: scripts/basic/fixdep: 1: scripts/basic/fixdep: Syntax error: "(" unexpected

sonnyhcl, 2018-03-12:
I am trying transplanting the seeed-voicecard into PYNQ Board ( ARM+FPGA ). When I run make, something wrong shows "scripts/basic/fixdep: 1: scripts/basic/fixdep: Syntax error: "(" unexpected" and its whole log shows below.
What can I do to figure out what causes this bug?

```console
xilinx@pynq:~/seeed-voicecard$ make
make -C /lib/modules/4.9.0-xilinx/build M=/home/xilinx/seeed-voicecard modules
make[1]: Entering directory '/usr/src/linux-headers-4.9.0-xilinx'
  CC [M]  /home/xilinx/seeed-voicecard/wm8960.o
scripts/basic/fixdep: 1: scripts/basic/fixdep: Syntax error: "(" unexpected
scripts/Makefile.build:293: recipe for target '/home/xilinx/seeed-voicecard/wm8960.o' failed
make[2]: *** [/home/xilinx/seeed-voicecard/wm8960.o] Error 2
Makefile:1490: recipe for target '_module_/home/xilinx/seeed-voicecard' failed
make[1]: *** [_module_/home/xilinx/seeed-voicecard] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.9.0-xilinx'
Makefile:12: recipe for target 'all' failed
make: *** [all] Error 2
```

Pillar1989, 2018-03-13:
这是你内核源码的问题，你内核源码放在哪儿？ 我或许可以帮你看一下

sonnyhcl, 2018-03-13:
kernel source code link is here [linux-xlnx](https://github.com/Xilinx/linux-xlnx). 

Thank you.

Pillar1989, 2018-03-13:
你的Linux-kernel-header是在x86上面编译生成的，所以在你的板子上面运行scripts/basic/fixdep会出现这种架构不兼容的问题。

sonnyhcl, 2018-03-13:
我重新编译了一下fixdep，这个问题就过去了（虽然后面出现了其它的问题）。

谢谢你的指点！

Pillar1989, 2018-03-13:
什么问题？

sonnyhcl, 2018-03-13:
`script/mod/modpost`出现了同样的问题，估计需要linux source code整体重新编译一下了

closed 2018-03-14 by sonnyhcl