respeaker/seeed-voicecard #45: integrating DOA with kms(pocketsphinx)  with 4-mic array

IshantPundir, 2018-03-18:
hi so I am having  a really bad time for a week now trying to integrate pocketsphinx for KWS instead of snowboy(I don't want to use snow boy  in my project). and at the same time I also want to do DOA 
I have tried to hack and modify the codes in voice-engine. but nothing seems to works  for me can anyone please help me in modifying the codes so that I can use KWS(using pocketsphinx) and DOA at the same time 

I know there is a library for respeaker as well and it supports pocketsphinx but how can I use it to do both KWS and DOA at the same time with respeaker 4-mics  hat with SPI based connection

IshantPundir, 2018-03-18:
there are many different problems with integrating pocketsphinx instead of Snowboy and one of them is that 
for some reason pocketsphinx doesn't work with 4 channel input but for DOA I need 4 channels

xiongyihui, 2018-03-18:
@IshantPundir You can use the channel_picker element. The audio stream between elements may like:

```
source -> channel_picker -> pocketsphinx
  |
  V
 doa
```

Code may like:
```
src = Source(channels=4)
doa = DOA()
ch0 = ChannelPicker(channels=src.channels, pick=0)
kws = PocketsphinxKWS()

src.link(doa)
src.link(ch0)
ch0.link(kws)
```

IshantPundir, 2018-03-19:
hi thanks for the help it really helped  me a lot I am able to implement it successfully and thanks you a lot for this 

these are the codes I am using for kws

`
# -*- coding: utf-8 -*-

"""
Keyword spotting using snowboy
"""

import os
import sys
import threading
from pocketsphinx import *

if sys.version_info[0] < 3:
    import Queue as queue
else:
    import queue

from snowboy import snowboydetect

from voice_engine.element import Element

class KWS(Element):
    def __init__(self):
        super(KWS, self).__init__()

        self.config = Decoder.default_config()
        self.config.set_string('-hmm', 'cmusphinx-5prealpha-en-us-ptm-2.0')
#        self.config.set_string('-lm', '0576.lm')
        self.config.set_string('-dict', '0576.dic')
        self.config.set_string('-kws', 'corpus.list')

        self.decoder = Decoder(self.config)
        self.decoder.start_utt()

        self.queue = queue.Queue()
        self.done = False

        self.on_detected = None

    def put(self, data):
        self.queue.put(data)

    def start(self):
        self.done = False
        thread = threading.Thread(target=self.run)
        thread.daemon = True
        thread.start()

    def stop(self):
        self.done = True
    
    
    def run(self):
        while not self.done:
            buf = self.queue.get()
            if buf:
                self.decoder.process_raw(buf, False, False)

                if self.decoder.hyp() != None:
                    print ([(seg.word, seg.prob, seg.start_frame, seg.end_frame) for seg in self.decoder.seg()])
                    print ("Detected keyword, restarting search")
                            
                    if callable(self.on_detected):

                        self.on_detected('hey asper')
                    super(KWS, self).put(buf)
                    self.decoder.end_utt()
                    self.decoder.start_utt()

                            
    def set_callback(self, callback):
        self.on_detected = callback

`

when I run similar code for pocketsphinx KWS on my MacBook Air it  works fine with no problem but when running  the above script on raspberry pi it is not recognising anything 
my keyword is "hey asper "
inside my corpus.list  
I have tried threshold from ie-10 to ie-50

but it's not recognising my voice on pi it works once in a while but then stuck  can you guys please check the above codes and  suggest me any changes 

it would be really great as you guys can also add this feature in your latest updates

IshantPundir, 2018-03-20:
hi so I am totally new to pocketsphinx 

I was able to integrate my DOA with Pocketsphinx-KWS successfully all thanks to you 
but the problem I am facing right now is that I am unable to detect the keyword from my  Pocketsphinx-KWS 
my codes are

def run(self):
        a  = None
        while not self.done:
            buf = self.queue.get()
            
            self.decoder.process_raw(buf, False, False)
            if self.decoder.hyp() != None:
#                print 'decoder.hyp detected >>>>>>'
                print ([(seg.word, seg.prob, seg.start_frame, seg.end_frame) for seg in self.decoder.seg()])
#                print ("Detected keyword, restarting search")
                if callable(self.on_detected):
                    self.on_detected('hey asper')
            super(KWS, self).put(buf)
            self.decoder.end_utt()
            self.decoder.start_utt()

but it is not recognising the key word , I suspect that may be buf = self.queue.get() is the culprit 
as in the original codes for my it was like buf = stream.read(200).  where 200 is number of frames  is there a a way so that I can specify the number of framers using self.queue.get() command

xiongyihui, 2018-03-21:
You can specify the frame size by passing a `frame_size` parameter to the source element

```
class Source(Element):
    def __init__(self, rate=16000, frames_size=None, channels=None, device_index=None):
```

By default, the `frames_size` is rate / 100 (10 ms frames). To change it, for example

```
src = Source(frames_size=200, channels=4)
```

closed 2018-04-02 by Pillar1989