respeaker/seeed-voicecard #46: Cannot compile on ArchlinuxARM

bahamut657, 2018-03-24:
Hi 
Hi just cloned your repository under my RPIZero
I correct the install.sh to use pacman (and install the right packages for ArchlinuxARM) rather than apt
When DKMS goes to build the module make exit with this error (using GCC7.2.1 bundled in ArchlinuxARM)

```
/var/lib/dkms/seeed-voicecard/0.3/build/simple-card.c: In function 'asoc_simple_card_dai_link_of':
/var/lib/dkms/seeed-voicecard/0.3/build/simple-card.c:360:61: error: macro "asoc_simple_card_parse_clk_cpu" requires 4 arguments, but only 3 given
  ret = asoc_simple_card_parse_clk_cpu(cpu, dai_link, cpu_dai);
                                                             ^
/var/lib/dkms/seeed-voicecard/0.3/build/simple-card.c:360:8: error: 'asoc_simple_card_parse_clk_cpu' undeclared (first use in this function); did you mean 'asoc_simple_card_parse_clk'?
  ret = asoc_simple_card_parse_clk_cpu(cpu, dai_link, cpu_dai);
        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
        asoc_simple_card_parse_clk
/var/lib/dkms/seeed-voicecard/0.3/build/simple-card.c:360:8: note: each undeclared identifier is reported only once for each function it appears in
/var/lib/dkms/seeed-voicecard/0.3/build/simple-card.c:364:67: error: macro "asoc_simple_card_parse_clk_codec" requires 4 arguments, but only 3 given
  ret = asoc_simple_card_parse_clk_codec(codec, dai_link, codec_dai);
                                                                   ^
/var/lib/dkms/seeed-voicecard/0.3/build/simple-card.c:364:8: error: 'asoc_simple_card_parse_clk_codec' undeclared (first use in this function); did you mean 'asoc_simple_card_parse_clk_cpu'?
  ret = asoc_simple_card_parse_clk_codec(codec, dai_link, codec_dai);
        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
        asoc_simple_card_parse_clk_cpu
make[1]: *** [scripts/Makefile.build:329: /var/lib/dkms/seeed-voicecard/0.3/build/simple-card.o] Error 1
make: *** [Makefile:1519: _module_/var/lib/dkms/seeed-voicecard/0.3/build] Error 2
make: Leaving directory '/usr/lib/modules/4.14.29-1-ARCH/build'
```

Pillar1989, 2018-03-26:
#38 
let's open a vote, If 50+ people need this feature, we'll develop it.

closed 2018-03-26 by Pillar1989

1158114251, 2018-03-28:
we are all need

1158114251, 2018-03-28:
Strictly speaking, it is your sales website promotion, your 4Mic support zero w, so. . . You need to develop or tell us how to do it.

1158114251, 2018-03-30:
@bahamut657 In fact, as we face the problem, but the environment is different, I is the default system in raspberry pie Debian, update the system after installation problems can be solved according to the driver, I https://github.com/respeaker/seeed-voicecard/issues/48, and now the seeed team has repaired the bug