respeaker/seeed-voicecard #47: Can't find seeed-4mic drivers on RPI3 Raspbian Stretch

melzoghbi, 2018-03-27:
Hi,

I am running Raspbian Stretch on RPI3. After i installed the drivers and rebooted. i can't see the 4mic array drivers. Below is what i see when i list my audio drivers:

[image]

I have installed google assistant sdk previously and that's why you see voice drivers. I don't know what would be the ideal way to reset these drivers with respeaker ones.

Thoughts on how to resolve this issue?

Thanks.

turmary, 2018-03-28:
please paste result of command 'cat /var/log/seeed-voicecard.log' here.

melzoghbi, 2018-03-28:
here you go:

+ dtparam i2c_arm=on
+ modprobe i2c-dev
+ dtparam spi=on
+ sleep 1
++ i2cdetect -y 1 0x1a 0x1a
++ grep 1a
++ awk '{print $2}'
+ is_1a=
++ i2cdetect -y 1 0x35 0x35
++ grep 35
++ awk '{print $2}'
+ is_35=
++ i2cdetect -y 1 0x3b 0x3b
++ grep 3b
++ awk '{print $2}'
+ is_3b=3b
+ '[' x == x1a ']'
+ '[' x3b == x3b ']'
+ '[' x == x ']'
+ echo 'install 4mic'
install 4mic
+ dtoverlay seeed-4mic-voicecard
* Failed to apply overlay '5_seeed-4mic-voicecard' (kernel)
+ sleep 1
++ arecord -l
++ grep seeed4micvoicec
++ awk '{print $2}'
++ sed s/://
+ hw=
+ cp /etc/voicecard/asound_4mic.conf /etc/asound.conf
+ echo 'get slavepcm line number'
get slavepcm line number
++ grep -n ac108-slavepcm /etc/asound.conf
++ awk '{print $1}'
++ sed s/://
+ line=28
+ echo 'delete 28 slavepcm hw number'
delete 28 slavepcm hw number
+ sed -i -e 28d /etc/asound.conf
+ echo 'insert slavepcm hw:,0'
insert slavepcm hw:,0
+ sed -i '28i slave.pcm "hw:,0"' /etc/asound.conf
+ cp /etc/voicecard/ac108_asound.state /var/lib/alsa/asound.state
+ '[' x3b == x3b ']'
+ '[' x == x35 ']'
+ alsactl restore
+ amixer cset numid=3 1
numid=3,iface=MIXER,name='PCM Playback Route'
  ; type=INTEGER,access=rw------,values=1,min=0,max=2,step=0
  : values=1

turmary, 2018-03-29:
you need to remove 'dtoverlay=googlevoicehat-soundcard' in /boot/config.txt, due to they are conflict, maybe.

melzoghbi, 2018-03-29:
Please ignore the previous output since i started from a fresh raspbian OS and here is the current output:

[image]

turmary, 2018-03-30:
Hi melzoghbi,
please try install the new [version](https://github.com/respeaker/seeed-voicecard/tree/031bbb3a1a19c5c0bcdf2b371439bd1aeca41158).

turmary, 2018-03-30:
Please don't run rpi-update or else driver will fail due to kernel-headers for 4.14.31 unavailable.

chhsiao90, 2018-03-30:
Same problem and same log here.
Is there anything I can do? Thanks!

```
Raspberry Pi: RASPBERRY PI 2 MODEL B
seeed-voicecard: 86329f0719c61405a17a0e5c6c04806d234ba086
Kernel: Linux raspberrypi 4.14.30-v7+ #1102 SMP Mon Mar 26 16:45:49 BST 2018 armv7l GNU/Linux
```

chhsiao90, 2018-03-30:
My problem solved, it looks like that I had config conflict in /boot/config.txt

melzoghbi, 2018-03-30:
Hi @turmary 

I installed the latest version but still i am having the same issue. 

Here is the output of -uname as per your request:

[image]

Pillar1989, 2018-04-02:
@melzoghbi  Please close all of your issues.

melzoghbi, 2018-04-02:
ok.

closed 2018-04-02 by melzoghbi