respeaker/seeed-voicecard #48: driver does not build on stretch lite: macro "asoc_simple_card_parse_clk_codec" requires 4 arguments, but only 3 given

mvdlinde, 2018-03-28:
Its exact the same issue as described in #46, but it fails on Raspberry Zero W with (up2date) raspbian stretch lite.

cat /proc/device-tree/model
Raspberry Pi Zero W Rev 1.1

cat /etc/debian_version 
9.4

[make.log](https://github.com/respeaker/seeed-voicecard/files/1856580/make.log)

mvdlinde, 2018-03-28:
As the recent kernel version moved forward to 4.14.30 this patch fixes the obove mentioned build failure:
diff --git a/simple-card.c b/simple-card.c
index 0730216..0a8333c 100644
--- a/simple-card.c
+++ b/simple-card.c
@@ -361,11 +361,11 @@ static int asoc_simple_card_dai_link_of(struct device_node *node,
        if (ret < 0)
                goto dai_link_of_err;
 
-       ret = asoc_simple_card_parse_clk_cpu(cpu, dai_link, cpu_dai);
+       ret = asoc_simple_card_parse_clk_cpu(dev, cpu, dai_link, cpu_dai);
        if (ret < 0)
                goto dai_link_of_err;
 
-       ret = asoc_simple_card_parse_clk_codec(codec, dai_link, codec_dai);
+       ret = asoc_simple_card_parse_clk_codec(dev, codec, dai_link, codec_dai);
        if (ret < 0)
                goto dai_link_of_err;
But ther is more work needed to smoothly integrate on all kernel versions... I would say, this is up to you... ;)

turmary, 2018-03-29:
Hi mvdlinde,
I can't find raspberrypi image with 4.14 kernel in the raspberrypi.org.
And in the image with 4.9 kernel, I can't upgrade to 4.14 kernel version too.

feiticeir0, 2018-03-29:
One workaround that I did (before I found out about this patch) is to compile using the install branch and not the master - even with the latest kernel updated via rpi-update...
git clone -b install https://github.com/respeaker/seeed-voicecard.git

mvdlinde, 2018-03-29:
so this should build on all kernel versions:
```
diff --git a/simple-card.c b/simple-card.c
index 0730216..4e7b07c 100644
--- a/simple-card.c
+++ b/simple-card.c
@@ -22,6 +22,9 @@
 #include <sound/soc-dai.h>
 #include <sound/soc.h>
 
+#include <linux/version.h>
+
+
 /*
  * single codec:
  *     0 - allow multi codec
@@ -361,11 +364,19 @@ static int asoc_simple_card_dai_link_of(struct device_node *node,
        if (ret < 0)
                goto dai_link_of_err;
 
+#if LINUX_VERSION_CODE <= KERNEL_VERSION(4,10,0)
        ret = asoc_simple_card_parse_clk_cpu(cpu, dai_link, cpu_dai);
+#else
+       ret = asoc_simple_card_parse_clk_cpu(dev, cpu, dai_link, cpu_dai);
+#endif
        if (ret < 0)
                goto dai_link_of_err;
 
+#if LINUX_VERSION_CODE <= KERNEL_VERSION(4,10,0)
        ret = asoc_simple_card_parse_clk_codec(codec, dai_link, codec_dai);
+#else
+       ret = asoc_simple_card_parse_clk_codec(dev, codec, dai_link, codec_dai);
+#endif
        if (ret < 0)
                goto dai_link_of_err;
```

mvdlinde, 2018-03-29:
at least on actual kernel 4.14.30 this builds and the card works as expected.

feiticeir0, 2018-03-29:
Working with latest kernel ! Thank you for the patch !

 Seeedstudio should include this in the wiki, because people will follow the wiki and it will not build..

turmary, 2018-03-30:
Hi mvdlinde,
thanks, I will merge the patch soon.

closed 2018-04-08 by Pillar1989