respeaker/seeed-voicecard #49: 树莓派zero w 上面安装4mic

1158114251, 2018-03-28:
我在树莓派zero w上面插入4mic 安装4mic驱动，显示安装成功，但是 arecord -L的时候显示
default
playback
dmixed
ac108
multiapps
就没有声卡这些信息。像default:CARD=seeed4micvoicec 这些东西，然后录音也没有设备

1158114251, 2018-03-28:
安装信息
DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area......
make KERNELRELEASE=4.14.30+ -C /lib/modules/4.14.30+/build M=/var/lib/dkms/seeed-voicecard/0.3/build....................(bad exit status: 2)
Error! Bad return status for module build on kernel: 4.14.30+ (armv6l)
Consult /var/lib/dkms/seeed-voicecard/0.3/build/make.log for more information.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area......
make KERNELRELEASE=4.14.30-v7+ -C /lib/modules/4.14.30-v7+/build M=/var/lib/dkms/seeed-voicecard/0.3/build........^C
root@raspberrypi:/home/pi/seeed-voicecard# vim /var/lib/dkms/seeed-voicecard/0.3/build/make.log 

DKMS make.log for seeed-voicecard-0.3 for kernel 4.14.30-v7+ (armv6l)
Wed 28 Mar 17:08:49 UTC 2018
make: Entering directory '/usr/src/linux-headers-4.14.30-v7+'
  AR      /var/lib/dkms/seeed-voicecard/0.3/build/built-in.o
  CC [M]  /var/lib/dkms/seeed-voicecard/0.3/build/wm8960.o
make[1]: *** wait: No child processes.  Stop.
make[1]: *** Waiting for unfinished jobs....
make[1]: *** wait: No child processes.  Stop.
Makefile:1519: recipe for target '_module_/var/lib/dkms/seeed-voicecard/0.3/build' failed
make: *** [_module_/var/lib/dkms/seeed-voicecard/0.3/build] Error 2

1158114251, 2018-03-28:
pi@raspberrypi:~/seeed-voicecard1 $ vim /var/lib/dkms/seeed-voicecard/0.3/build/make.log

  1 DKMS make.log for seeed-voicecard-0.3 for kernel 4.14.30-v7+ (armv6l)
  2 Wed 28 Mar 18:14:45 UTC 2018
  5   CC [M]  /var/lib/dkms/seeed-voicecard/0.3/build/wm8960.o
  6   LD [M]  /var/lib/dkms/seeed-voicecard/0.3/build/snd-soc-wm8960.o
  7   CC [M]  /var/lib/dkms/seeed-voicecard/0.3/build/ac108.o
  8   CC [M]  /var/lib/dkms/seeed-voicecard/0.3/build/ac101.o
  9   LD [M]  /var/lib/dkms/seeed-voicecard/0.3/build/snd-soc-ac108.o
 10   CC [M]  /var/lib/dkms/seeed-voicecard/0.3/build/simple-card.o
 11 /var/lib/dkms/seeed-voicecard/0.3/build/simple-card.c: In function ‘asoc_simple_card_dai_link_of’:
 12 /var/lib/dkms/seeed-voicecard/0.3/build/simple-card.c:364:61: error: macro "asoc_simple_card_parse_clk_cpu" requir    es 4 arguments, but only 3 given
 13   ret = asoc_simple_card_parse_clk_cpu(cpu, dai_link, cpu_dai);
 14                                                              ^
 15 /var/lib/dkms/seeed-voicecard/0.3/build/simple-card.c:364:8: error: ‘asoc_simple_card_parse_clk_cpu’ undeclared (f    irst use in this function)
 16   ret = asoc_simple_card_parse_clk_cpu(cpu, dai_link, cpu_dai);
 17         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
 18 /var/lib/dkms/seeed-voicecard/0.3/build/simple-card.c:364:8: note: each undeclared identifier is reported only onc    e for each function it appears in
 19 /var/lib/dkms/seeed-voicecard/0.3/build/simple-card.c:368:67: error: macro "asoc_simple_card_parse_clk_codec" requ    ires 4 arguments, but only 3 given
 20   ret = asoc_simple_card_parse_clk_codec(codec, dai_link, codec_dai);
 21                                                                    ^
 22 /var/lib/dkms/seeed-voicecard/0.3/build/simple-card.c:368:8: error: ‘asoc_simple_card_parse_clk_codec’ undeclared     (first use in this function)
 23   ret = asoc_simple_card_parse_clk_codec(codec, dai_link, codec_dai);
 24         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
 25 scripts/Makefile.build:328: recipe for target '/var/lib/dkms/seeed-voicecard/0.3/build/simple-card.o' failed
 26 make[1]: *** [/var/lib/dkms/seeed-voicecard/0.3/build/simple-card.o] Error 1
 27 Makefile:1519: recipe for target '_module_/var/lib/dkms/seeed-voicecard/0.3/build' failed
 28 make: *** [_module_/var/lib/dkms/seeed-voicecard/0.3/build] Error 2
 29 make: Leaving directory '/usr/src/linux-headers-4.14.30-v7+'

turmary, 2018-03-29:
参考 #48 修改代码再安装一次。

turmary, 2018-03-29:
Duplicate of #48

closed 2018-03-30 by Pillar1989