respeaker/seeed-voicecard #50: arecord doesn't show available devices/drivers

melzoghbi, 2018-03-28:
Hi, 

I am running raspbian stretch on RPI3. After install seeed driver for my respeaker 4 mic array, i am not able to use audicty because driver issues.

I am not able to list my drivers, here is the return when i execute "arecord -L" in terminal window:

[image]

do you know what i need to do to fix this issue ? so i can see the 4mic array driver as it is supposed to be.

Thanks.

turmary, 2018-03-29:
please paste result of command 'cat /var/log/seeed-voicecard.log' here again.

melzoghbi, 2018-03-29:
Here you go:

[image]