respeaker/seeed-voicecard #51: Doesn't compile in Raspberry PI zero W (or any other) with latest versions of Rasbian Stretch

feiticeir0, 2018-03-29:
Following the instructions on the wiki, I haven't been able to add support for the ReSpeaker 2Mic because it doesn't compile... The module is never created 

Here are the logs when issuing the command: sudo ./install.sh 2mic related to the module building:

`Building module:
cleaning build area.....
make -j1 KERNELRELEASE=4.14.30+ -C /lib/modules/4.14.30+/build M=/var/lib/dkms/seeed-voicecard/0.3/build............................(bad exit status: 2)
Error! Bad return status for module build on kernel: 4.14.30+ (armv6l)
Consult /var/lib/dkms/seeed-voicecard/0.3/build/make.log for more information.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area.....
make -j1 KERNELRELEASE=4.14.30-v7+ -C /lib/modules/4.14.30-v7+/build M=/var/lib/dkms/seeed-voicecard/0.3/build.......................(bad exit status: 2)
Error! Bad return status for module build on kernel: 4.14.30-v7+ (armv6l)
Consult /var/lib/dkms/seeed-voicecard/0.3/build/make.log for more information.
Created symlink /etc/systemd/system/sysinit.target.wants/seeed-voicecard.service → /lib/systemd/system/seeed-voicecard.service.
------------------------------------------------------
Please reboot your raspberry pi to apply all settings
Enjoy!
`

Reviewing the logs:
`DKMS make.log for seeed-voicecard-0.3 for kernel 4.14.30-v7+ (armv6l)
Wed Mar 28 23:12:19 UTC 2018
make: Entering directory '/usr/src/linux-headers-4.14.30-v7+'
  AR      /var/lib/dkms/seeed-voicecard/0.3/build/built-in.o
  CC [M]  /var/lib/dkms/seeed-voicecard/0.3/build/wm8960.o
  LD [M]  /var/lib/dkms/seeed-voicecard/0.3/build/snd-soc-wm8960.o
  CC [M]  /var/lib/dkms/seeed-voicecard/0.3/build/ac108.o
  CC [M]  /var/lib/dkms/seeed-voicecard/0.3/build/ac101.o
  LD [M]  /var/lib/dkms/seeed-voicecard/0.3/build/snd-soc-ac108.o
  CC [M]  /var/lib/dkms/seeed-voicecard/0.3/build/simple-card.o
/var/lib/dkms/seeed-voicecard/0.3/build/simple-card.c: In function <E2><80><98>asoc_simple_card_dai_link_of<E2><80><99>:
/var/lib/dkms/seeed-voicecard/0.3/build/simple-card.c:364:61: error: macro "asoc_simple_card_parse_clk_cpu" requires 4 arguments, but only 3 given
  ret = asoc_simple_card_parse_clk_cpu(cpu, dai_link, cpu_dai);
                                                             ^
/var/lib/dkms/seeed-voicecard/0.3/build/simple-card.c:364:8: error: <E2><80><98>asoc_simple_card_parse_clk_cpu<E2><80><99> undeclared (first use in this function)
  ret = asoc_simple_card_parse_clk_cpu(cpu, dai_link, cpu_dai);
        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/seeed-voicecard/0.3/build/simple-card.c:364:8: note: each undeclared identifier is reported only once for each function it appears in
/var/lib/dkms/seeed-voicecard/0.3/build/simple-card.c:368:67: error: macro "asoc_simple_card_parse_clk_codec" requires 4 arguments, but only 3 given
  ret = asoc_simple_card_parse_clk_codec(codec, dai_link, codec_dai);
                                                                   ^
/var/lib/dkms/seeed-voicecard/0.3/build/simple-card.c:368:8: error: <E2><80><98>asoc_simple_card_parse_clk_codec<E2><80><99> undeclared (first use in this function)
  ret = asoc_simple_card_parse_clk_codec(codec, dai_link, codec_dai);
        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
scripts/Makefile.build:328: recipe for target '/var/lib/dkms/seeed-voicecard/0.3/build/simple-card.o' failed
make[1]: *** [/var/lib/dkms/seeed-voicecard/0.3/build/simple-card.o] Error 1
Makefile:1519: recipe for target '_module_/var/lib/dkms/seeed-voicecard/0.3/build' failed
make: *** [_module_/var/lib/dkms/seeed-voicecard/0.3/build] Error 2
make: Leaving directory '/usr/src/linux-headers-4.14.30-v7+'
`

For two days I've been trying different versions of Stretch, even jessie, but still no luck...

feiticeir0, 2018-03-29:
After a bit of struggling, I was able to compile the code, but using the install branch and not the master..

turmary, 2018-03-29:
Duplicate of #48

feiticeir0, 2018-03-29:
Yes it is ! I've searched for issues, but didn't found any... Going to close it !

closed 2018-03-29 by feiticeir0

Kuldeepwadaskar, 2020-02-06:
> After a bit of struggling, I was able to compile the code, but using the install branch and not the master..

can you explain how did you compiled the code? i am facing similar problem
thanks!

feiticeir0, 2020-02-13:
> 
> 
> > After a bit of struggling, I was able to compile the code, but using the install branch and not the master..
> 
> can you explain how did you compiled the code? i am facing similar problem
> thanks!

I've used the instruction on the wiki to compile the code:
http://wiki.seeedstudio.com/ReSpeaker_2_Mics_Pi_HAT/