respeaker/seeed-voicecard #55: 不知道是不是新版驱动问题，我的4录音一直在阻塞

1158114251, 2018-04-01:
我的录音代码如下：
`

"""PyAudio example: Record a few seconds of audio and save to a WAVE file."""

import pyaudio
import wave

CHUNK = 1024
FORMAT = pyaudio.paInt16
CHANNELS = 2
RATE = 44100
RECORD_SECONDS = 5
WAVE_OUTPUT_FILENAME = "output.wav"

p = pyaudio.PyAudio()

stream = p.open(format=FORMAT,
                channels=CHANNELS,
                rate=RATE,
                input=True,
                frames_per_buffer=CHUNK)

print("* recording")

frames = []

for i in range(0, int(RATE / CHUNK * RECORD_SECONDS)):
    data = stream.read(CHUNK)
    frames.append(data)

print("* done recording")

stream.stop_stream()
stream.close()
p.terminate()

wf = wave.open(WAVE_OUTPUT_FILENAME, 'wb')
wf.setnchannels(CHANNELS)
wf.setsampwidth(p.get_sample_size(FORMAT))
wf.setframerate(RATE)
wf.writeframes(b''.join(frames))
wf.close()

`

pi@raspberrypi:~ $ python3 1.py 
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.front
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround40
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround41
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround50
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround51
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround71
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4996:(snd_config_expand) Args evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4996:(snd_config_expand) Args evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa
ALSA lib pcm_dmix.c:990:(snd_pcm_dmix_open) The dmix plugin supports only playback stream
ALSA lib pcm_dsnoop.c:556:(snd_pcm_dsnoop_open) The dsnoop plugin supports only capture stream
ALSA lib pcm_dsnoop.c:556:(snd_pcm_dsnoop_open) The dsnoop plugin supports only capture stream
connect(2) call to /tmp/jack-1000/default/jack_0 failed (err=No such file or directory)
attempt to connect to server failed
* recording

cs301cs301, 2018-04-01:
你好。你从哪里得到2.0版本？我也许有与SOPARE项目（声音模式识别）相同的问题

1158114251, 2018-04-02:
2.0就在仓库的发行版里

Pillar1989, 2018-04-02:
@cs301cs301  SOPARE项目对中文的识别，效果好吗？

1158114251, 2018-04-02:
@Pillar1989 大佬如何看待这个问题，是不是新版驱动，更新内核后引起的，因为出现在系统调用上

Pillar1989, 2018-04-02:
@turmary 帮忙看一下

turmary, 2018-04-02:
@1158114251  我尝试修复了这个问题，用你的脚本测试可以，请更新到新版本试下。

1158114251, 2018-04-02:
好的，晚上我试试，你们的效率一惯的高啊！

1158114251, 2018-04-02:
哈哈哈，是中断导致死锁？

cs301cs301, 2018-04-02:
@Pillar1989 Sry no idea about that. But i think the language does not matter. The program compares the recived command with the learned commands. https://www.bishoph.org Better ask the developer for details.
My problem still persists, the 4MicArray does not work with SOPARE, no matter which branch i try.
@ Pillar1989抱歉，不知道这件事。但我认为这种语言并不重要。该程序将收到的命令与学习到的命令进行比较。 https://www.bishoph.org更好地向开发者询问详细信息。
我的问题仍然存在，4MicArray不适用于SOPARE，无论我尝试哪个分支。

cs301cs301, 2018-04-06:
My problem with 4mic-array and 2mic-hat is, that sopare did not use the mic-device. pyaudio uses the default input device: „DEBUG:sopare.audio_factory:maxInputChannels: 128
DEBUG:sopare.audio_factory:name: default“
But there has to be as name "seeed-2mic-voicecard" or the 4mic array. And input channels are only 2, or 4 for the array, not 128?!
I did not change anything on the alsa-config.
I can record without a problem with arecord but sopare has many problems.  

我的问题与4微阵列和2米帽子是，该sopare没有使用麦克风设备。 pyaudio使用默认输入设备：“DEBUG：sopare.audio_factory：maxInputChannels：128
DEBUG：sopare.audio_factory：name：default“
但必须有名称“seeed-2mic-voicecard”或4mic数组。 而输入通道只有2或4个阵列，而不是128？！
我没有更改alsa-config上的任何内容。
我可以记录没有与arecord问题，但sopare有很多问题。

cs301cs301, 2018-04-06:
@Pillar1989 你有没有尝试过自己的“sopare”？ 如果它适用于4微米或2微米的你，你能告诉我alsa配置吗？

cs301cs301, 2018-04-06:
奇怪的事情，如果我从主回购安装驱动它不起作用。 如果我删除主驱动程序，然后安装wm8960_debug它显示正确与2输入通道。 如果我做了一个干净的安装，系统新的驱动程序wm8960_debug，它将无法正常工作。 我变得怪异...

Pillar1989, 2018-04-08:
@cs301cs301  I'll test 4mic hat with sopare tomorrow

Pillar1989, 2018-04-09:
@cs301cs301 Please keep seeed-voicecard's asound.conf, donnt modify any line.
and change https://github.com/bishoph/sopare/blob/master/sopare/audio_factory.py#L41 channels = 1 to 4. 
then test 4mic hat again.

cs301cs301, 2018-04-09:
Thanks for testing, but that was the first thing I did, and i did never change any file from the driver! The problem is that sopare / pyaudio has a high threshold of >2000 to 24000 with the arrays, normal would be a value around 100. And if I set the recognized threshold in the config to this high value, then sopare does not recognize a word. Next, when I lower the threshold, Sopare does not recognize anything or it has detected background noise as a language. Also, the script does not stop at endless recording, but aborts with error because the threshold is too low. If I lower the gain in alsamixer, then I get a lower threshold, but neither Sopare nor arecord can record anything, there is nothing to hear ... I tested with an unemplified low cost mic, and got a treshold of 120. But the long range detection of the mic was crap. Therefore i did choose the mic array. I'm really out of ideas.

Can you explain, why pyaudio detects 128 in-/out channels?

Pillar1989, 2018-04-09:
You can use `python test/test_audio.py` test audio first. I did not In-depth look at sopare.py  file.
Maybe you can ask for help from sopare project?

cs301cs301, 2018-04-09:
i did test_audio.py first, and set the parameter in config.txt...  the sopare developer ist out of ideas, too...
is there a way to downmix the channels to mono? he mentioned something, because of high cpu and memory usage

big thx for help, but i will stop here with the arrays. :(

closed 2018-04-10 by Pillar1989