respeaker/seeed-voicecard #62: "Illegal Instruction" in env/bin/google-assistant-demo on Raspberry-Zero

gerardwr, 2018-04-10:
Hi,

I installed:
- Respeaker 2-mic HAT on my Raspberry ZeroW
- Latest Raspbian Stretch
- Respeaker software and Google Assistant installed acc. http://wiki.seeedstudio.com/ReSpeaker_2_Mics_Pi_HAT/

The installation of the Respeaker software and Google Assistant software worked fine, no errors.

But his results in an "illegal instruction" error:
```
(env) pi@raspberrypi:~/Google-Assistant-SDK $ env/bin/google-assistant-demo --device_model_id gwr-light
Illegal instruction
(env) pi@raspberrypi:~/Google-Assistant-SDK $
```

While this runs fine, voice is recorded and played via Respeaker:
```
(env) pi@raspberrypi:~/Google-Assistant-SDK $ env/bin/googlesamples-assistant-audiotest 
INFO:root:Starting audio test.
INFO:root:Recording samples.
INFO:root:Finished recording.
INFO:root:Playing back samples.
INFO:root:Finished playback.
INFO:root:audio test completed.
(env) pi@raspberrypi:~/Google-Assistant-SDK $
```

The following commands als result in an "illegal instruction":
```
(env) pi@raspberrypi:~/Google-Assistant-SDK $ env/bin/googlesamples-assistant-hotword --device_model_id gwr-light
Illegal instruction
(env) pi@raspberrypi:~/Google-Assistant-SDK $ env/bin/googlesamples-assistant-pushtotalk --device_model_id gwr-light
Illegal instruction
(env) pi@raspberrypi:~/Google-Assistant-SDK $
```

Is the software installation instruction applicable for the Raspbbery ZeroW?

Should the "hotword" example work OK on het ZeroW?

Any suggestions to what's my problem might be?

Pillar1989, 2018-04-11:
@gerardwr https://github.com/google/aiyprojects-raspbian/issues/177

gerardwr, 2018-04-11:
@Pillar1989 

Acc. Your link Zero is not supported for the latest GA software.

If that would have been clearly on the READ.ME it would have saved me an hour of my live.

On another Zero I had the GA version of feb18 working for the textinput and pushtotalk examples working fine without the illegal instruction. Only the hotword example resulteerde in an illegal instruction.

Thx for your response.

closed 2018-04-11 by gerardwr