respeaker/seeed-voicecard #65: Volume Respeaker 2-mic HAT too low with Google Assistant

gerardwr, 2018-04-16:
Hi,

Not sure this is a Respeaker issue.

Installed a Respeaker 2-mic HAT on my Raspberry Pi 3+, and connected a speaker to the JST connector.

The Respeaker mic and speaker work fine, with a high volume on the speaker:
- record audio with "arecord" and then re-play with "aplay"
- Alexa using the avs-device-sdk from https://github.com/alexa/avs-device-sdk
- can increase/decrease the volume level with "alsamixer". It's now set to 50%

So far, so good, BUT:
Using Google Assistant from https://github.com/googlesamples/assistant-sdk-python the volume of the voice response is very low, MUCH lower than using "aplay" or Alexa. Increasing the volume using alsamixer makes no difference in the volume.

It seems as if the sound-output from Google Assistant is not using the alsamixer settings.

How come that the Respeaker volume is too low using Google Assitant, while it is fine using "apaly" nad Alexa?

KillingJacky, 2018-04-17:
@gerardwr Have you tried tune the volume on the Google Assistant's cloud side? I think the default volume of the audio stream returned by Google Assistant is 50%, I met the low volume issue as well, I solved it with speaking command: "[wake word] set your volume to 100"

gerardwr, 2018-04-17:
@KillingJacky 
Thx, I re-checked to be sure, it is set to MAX=100%

gerardwr, 2018-04-17:
@xiongyihui 
Thx.
I agree that the problem seems to be ig GA, conidering the volume is fine using Alexa and aplay.

Will raise an issue in the GA SDK github.

gerardwr, 2018-04-18:
I solved my issue.

The problem was caused by differences between the files ~/.asound and /etc/asoundrc.

After I delete the file ~/.asoundrc and rebooted the volume of the Google Assistant was on the same high level as with With Alexa en aplay.

FYI I list the contents of both files:
```(env) pi@raspberrypi3:~ $ cat .asoundrc
pcm.!default {
 type asym
 capture.pcm "mic"
 playback.pcm "speaker"
}
pcm.mic {
 type plug
 slave {
 pcm "hw:1,0"
 }
}
pcm.speaker {
 type plug
 slave {
 pcm "hw:1,0"
 }
}

(env) pi@raspberrypi3:~ $ cat /etc/asound.conf 
# The IPC key of dmix or dsnoop plugin must be unique
# If 555555 or 666666 is used by other processes, use another one

pcm.!default {
    type asym
    playback.pcm "playback"
    capture.pcm "capture"
}

pcm.playback {
    type plug
    slave.pcm "dmixed"
}

pcm.capture {
    type plug
    slave.pcm "array"
}

pcm.dmixed {
    type dmix
    slave.pcm "hw:1,0"
    ipc_key 555555 
}

pcm.array {
    type dsnoop
    slave {
        pcm "hw:1,0"
        channels 2
    }
    ipc_key 666666
}

(env) pi@raspberrypi3:~ $ 
```

closed 2018-04-24 by Pillar1989