respeaker/seeed-voicecard #66: Respeaker 2-mic array not working after rpi kernel update

jay7583, 2018-04-16:
Hi,
I am using Raspbian strech lite.when i tried to install after rpi-update,i am getting this error.
Error! echo
Your kernel headers for kernel 4.14.33+ cannot be found at
/lib/modules/4.14.33+/build or /lib/modules/4.14.33+/source.
Error! echo
Your kernel headers for kernel 4.14.33-v7+ cannot be found at
/lib/modules/4.14.33-v7+/build or /lib/modules/4.14.33-v7+/source.

Is there any patch for this.

Pillar1989, 2018-04-17:
@jay7583 Please note:
**Your kernel headers for kernel 4.14.34+ cannot be found at
/lib/modules/4.14.34+/build or /lib/modules/4.14.34+/source**
Which library depend on  rpi-update ?

psixilambda, 2018-04-18:
this helped me: 

https://github.com/respeaker/seeed-voicecard/issues/40#issuecomment-373235991

OK，the lastest driver changed kernel module snd-soc-simple-card.ko.So, please run commandssudo rm -rf `find /lib/modules/ -name "*simple-card.ko*"` sudo apt-get install --reinstall raspberrypi-kernel to recover it, then reboot.
--

jay7583, 2018-04-19:
Thanks guys,I solved it by updating kernel to 4.14.30 version.

closed 2018-04-19 by jay7583