respeaker/seeed-voicecard #67: ALSAMIXER Optimal Settings

ash-lat, 2018-04-20:
Hi 

Could anyone help with alsamixer's optimal settings to get the most out of 2-mic HAT.

Or provide details of what each settings/bar mean (and has impact on recording quality and intensity) so I could set it up and share optimal settings here.

(I tried to google those acronyms "ALC, ALC Target, ALC Decay and more" from alsamixer but I was not getting optimal results after changing them according to the definations provided)

Thanks in advance

Pillar1989, 2018-04-24:
@ash-lat  alsamixer's settings is works best now, what's do you want do?

ash-lat, 2018-04-24:
I see

I wanted to improve its performance , as it was not performing well for my space and during noise from Fan or TV. So if you could provide the details of what each settings does, so maybe I could optimise it for my space.

Also I tried to save the alsamixer settings by - 
**$ sudo alsactl --file=asound.state store**
but after reboot it turn back to previous settings

And how would you rate the performance of 2-mic Vs 4-mic array in terms of range (if you may have tried them together)?

Lastly I saw ReSpeaker Mic Array v2.0 on seeed studio site, any clue would it shall be released?

Pillar1989, 2018-04-25:
[image]
Our readme file  write very clear. 

About ReSpeaker Mic Array v2.0, Please pay attention to our website

ash-lat, 2018-04-25:
Thanks Pillar1989
I was following wiki guide - http://wiki.seeedstudio.com/ReSpeaker_2_Mics_Pi_HAT/

Any info regarding the different states in alsamixer?
Also is 4-Mics **Linear** Array Kit available, I couldnt find it on seeed studio site?

Pillar1989, 2018-04-25:
About 4-Mics Linear Array Kit ,  Please pay attention to our website

closed 2018-05-03 by Pillar1989

cogmeta, 2018-10-25:
Can you help with alsamixer settings for respeaker v2. Also, is there c++ implementation of voice engine library?