respeaker/seeed-voicecard #68: Using Neopixel RGB 24 LED ring with Respeaker 2mic hat?

jabulonboaz, 2018-04-21:
Hello is it possible to connect a Neopixel 24 RGB LED ring with Respeaker 2mic hat?

Normally a neopixel ring control pin can be connected to one of the Pi PWM GPIOs, typocaly GPIO13.
Where can i connect the  neopixel ring when i have the respeaker 2mic hat on the Pi3 ?

A similar question has been asked on the raspberrypi forum bit there is no answer so far:
https://www.raspberrypi.org/forums/viewtopic.php?t=207684

qunanan, 2018-04-24:
Hi there,

Please check this:
[https://pinout.xyz/pinout/unicorn_hat](https://pinout.xyz/pinout/unicorn_hat)
and this:
[image]

jabulonboaz, 2018-04-24:
@qunanan i have tried pin 12 but the LEDs do not light in the neopixel ring. I believe i have not asked you my question correctly.
I have tried also pin BCM 10 that is the Respeaker LEDs data pin but only one LED of the neopixel ring lights up.
What i want to do is drive the LEDs of the neopoxel ring to light up the same way the Respeaker LEDs light up.
I have build this raspberry pi 3 with Alexa and Google Assistant SDKs and the Respeaker LEDs light up correctly when each assistant is triggered. I have done this using the pixel.py script for Google assitant and another script for Alexa and the Respeker LEDs are driven via the Respeaker SPI interface. 
The question is how can i drive the neopixel LEDs with the Respeaker SPI interface or BCM 10 pin or any other pin on the Respeaker board so they light up with the same pattern the Respeaker LEDs light up?

xiongyihui, 2018-04-26:
@jabulonboaz The ReSpeaker LEDs are driven SPI interface, but the neopixel is not. You need to change the code to light up the neopixel ring.
https://github.com/jazzycamel/ws28128-rpi is a reference to drive the neopixel.

jabulonboaz, 2018-04-28:
@xiongyihui then what if i use APA102c or APA102 LEDs like the Respeaker's ?