respeaker/seeed-voicecard #69: Error when Installing Respeaker 2-mic HAT drivers on Orange Pi

gerardwr, 2018-04-23:
I tried setting up the Respeaker software acc. this Github on an Orange Pi, but get the following error:
```
pi@orangepione:~/seeed-voicecard$ sudo ./install.sh 
cat: /proc/device-tree/model: No such file or directory
Sorry, this drivers only works on raspberry pi
```

In general installing software for the Raspberry Pi will install fine on an Orange Pi.

Can I change the install.sh for the Orange PI?

Pillar1989, 2018-04-23:
Sorry, this drivers only works on raspberry pi

gerardwr, 2018-04-24:
@Pillar1989 
That's a shame. 

The CPU is on the Orange PI is an Arm, like on the RPi. Is there an obvious reason why the Respeaker would not run on the Orange?

Pillar1989, 2018-04-24:
Thay are different. You can compare bcm2835 and allwinner h3/h5/h6 datasheet.

gerardwr, 2018-04-24:
@Pillar1989 

OK, thanks for your confirmation.

closed 2018-04-24 by gerardwr