respeaker/seeed-voicecard #71: White noise

Psychokiller1888, 2018-04-26:
Hi!

I'm using respeaker 4 and respeaker 2 on many devices, on raspi 3B, 3B+ and pi zero.... On every devices the sound quality is horrible, full of white noise, hard to understand what I'm saying... Is there a way to obtain a good sound quality?

Psychokiller1888, 2018-04-27:
I can't, I'm using the devices as voice assistants

Psychokiller1888, 2018-04-27:
This is respeaker 2, not too bad, but not good either
https://soundcloud.com/laurent-chervet-psychokiller1888/test3

Respeaker 4 is far worse than that, will upload a sound test

Pillar1989, 2018-04-28:
@Psychokiller1888 any update ? I did not see your 4mic hat sound test file.

Psychokiller1888, 2018-04-28:
Oh, yeah, there you go :)

https://soundcloud.com/laurent-chervet-psychokiller1888/respeaker4

Psychokiller1888, 2018-05-02:
Any update? I don't see any reaction since I shared the audio?

Pillar1989, 2018-05-03:
@Psychokiller1888 Sorry, we just passed China's labor day, back to your issue now. Do you modify the alsamixer's setting?

Psychokiller1888, 2018-05-03:
No, it's not modified

```The IPC key of dmix or dsnoop plugin must be unique
# If 555555 or 666666 is used by other processes, use another one

pcm.!default {
    type asym
    playback.pcm "playback"
    capture.pcm "ac108"
}

pcm.playback {
    type plug
    slave.pcm "hw:0,0"
}

pcm.dmixed {
    type dmix
    slave.pcm "hw:0,0"
    ipc_key 555555
}

pcm.ac108 {
    type plug
    slave.pcm "multiapps"
}

pcm.multiapps {
    type dsnoop
slave.pcm "hw:1,0"
    ipc_key 666666
}
```

bkchr, 2019-03-17:
@Psychokiller1888 could you solve this problem in any way? I have the exact same problem and getting crazy that this is not fixable :(

Psychokiller1888, 2019-03-17:
Nope.... Seeed and @Pillar1989 don't care about this problem and won't solve it. Get rid of your respeaker 4 and get something more reliable, you can't fix it

bkchr, 2019-03-17:
Did you found any microphone without noise? I already tried several USB microphones and the respeaker 2, but all have some sort of noise :(

Psychokiller1888, 2019-03-17:
well, on a matrix voice or creator you're better qualitywise. Depends a lot on what you want to do

bkchr, 2019-03-17:
Just want to have a microphone that can record without any background noise. Don't need it for any voice activation stuff, just sound recording.

Psychokiller1888, 2019-03-17:
I don't know then, I'm not using it this way, I'm using it for voice assistants

bkchr, 2019-03-17:
Okay, thanks for your help :)

bkchr, 2019-03-17:
@Psychokiller1888 One last question, do you use Matrix Voice and have better results regarding noise?

Psychokiller1888, 2019-03-17:
sure, matrix voice is light years over respeaker