respeaker/seeed-voicecard #72: 当前4mic驱动在树莓派zerow上还是有bug

1158114251, 2018-04-29:
当前 树莓派驱动还是有问题 ，录音的时候依旧会阻塞卡死，只是有时候可以录音，输入 alsamixer  CH3 digital  经常显示0，调好后下次启动又是0，偶尔可以成功！

Pillar1989, 2018-04-30:
那应该是你自己写的代码才会那样，你可以用arecord测试一下

1158114251, 2018-04-30:
我进一步定位发现倒不是阻塞，而是录不到音，arecord也不行，也录不到音，而且我感觉和我的摄像头似乎有冲突，我的摄像头是标准的树莓派摄像头，不是usb，我看树莓派硬件也没和麦克风阵列有什么冲突啊，

1158114251, 2018-04-30:
日志
i2cdetect -y -r 1日志

  0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- UU -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --       

看dmesg
[    4.227853]  i2c_id number :0
[    4.227864]  ac108  codec_index :0
[    4.227868]  ac108  I2S data protocol type :1
[    4.231836] ac108_write error->[REG-0x00,val-0x12]

[   19.254805] AC108 PLL freq_in match:24000000, freq_out:22579200

[   19.379051] AC108 PLL freq_in match:24000000, freq_out:22579200

[   71.308609] random: crng init done
[  256.972153] AC108 PLL freq_in match:24000000, freq_out:24576000

Pillar1989, 2018-04-30:
这不是bug，树莓派所有平台的都有这些日志

1158114251, 2018-04-30:
哦，我说下情况

1158114251, 2018-04-30:
现象是这样的有时候就好使一点问题都没用，有时候就彻底卡死，然后一直卡死，不管是调节录音音量还是重启都不行

1158114251, 2018-04-30:
非常奇怪，有什么调试手段，我确认不是录音程序问题，arecord也是这个问题

1158114251, 2018-04-30:
嗯，还有个现象就是卡死重启后有时候看alsamixer里面的录音音量全部是0

Pillar1989, 2018-04-30:
alsamixer音量为0，说明你系统里面的alsa服务没有启动

1158114251, 2018-04-30:
意思alsa老是关闭

1158114251, 2018-04-30:
稍等我回去看看alsa服务，但是我的声卡是一直可以打开的，就是在录音的时候录不到卡死

1158114251, 2018-04-30:
再问下alsa的服务启动情况如何查看，和重启

1158114251, 2018-05-01:
问题好像解决了，我试了各种方法，今天上午我把 PulseAudio 卸载了，视乎就没有出现过生命问题了！

closed 2018-05-01 by 1158114251