respeaker/seeed-voicecard #74: 新版3.0 4mic 驱动安装成功后使用的时候卡死

1158114251, 2018-05-09:
dmesg输出：

[   40.345207] bcm2835-i2s 3f203000.i2s: I2S SYNC error!
[   40.348977] bcm2835-dma 3f007000.dma: DMA transfer could not be terminated
[   40.356404] bcm2835-i2s 3f203000.i2s: I2S SYNC error!
[   40.358628] bcm2835-dma 3f007000.dma: DMA transfer could not be terminated
[   40.441936] bcm2835-i2s 3f203000.i2s: I2S SYNC error!

我一直使用2.0版本驱动，昨晚更新内核成功安装了3.0 但是打开卡死阻塞在读流的时候，看dmesg输出。
我看到树莓派官方 https://github.com/raspberrypi/linux/issues/2473 也出了些东西。

turmary, 2018-05-10:
先尝试删除以下文件
/usr/lib/arm-linux-gnueabihf/alsa-lib/libasound_module_pcm_ac108.so

1158114251, 2018-05-10:
好的晚上下班我试试!

1158114251, 2018-05-10:
还有就是2.0版本驱动在alsamixer里面录音音量调高到80以上就是出现爆破，杂音，不知道在3.0有没有提升！

1158114251, 2018-05-10:
里面没有这个文件

1158114251, 2018-05-10:
我重新下载了最新系统，勉强可以录音了，但是i2s同步错误依旧，只是没有dma数据错误了，
我还发现一个bug，只要打开alsamixer ，那么关闭后录音肯定卡死，重启都不行，除非断电。
同时我发现我们的麦克风阵列录音不灵敏，录得声音特别低。我距离60cm，按照正常人说话，发现录制的声音也特别低沉，不是特别清晰干脆，连我pc的录音效果好都没有，导致语音识别的时候效果非常差。

turmary, 2018-05-11:
hi 1158114251,
我更新了一个版本v3.1，解决了不正常工作后必须断电的问题。

录音低的问题需要调节音量。
爆音的问题要看调节的模拟还是数据音量，
模拟调到100%都没问题的，数字的80就很高了。

开始录音的时候报I2S同步错误算是正常工作的。
你提的其它问题暂时没有发现。

1158114251, 2018-05-13:
好吧，弄了2天，录音效果很差，不是那么清晰，没办法

closed 2018-05-13 by 1158114251