respeaker/seeed-voicecard #77: Cannot reinstall

Psychokiller1888, 2018-05-28:
I have been trying since over a week, I cannot reinstall my respeaker

It went missing after a rpi-update

These are the logs:

https://pastebin.com/aUy6UB2n

Yes, I did remove the kernel so it would clean, yes I did try to uninstall.sh, nothing, it always fails. We see that it works in kernel 4.14.34 but tries to search, at the end, in 4.14.43

Psychokiller1888, 2018-05-28:
sudo dkms build 4.14.43-v7+ -m seeed-voicecard -v 0.3 && sudo dkms install --force 4.14.43-v7+ -m seeed-voicecard -v 0.3

As the install was failing, I tried running the dkms manually on the correct header and somehow the respeaker is now detected

meischdaddy30, 2018-06-08:
Any idea on what to do if you're getting this error for multiple files?  ex. 3.14.41-v7+ and 3.14.42-v7+

Psychokiller1888, 2018-06-08:
Dunno, and it looks like seeed doesn't care either

meischdaddy30, 2018-06-08:
Oh wait so your kernel was 4.14.43 not the 4.14.34?  My problem is i rpi-updated two times and then reverted back, so have 2 kernels higher than my current one that it's searching in..might try to either rpi again or try to delete the newer kernels

Psychokiller1888, 2018-06-09:
Yes, after the upgrade you'll get 4.14.43

Imo you can get it working by removing the newest kernel and force dkms as I wrote above

meischdaddy30, 2018-06-11:
Yeah i actually ended up removing the later kernel mods and tried reinstalling and it worked perfectly.  I just hadn't installed the headers for those later versions when i was using them so that's why it was searching and failing.  Thanks though!

Pillar1989, 2018-06-19:
@Psychokiller1888 any update ?

Psychokiller1888, 2018-06-19:
Update on what? It's you that should update.... And why did you close the other bug report where I can't install on an updated pi?

Pillar1989, 2018-06-19:
@Psychokiller1888 
is that working for you ?
```
sudo dkms build 4.14.43-v7+ -m seeed-voicecard -v 0.3 && sudo dkms install --force 4.14.43-v7+ -m seeed-voicecard -v 0.3
```

Psychokiller1888, 2018-06-19:
No, not since last raspbian build

kali, 2018-06-19:
Or to clarify (I have followed @Psychokiller1888 issue for Snips) it's since the last kernel update using rpi-update.

Pillar1989, 2018-06-20:
@Psychokiller1888 Maybe you can flash the last raspbian, then reinstall it

Psychokiller1888, 2018-06-20:
Doesn't change anything

Pillar1989, 2018-06-20:
fine

closed 2018-06-20 by Pillar1989

Psychokiller1888, 2018-06-20:
Is that how you solve install problems? You close the bug reports? Time to find something else than seeed I guess

KillingJacky, 2018-06-20:
Hi @Psychokiller1888 I think @Pillar1989 had some problem in understanding you, he thought you've solved your problem yourself. Thanks to the investigations from you guys here and https://github.com/respeaker/seeed-voicecard/issues/83 , I think the installer script can be improved indeed.
@Pillar1989 tried to merge the two issues, but I think we should keep the later one https://github.com/respeaker/seeed-voicecard/issues/83 , in which we should discuss how to improve the installer script to automatically adapt different kernel versions through updates. So gentlemen I'm gonna keep this issue closed and reopen https://github.com/respeaker/seeed-voicecard/issues/83