respeaker/seeed-voicecard #79: Support for Ubuntu Mate 16.04 Raspberry Pi version

auvilink, 2018-06-02:
Hi, there, 

Just installed Ubuntu Mate 16.04 Raspberry Pi version (https://ubuntu-mate.org/raspberry-pi/) as well as the 4-mic array for raspberry pi driver according to wiki 
cd seeed-voicecard
sudo ./install.sh

after reboot

arecord -L shows
null
    Discard all samples (playback) or generate zero samples (capture)
pulse
    PulseAudio Sound Server
default
    Playback/recording through the PulseAudio sound server
playback
dmixed
ac108
multiapps

Does that mean this platform is not supported?

Thank you.

auvilink, 2018-06-02:
make KERNELRELEASE=4.4.38-v7+ -C /lib/modules/4.4.38-v7+/build M=/var/lib/dkms/seeed-voicecard/0.3/build.................(bad exit status: 2)
ERROR (dkms apport): binary package for seeed-voicecard: 0.3 not found
Error! Bad return status for module build on kernel: 4.4.38-v7+ (armv7l)

Pillar1989, 2018-06-02:
#38

Pillar1989, 2018-06-02:
#38

closed 2018-06-02 by Pillar1989

whb883, 2019-01-18:
Still no support plan?