respeaker/seeed-voicecard #81: seed 4-mic array with hifiberry amp2

marcelveldt, 2018-06-06:
Hi,

So I'm building my own smart speaker and purchased the 4mic respeaker hat to add Alexa support.
On the Pi I already have a HifiBerry AMP2 connected for the audio output (as the onboard audio is crap on the pi).

Well, this doesn't seem to work as the respeaker wants to use the i2s interface (which is off course already used by the hifiberry). Somehow it should be possible to merge the two overlays where the HifiBerry can use the i2s bus for output while the respeaker uses it for input but I have no clue how.

Any ideas to get this working or should I just return the ReSpeaker HAT ?

Pillar1989, 2018-06-08:
I dont have any ideas. Maybe you should return the HifiBerry.

marcelveldt, 2018-06-08:
OK thanks! I'll switch to the MicArray board instead.

closed 2018-06-11 by Pillar1989