respeaker/seeed-voicecard #82: Respeaker 2-Mics PI Hat not found

rgall218, 2018-06-08:
Greetings 
I have two Respeaker 2-mic PI Hat's. I have followed instructions for installing the drivers.
When I enter  play -l  The ReSpeaker pi hat is not found also with arecord -L no ReSpeaker mic is listed. I have tried using  both a Raspi 2 and Raspi 3.

I have tried an number of times for past couple of months both uninstalling and removing the directory.
and downloading  and installing again.
This message was in the kernel log
Jun  8 20:11:40 raspberrypi kernel: [    2.596589] snd_soc_seeed_voicecard: loading out-of-tree module taints kernel

which is normal I think.

I looked in the 'messages' log there was nothing related to ReSpeaker after reboot

Pillar1989, 2018-06-09:
Maybe you try a new raspbian image and install seeed-voicecard again. If you still have this issue, Please send me the install  log.

rgall218, 2018-06-10:
Created a new image and the 2-mic PI Hat was found using aplay -l.

Thanks It rough way to solve an issue,

Pillar1989, 2018-06-11:
[image]

Sorry, Raspbian didnt support jessie os.

ianr78, 2018-06-18:
Hi. I'm having similar problem of ReSpeaker not showing up. I've installed and reinstalled a few times now, done fresh builds of raspbian, full and lite. The installation process goes through error free ok but after reboot, running aplay -l or arecord -l, just not seeing the voicecard at all. Any hints where to start looking with this? I'm new to this device. I have it connected up to a Pi Zero W. Thanks

closed 2018-06-19 by Pillar1989