respeaker/seeed-voicecard #83: Can't install

Psychokiller1888, 2018-06-11:
Raspberry pi 3b+ after rpi-update on 4.14.48+ fails installing:

Building module:
cleaning build area...
make -j4 KERNELRELEASE=4.14.34-v7+ -C /lib/modules/4.14.34-v7+/build M=/var/lib/dkms/seeed-voicecard/0.3/build......
cleaning build area...

DKMS: build completed.
Error! echo
Your kernel headers for kernel 4.14.48+ cannot be found at
/lib/modules/4.14.48+/build or /lib/modules/4.14.48+/source.
Error! echo
Your kernel headers for kernel 4.14.48-v7+ cannot be found at
/lib/modules/4.14.48-v7+/build or /lib/modules/4.14.48-v7+/source.
mkdir: cannot create directory ‘/etc/voicecard’: File exists
------------------------------------------------------
Please reboot your raspberry pi to apply all settings
Enjoy!
------------------------------------------------------

meischdaddy30, 2018-06-11:
Didn't you just have this same problem like 2 weeks ago?

sudo apt-get install linux-headers-$(uname -r)

If not rollback before the rpi-update, remove the 4.14.48+ and 4.14.48-v7+ module directories and then it'll work.

Psychokiller1888, 2018-06-12:
Isn't that not my problem and seeed installer to fix and make it work? This is a bug reporter and technically a help seeking submit system. As you can see, Seeed did not even bother answering

Pillar1989, 2018-06-19:
#77

KillingJacky, 2018-06-20:
Hey @Psychokiller1888 I'm trying to copy what steps you took, please correct me if I'm wrong about anywhere.
1. You're using the `4.14.34` kernel, you updated your kernel to `4.14.43` with rpi-update command, you tried to uninstall the driver and reinstall it, but failed in `uninstall.sh`
2. `sudo dkms build 4.14.43-v7+ -m seeed-voicecard -v 0.3 && sudo dkms install --force 4.14.43-v7+ -m seeed-voicecard -v 0.3` temporarily let you install the driver for kernel `4.14.43`
3. You updated your kernel to `4.14.48` with rpi-update, the same situation happened again.

Actually I'm totally agree with you, flashing a fresh image for every kernel update is kind of time-consuming thing. But what shocks me is that, the version iteration of raspbian's kernel is really frequent🤪 Now we have to provide friendly support for the update. We will find out a way to better solve this.

Psychokiller1888, 2018-06-20:
- I'm using the latest Raspbian
- I do sudo raspi-update that last time I tried updated to 4.14.48
- Reboot
- Respeaker doesn't work anymore, arecord is empty
- Uninstall.sh respeaker
- Try to install it again

yuehaowang, 2018-06-28:
I solved this problem with these steps:

1. Update your operating system:

```
sudo rpi-update
```

After update, reboot is required.

2. Then, update the linux headers because I came across some problems on finding **the latest** linux headers.

```
sudo rpi-source --skip-gcc
```

Reboot is also required here.

**Note:** if you don't have rpi-source on your system, try to download and install it with the help of [the documentation](https://github.com/notro/rpi-source/wiki).

closed 2018-07-16 by Pillar1989