respeaker/seeed-voicecard #85: Artikpi - need binary compiled for the DD60 version

krishkal, 2018-06-11:
The binary in the Artikpi branch is for the older 01Q5 version of the firmware. The latest (and ONLY) version of the firmware available on the Artik.io website is version DD60. It appears that the 01Q5 binary is not compatible with that version, because it does not work.

Please update the binary version to be compatible with the Artik software.

Pillar1989, 2018-06-12:
[image]
01Q5  is latest version. Where can I find version DD60 ?

krishkal, 2018-06-12:
Hi Pillar1989, DD60 is the image that is on the official developer website, which was released on 06/05/2018. In fact, I am unable to download the 01Q5 at this time. Can you give me a link to where I can download it?

DD60 can be found here: https://developer.artik.io/documentation/downloads.html. It is the first link on that page which is found at: https://developer.artik.io/downloads/2768bd32-383a-44cf-9e90-9292712a96ba/download.

Pillar1989, 2018-06-12:
I think you chose the wrong image. Please checkout
[image]
https://developer.artik.io/documentation/downloads.html

krishkal, 2018-06-12:
Yes, as I said in the previous message, I did find that image, and it works.

However, the version I chose IS  the latest version. It is NOT the "wrong image" as you say. It works fine on the Artik board. But, the binaries on this repo are not compatible with this latest version. You really should fix that.

Pillar1989, 2018-06-13:
Let me confrim the latest artik board BSP source code is https://github.com/SamsungARTIK/manifest/blob/master/artik533s_bsp.xml @krishkal

Pillar1989, 2018-06-19:
https://github.com/respeaker/seeed-voicecard/commit/111fb0be89197024ddc5920e4cba0216ca735899

closed 2018-06-19 by Pillar1989