respeaker/seeed-voicecard #86: Question: artikpi wm8960 formats

ezaluzec, 2018-06-27:
I'm using the Samsung Artik530s Eagleye platform. 
On the artikpi branch, I see that the wm8960 driver supports i2s `SNDRV_PCM_FMTBIT_S32_LE` in `WM8960_FORMATS`. 
Is the wm8960 driver compatible with i2s interface on artik530s in SNDRV_PCM_FMTBIT_S32_LE format?

Pillar1989, 2018-06-28:
artiki530s i2s driver only support SNDRV_PCM_FMTBIT_S24_LE or SNDRV_PCM_FMTBIT_S16_LE .
Please refer to 
https://github.com/SamsungARTIK/linux-artik/blob/A530s_os_3.2.0/sound/soc/nexell/nexell-i2s.c#L537-L543

closed 2018-07-16 by Pillar1989

ezaluzec, 2018-07-16:
Correct reference to artik530s i2s driver. 
@Pillar1989 , could you achieve pseudo S32 format by setting the BLC to 24 format reading the L/R clk down the FIFO?