respeaker/seeed-voicecard #89: Can't install, reboot pi when attempting building the module

Overdrive83, 2018-07-13:
Hi,
I tried to install the driver from the last raspbian stretch lite image and from a raspbian stretch from March with no luck.
Each time I run the installation script it goes on until it reach the building the module part and then the pi reboot the "make" command is launched.
I have obviously no message because it is instantly rebooting.
I saw the other issue from someone having trouble with the installation but I can't figure out if my case can be related to his.

Below the attempt on the older stretch :
`...
Creating symlink /var/lib/dkms/seeed-voicecard/0.3/source ->
                 /usr/src/seeed-voicecard-0.3

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area....
make -j4 KERNELRELEASE=4.14.52+ -C /lib/modules/4.14.52+/build M=/var/lib/dkms/seeed-voicecard/0.3/build...
`

4.14.52+ seems right, I can find it in /lib/modules/
Any idea ? I just can't use my 2 mics hat right now...

Thx

Pillar1989, 2018-07-16:
@Overdrive83  I'm very sorry for what happened to you. 
You can  check your problems in three ways:
-  There was no installation error while executing the `./install.sh` script. Failure to download raspberry-kernels-headers is a frequent occurrence because of network problems,  Switch Raspbian Repository Mirrors（https://www.raspbian.org/RaspbianMirrors） to your local network.

- In retrospect, have you ever updated your system with "sudo rpi-update". Currently seeed-voicecard does not support updating the system in this way, and the correct way to update the system should be "sudo apt upgade".

- Check that the compiled version of your kernel matches the final version used. eg ,You  compiled version is 4.14.52+, and last your runtime system linux kernel version should be 4.14.52+(`uname -a`). The Linux kernel version is important for the dependent software, so it's best to know what you're doing before you change it

Overdrive83, 2018-07-16:
Hi,
Thx for your response.
For your questions : 
- There was no error indeed during the ./install.sh until the reboot;
- My first try was with the latest image (2018-06-27) from raspberrypi.org so the only update I made was using apt update/upgrade. After that, among other things I did try using @yuehaowang indications from the issue #83 (based on a rpi-update). It lead me no were;
- My linux kernel was the one mentioned during compilation (at the time I checked with uname -a as you mentioned).

As a solution I installed a stretch version from March and did a git checkout from the "install" branch. Don't ask me why, it wasn't working with the main one (same reboot problem when compiling), but it did with the "install" one. I didn't even bothered comparing them cause I was just glad having a solution.
So I won't update my kernel for a while and everything will be find for me.

I would suggest you to get your hands on a RPI 3b+ and the june stretch raspbian to check it out for yourself if not done already. The pb was a bit disappointing for someone like me who never bought anything from you before. It's easy to follow my steps : I was just flashing the version and then following directly after that your wiki like 6 command lines install instructions !

Thx again anyway.

jerryyip, 2018-07-18:
@Overdrive83, I have tried on RPI 3B+ with 0627 stretch image. It works well with 2mic, 4mic, and 8mic. All I did was just `git clone https://github.com/respeaker/seeed-voicecard` and `sudo ./install.sh` and `sudo reboot`. I didn't make a `apt upgrade`.

jerryyip, 2018-07-18:
@Overdrive83 I try `apt upgrade` too. I am using 0627 stretch image, so my kernel version `4.14.52-v7+` doesn't change after `apt upgrade`. And 2mic works.
So please tell us what you did after flashing the 0627 stretch image to your SD card.

Overdrive83, 2018-07-18:
@jerryyip My only commands after flashing the raspbian image with etcher were exactly :
sudo apt-get update
sudo apt-get upgrade
git clone https://github.com/respeaker/seeed-voicecard.git
cd seeed-voicecard
sudo ./install.sh
sudo reboot
It's pretty simple cause it was a copy/paste from http://wiki.seeedstudio.com/ReSpeaker_4_Mic_Array_for_Raspberry_Pi/

jerryyip, 2018-07-20:
@Overdrive83 Wait...you said raspberrypi would reboot instantly when installing. So why were you `sudo reboot ` again?

jerryyip, 2018-07-20:
In your case, the installation cannot be finished because that "instant reboot" happens every time. So I advise you just don't `sudo apt update` and `sudo apt upgrade` after flashing the 0627 raspbian image. And try again:
```
git clone https://github.com/respeaker/seeed-voicecard.git
cd seeed-voicecard
sudo ./install.sh
sudo reboot
```
pls let me know if it reboots instantly again or not.

Overdrive83, 2018-07-20:
My mistake, of course I didn't went to the `sudo reboot` part because of the crash during the `install.sh` script.
Unfortunately as I stated before I found a solution with an old raspbian image and the install branch of you software.
It pretty much suits me like this. I just lost a lot of time but it's somewhat part of the DIY journey I suppose !

My pi is installed in my car right now, the mic hat used for the "ok google" of my open auto project. It's working well so in the feature if I update the thing I well try again. No doubt it will work this time :)
Thx for your time, as I won't reproduce the issue, feel free to close it or not.

closed 2018-07-28 by jerryyip